espressif/esp-mesh-lite #78: How to use esp_mesh_lite_set_fusion_config to set the fusion signal strength

yel-best, 2024-04-17:
use last version mesh-lite
use esp-idf 5.1.3

I set the fusion parameters directly when calling the `app_wifi_set_softap_info `function when initializing mesh-lite, but it didn't seem to achieve the desired effect.

I have some confusion about the parameters of this fusion signal strength, as follows:

* **I understand that with the same mesh ID, if the nearest mesh network signal is detected to be lower than the threshold, the connection will not be selected. Will it allow itself to become a root node?**
* **I am currently testing that the connection will continue even if it is lower than the threshold. For example, if I configure the value to be -75, I will also see that I will connect to a network with an RSSI strength of -85.**

The control I understand is that under the same mesh, if certain conditions are not met, it will automatically become a root node. This allows me to ensure that devices on multiple floors are present when using a mesh with the same SSID. Network; it is possible to elect to become the root node or automatically become the root node by detecting network signals that do not meet the threshold;

```C

// Initialize fusion configuration with specified values.
esp_mesh_lite_fusion_config_t fusion_config = {
    .fusion_start_time_sec = 0, // Set the fusion to start 60 seconds after the mesh network is up.
    .fusion_frequency_sec = 120, // Set the network to perform fusion every 120 seconds thereafter.
    .fusion_rssi_threshold = -75,
};

void app_wifi_set_softap_info(void)
{
    char softap_ssid[32];
    uint8_t softap_mac[6];
    esp_wifi_get_mac(WIFI_IF_AP, softap_mac);
    memset(softap_ssid, 0x0, sizeof(softap_ssid));

#ifdef CONFIG_BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC
    snprintf(softap_ssid, sizeof(softap_ssid), "%.25s_%02x%02x%02x", CONFIG_BRIDGE_SOFTAP_SSID, softap_mac[3], softap_mac[4], softap_mac[5]);
#else
    snprintf(softap_ssid, sizeof(softap_ssid), "%.32s", CONFIG_BRIDGE_SOFTAP_SSID);
#endif

    esp_mesh_lite_set_softap_ssid_to_nvs(softap_ssid);
    esp_mesh_lite_set_softap_psw_to_nvs(CONFIG_BRIDGE_SOFTAP_PASSWORD);
    esp_mesh_lite_set_softap_info(softap_ssid, CONFIG_BRIDGE_SOFTAP_PASSWORD);

    
    esp_err_t result = esp_mesh_lite_set_fusion_config(&fusion_config);
    if (result == ESP_OK)
    {
       // RSSI threshold list set successfully
        ESP_LOGI(TAG, "RSSI esp_mesh_lite_set_fusion_config set successfully");
     }
    else
    {
       // Failed to set the RSSI threshold list
         ESP_LOGI(TAG, "RSSI esp_mesh_lite_set_fusion_config set Failed");
     }
}

main(){
esp_bridge_create_all_netif();
    app_wifi_init();

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);
    esp_event_handler_instance_register(ESP_MESH_LITE_EVENT, ESP_EVENT_ANY_ID, &esp_mesh_lite_event_cb, NULL, NULL);
    // set 1 level node ,is mesh root node
    //  esp_mesh_lite_set_allowed_level(1);
    app_wifi_set_route_info();
    app_wifi_set_softap_info();
    esp_mesh_lite_scan_cb_register(&mesh_lite_scan_cb);

}
```

tswen, 2024-04-19:
Based on my understanding of what you described, what you need is the settings for this API, not the fusion config settings. The fusion logic triggers when there are multiple root nodes. Devices with weaker relative router signal strengths will try to connect to the root node with the strongest signal unless the signal strength between them falls below the fusion_rssi_threshold.

https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/include/esp_mesh_lite_core.h#L427

esp_mesh_lite_set_rssi_threshold(NULL, 0, -75);

tswen, 2024-04-20:
Yes, ideally, for a Mesh network, there should only be one root node. However, we can't rule out the possibility of having multiple root nodes at certain times, such as when [multiple devices power on simultaneously](https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide.md#general-process) or when a device connects directly to a router because it didn't scan for a parent node due to some circumstances. 
Therefore, there's a fusion mechanism in place. When there are multiple root nodes in the network, it attempts to merge these root nodes into one.