espressif/esp-mesh-lite #82: Repeating error from no_router example root node:  E (32493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm

martenainult, 2024-05-05:
Specs:
- Project name:     no_router
- Mesh-Lite:       esp-mesh-lite component version: 0.10.3
- boot:               ESP-IDF v5.1.2-dirty 2nd stage bootloader
- Chip rev:         v3.1 (ESP32-WROOM-32E)

Problem: 
After booting up a ESP-MESH-LITE system without a network connection there is a timer triggered from the root node. Start time is 30 s alive time and triggers every 5 seconds. From digging through the included code it seems to come from the esp_modem  or AP setup side. Perhaps it's due to the fact that some watchers require router AP? 

Is there a quick fix available to suppress the error that pops up from the included libraries? 

`I (2488) [vendor_ie]: Mesh ID: 77
I (2494) [ESP_Mesh_Lite_Comm]: msg action add success 
I (2499) [ESP_Mesh_Lite_Comm]: Bind Socket 54, port 6364
I (2504) [ESP_Mesh_Lite_Comm]: Bind Socket 55, port 6363
I (2510) [ESP_Mesh_Lite_Comm]: Bind Socket 56, port 6366
I (2516) [ESP_Mesh_Lite_Comm]: Bind Socket 57, port 6365
I (2523) [mesh-lite-espnow]: Start espnow task
I (2527) [ESP_Mesh_Lite_Comm]: msg action add success
I (2535) no_router: Root node
I (2537) Mesh-Lite: Mesh-Lite connecting
I (2541) main_task: Returned from app_main()
I (7317) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<255,255>, prof:11
I (7319) wifi:station: fc:b4:67:53:39:a0 join, AID=1, bgn, 40D
I (7335) bridge_wifi: STA Connecting to the AP again...
I (7362) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (8376) wifi:<ba-add>idx:2 (ifx:1, fc:b4:67:53:39:a0), tid:0, ssn:0, winSize:64
E (32493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (37493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (42493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (47493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (52493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (57493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm 
E (62493) [ESP_Mesh_Lite_Comm]: ESP_Mesh_Lite_Comm `

Thank you for the help :)

tswen, 2024-05-09:
Hello, this print statement does not affect the functionality, and we will fix this issue later.

martenainult, 2024-05-09:
Hi @tswen. Thank you for the reply!

The print statement was bugging my nerves a lot when there was a need to monitor the ESP32 devices. As the package was used with a serial transfer system through the UART0 channel, the console errors started to affect the data transfer also. In the middle of development I switched the console log channel to a custom UART and the issue is under a rug for now.