espressif/esp-mesh-lite #83: IP获取失败

LYL14, 2024-05-07:
mesh-lite组网后，根节点掉线。有部分设备重组失败，两种情况：
1. 能获取到mesh层数，但IP是0; wifi:Haven't to connect to a suitable AP now!
2. 也有部分设备，IP正常但是TCP通讯不了。即使TCP服务端关闭，ESP32仍能TCP发送成功。

nefusmzj, 2024-05-07:
基于localcontrol例子，遇到了相同的问题。

30个设备组网，根节点意外断开后，等待几分钟后:

1. 少部分设备可自动恢复，与服务端TCP通信；
2. 一部分设备无法获得IP，因此也无法通信；
3. 一部分设备可以获得IP，tcp任务中的send或write函数一或write函数发送，一直返回直返回发送字节长度（即成功），但实际服务端没有收到，连接已经断开。

上述问题与楼上遇到的问题很相似，期待乐鑫技术能答疑解惑。

LYL14, 2024-05-07:
mesh-lite有断开、停止、重新连接的函数吗，mesh_lite_start（）后可以使用WIFI的api吗，例如WIFI的断开、重连等函数。

tswen, 2024-05-09:
1. 请问根节点意外断开是指：根节点下电、根节点软重启、根节点硬重启哪一种
2. 对于无法获取 ip 的问题：可以将 mesh_lite 更新到最新，然后重新编译，让 mesh lite 拉取最新的 iot_bridge 组件（0.11.4），使用最新的试一下。新的 iot-bridge 组件修复了获取不到 IP 的问题。另外需要在 idf/components/lwip/apps/dhcpserver/dhcpserver.c 里面加如下一行代码
[image]

3. 获得到 ip 但是应用层通信异常的问题：是否可能是其父节点遇到了第二点的情况，导致数据没法从其父节点出去？这个可以观察一下。

tswen, 2024-05-09:
你们使用的 idf 是多少呢？建议使用 5.0 或者 5.1

LYL14, 2024-05-09:
> 1. 请问根节点意外断开是指：根节点下电、根节点软重启、根节点硬重启哪一种
> 2. 对于无法获取 ip 的问题：可以将 mesh_lite 更新到最新，然后重新编译，让 mesh lite 拉取最新的 iot_bridge 组件（0.11.4），使用最新的试一下。新的 iot-bridge 组件修复了获取不到 IP 的问题。另外需要在 idf/components/lwip/apps/dhcpserver/dhcpserver.c 里面加如下一行代码
> 
> <img alt="企业微信截图_9eb8229d-61f5-476e-a217-8beb06bf1dc3" width="723" src="https://private-user-images.githubusercontent.com/55544524/329158496-3832591b-06dc-493d-b898-fdf3343ccba7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUyNDg4MTAsIm5iZiI6MTcxNTI0ODUxMCwicGF0aCI6Ii81NTU0NDUyNC8zMjkxNTg0OTYtMzgzMjU5MWItMDZkYy00OTNkLWI4OTgtZmRmMzM0M2NjYmE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA5VDA5NTUxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlOTgyMWRmMThiMzg5NWEzOTJjMmQxYzQ3MmRiNjViN2QyOTY4ZWJiMDlmOWQ1NTU5ZmY5Mjc5MGViYzI1MzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Tuwseji3kaG8VJcy4KiQBN1APOgXKbXZ10hkKFnSZFQ">
> 3. 获得到 ip 但是应用层通信异常的问题：是否可能是其父节点遇到了第二点的情况，导致数据没法从其父节点出去？这个可以观察一下。

好的，谢谢，更新到最新版本组网ip可以获取了

closed 2024-06-20 by LYL14