espressif/esp-mesh-lite #84: Mesh-lite 根节点关机，有概率无法重建Mesh-Lite网络，怎么解决

cuijunling00, 2024-05-07:
各位尊敬的作者，我在使用Mesh-Lite功能是遇到了一些问题，希望大家能帮我解决。具体问题如下：
我是使用mesh_local_control为模板创建的工程，并且使用TCP协议和上位机连接，主要功能是Mesh-Lite的重建网络功能。
我在测试Mesh-Lite网络丢失根节点后的重建网络的能力。发现会有很大的概率无法重建网络，或者是部分模块无法加入到重新建立的网络中。 具体的日志都停止在检测网段冲突的位置。下面是我的日志
个别模块重启后无法加入网络的日志：
I (1201) wifi:wifi driver task: 3fcc094c, prio:23, stack:6656, core=0
I (1211) wifi:wifi firmware version: 91b9630
I (1211) wifi:wifi certification version: v7.0
I (1211) wifi:config NVS flash: enabled
I (1211) wifi:config nano formating: disabled
I (1221) wifi:Init data frame dynamic rx buffer num: 32
I (1221) wifi:Init static rx mgmt buffer num: 5
I (1231) wifi:Init management short buffer num: 32
I (1231) wifi:Init dynamic tx buffer num: 32
I (1241) wifi:Init static tx FG buffer num: 2
I (1241) wifi:Init static rx buffer size: 1600
I (1241) wifi:Init static rx buffer num: 10
I (1251) wifi:Init dynamic rx buffer num: 32
I (1251) wifi_init: rx ba win: 6
I (1261) wifi_init: tcpip mbox: 32
I (1261) wifi_init: udp mbox: 6
I (1261) wifi_init: tcp mbox: 6
I (1271) wifi_init: tcp tx win: 5744
I (1271) wifi_init: tcp rx win: 5744
I (1281) wifi_init: tcp mss: 1440
I (1281) wifi_init: WiFi IRAM OP enabled
I (1291) wifi_init: WiFi RX IRAM OP enabled
I (1291) phy_init: phy_version 620,ec7ec30,Sep  5 2023,13:49:13
I (1341) wifi:mode : null
I (1341) wifi:mode : softAP (f4:12:fa:8a:50:31)
I (1341) wifi:Total power save buffer number: 16
I (1341) wifi:Init max length of beacon: 752/752
I (1341) wifi:Init max length of beacon: 752/752
I (1351) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1351) bridge_common: [WIFI_AP_DEF ]
I (1361) bridge_common: Set ip info:192.168.4.1
W (1361) bridge_common: SoftAP IP network segment has changed, deauth all station
Add netif ap with (commit id)
I (1371) bridge_common: netif list add success
I (1381) wifi:Disabled PMF config for SoftAP
I (1381) wifi:Total power save buffer number: 16
Add netif sta with (commit id)
I (1391) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1401) bridge_common: [WIFI_AP_DEF ]Name Server1: 114.114.114.114
I (1411) bridge_common: netif list add success
I (1411) wifi:mode : sta (f4:12:fa:8a:50:30) + softAP (f4:12:fa:8a:50:31)
I (1421) wifi:enable tsf
I (1421) bridge_wifi: Found ssid TPLOGIN001
I (1421) bridge_wifi: Found password kwhz1234
E (1431) NVS: Failed to read IP info from NVS
I (1441) wifi:Disabled PMF config for STA
I (1441) bridge_wifi: [esp_bridge_wifi_set_config] sta ssid: KWHZ_Line password: kwhz1234
I (1451) bridge_wifi: [esp_bridge_wifi_set_config] softap ssid: ESP_Bridge password: 12345678
W (1451) wifi:Haven't to connect to a suitable AP now!
I (1951) wifi:Total power save buffer number: 16
I (1961) Mesh-Lite: esp-mesh-lite component version: 0.10.2
W (1961) wifi:Haven't to connect to a suitable AP now!
I (1961) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1971) bridge_common: [WIFI_AP_DEF ]Name Server1: 114.114.114.114
I (1971) ESPNOW: espnow [version: 1.0] init
I (1981) [vendor_ie]: Mesh-Lite commit id: 6bca946

I (1981) [vendor_ie]: Mesh ID: 77 
I (1991) [ESP_Mesh_Lite_Comm]: msg action add success 
I (1991) [ESP_Mesh_Lite_Comm]: Bind Socket 54, port 6364 
I (2001) [ESP_Mesh_Lite_Comm]: Bind Socket 55, port 6363 
I (2011) [ESP_Mesh_Lite_Comm]: Bind Socket 56, port 6366 
I (2011) [ESP_Mesh_Lite_Comm]: Bind Socket 57, port 6365 
I (2021) [ESP_Mesh_Lite_Comm]: msg action add success 
I (2021) [mesh-lite-espnow]: Start espnow task 
I (2031) [ESP_Mesh_Lite_Comm]: msg action add success 
[... 467 lines cut ...]

I (429171) wifi:<ba-del>idx:0, tid:0
I (429171) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (429191) wifi:mode : sta (f4:12:fa:8a:50:10) + softAP (f4:12:fa:8a:50:11)
I (429191) wifi:Total power save buffer number: 16
W (429191) wifi:Haven't to connect to a suitable AP now!
W (429201) WIFI_EVENT: star
W (429201) WIFI_EVENT: WIFI_EVENT_STA_DISCONNECTED
W (429211) WIFI_EVENT: end

I (429211) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.6.1
I (429221) bridge_common: [WIFI_AP_DEF ]Name Server1: 192.168.1.1
I (429231) [ESP_Mesh_Lite_Comm]: Bind Socket 54, port 6364 
I (429231) [ESP_Mesh_Lite_Comm]: Bind Socket 55, port 6363 
I (429241) [ESP_Mesh_Lite_Comm]: Bind Socket 56, port 6366 
I (429241) [ESP_Mesh_Lite_Comm]: Bind Socket 57, port 6365 
W (429251) WIFI_EVENT: star
W (429251) WIFI_EVENT: WIFI_EVENT_AP_START
W (429261) WIFI_EVENT: end

W (429271) wifi:Haven't to connect to a suitable AP now!
W (429321) wifi:Haven't to connect to a suitable AP now!
W (429441) wifi:Haven't to connect to a suitable AP now!
希望大神给我一个解决方案。 
这个重建网络的稳定性对我很重要。 再次感谢各位

tswen, 2024-05-09:
可以将 mesh_lite 更新到最新，然后重新编译，让 mesh lite 拉取最新的 iot_bridge 组件（0.11.4），使用最新的试一下。新的 iot-bridge 组件修复了获取不到 IP 的问题。

另外需要在 idf/components/lwip/apps/dhcpserver/dhcpserver.c 里面加如下一行代码
[image]

cuijunling00, 2024-05-17:
tswen 你好。 我使用的编程环境是Windows10, VSCode和ESP-IDF插件 ESP-IDF版本是esp-idf-v5.1.2。我使用idf.py add-dependency "espressif/mesh_lite^0.10.3"进行下载使用Mesh-Lite，这下载的iot_bridge是0.11.3版本。我在github上下载了最新的https://github.com/espressif/esp-iot-bridge.git代码，将这个工程中的\esp-iot-bridge-master\components\iot_bridge这个目录下的所有文件复制到\wifi_paper_v1.0\managed_components\espressif__iot_bridge这个目录下，在将这个espressif__iot_bridge文件夹复制到wifi_paper_v1.0\components这个文件夹中。然后重新编译，结果爆出了以下的错误。

NOTICE: Skipping optional dependency: espressif/esp_modem
Manifest files have changed, solving dependencies.
Using component placed at D:\ESP32LearnCode\wifi_paper_v1.0\components\espressif__iot_bridge for dependency espressif/iot_bridge(0.*)(introduced by component espressif/mesh_lite), specified in D:/ESP32LearnCode/wifi_paper_v1.0/main\idf_component.yml
...........Using component placed at D:\ESP32LearnCode\wifi_paper_v1.0\components\espressif__iot_bridge for dependency espressif/iot_bridge(0.*)(introduced by component espressif/mesh_lite), specified in D:/ESP32LearnCode/wifi_paper_v1.0/main\idf_component.yml
..CMake Error at D:/Espressif/frameworks/esp-idf-v5.1.2/tools/cmake/build.cmake:540 (message):
  ERROR: 'git ls-remote --exit-code origin HEAD' failed with exit code 128

  error: RPC failed; curl 28 Recv failure: Connection was reset

  fatal: expected flush after ref listing

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.1.2/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "D:/ESP32LearnCode/wifi_paper_v1.0/build/CMakeFiles/CMakeOutput.log".

这个要怎么解决，要怎么用。 我的电脑网络是通畅的，简单的说就是能上google.

谢谢。 告诉我要怎么解决这个问题。

guo652917087, 2024-07-03:
> 可以将 mesh_lite 更新到最新，然后重新编译，让 mesh lite 拉取最新的 iot_bridge 组件（0.11.4），使用最新的试一下。新的 iot-bridge 组件修复了获取不到 IP 的问题。
>
> 另外需要在 idf/components/lwip/apps/dhcpserver/dhcpserver.c 里面加如下一行代码 [...]

最新版本【[esp-idf](https://github.com/espressif/esp-idf/tree/e7070e777a079695f69720ffb3c631c5fe620cc6)】
                [ [esp-mesh-lite](https://github.com/espressif/esp-mesh-lite/tree/master)](https://github.com/espressif/esp-mesh-lite/commit/80adbffb478d852256cc5fb36a1bca96956427af)

根节点断电也会出现不能组网。

我用例子“no_router”，三个模块测试。根节点断电前，两个模块分别在level2和level3.手动断电根节点，会出现level2会连接到level3的AP，level3连接到level2的AP。level2和level3两个模块会反复循环互相连接。根节点上电后也不能恢复组网。重启level3的节点无法修复网络，只有重启level2，等待几分钟才三个模块重新恢复mesh网络。日志方面就不贴了，已经告诉了测试复现的方法。希望对维护团队有帮助。

tswen, 2024-07-09:
```
diff --git a/examples/no_router/main/no_router.c b/examples/no_router/main/no_router.c
index 2fbf53c..8da76d2 100644
--- a/examples/no_router/main/no_router.c
+++ b/examples/no_router/main/no_router.c
@@ -128,6 +128,7 @@ void app_main()
     esp_mesh_lite_set_allowed_level(1);
 #else
     ESP_LOGI(TAG, "Child node");
+    esp_mesh_lite_set_disallowed_level(1);
 #endif
 
     esp_mesh_lite_start();
```

按照如上 patch 进行修改可以修复子节点相互连接的问题。

iwonako, 2024-12-10:
@guo652917087 @cuijunling00 ，两位好
请问这个库该怎么安装/使用，我是在 vsc 插件中安装的 idf-5.1.1，然后 git clone 了这个 mesh lite，在 vsc 打开例程，点 menuconfig 和 build 都报错了