espressif/esp-mesh-lite #86: 怎么为根节点设置静态IP

liuquanli1970, 2024-05-14:
我想根节点设置静态IP，便于TCP服务端连接。
试了几次，都会造成设备重启。
请高手指点一下。谢谢！

liuquanli1970, 2024-05-28:
是的，是使用 esp_netif_set_ip_info 接口来给根节点的 station 的 netif 设置静态 ip，单独设置，会造成mesh-lite重启。
esp_netif_create_default_wifi_sta();
esp_netif_create_default_wifi_ap();
wifi_config_t wifi_sta_cfg = {
    .sta = {
        .ssid = "YourSTASSID", // 替换为你要连接的WiFi名称
        .password = "YourSTAPassword", // 替换为密码
    },
};

wifi_config_t wifi_ap_cfg = {
    .ap = {
        .ssid = "YourAPSSID", // 设置你的AP热点名称
        .password = "YourAPPASSWORD", // 设置AP热点密码
        .authmode = WIFI_AUTH_WPA2_PSK, // 可选，设置认证方式
    },
};

esp_netif_t *wifi_netif = esp_netif_create_default_wifi_sta();
esp_netif_ip_info_t ip_info;
        memset(&ip_info, 0, sizeof(esp_netif_ip_info_t));
        ip_info.ip.addr = esp_ip4addr_aton("192.168.1.250");      // 设置静态IP
        ip_info.netmask.addr = esp_ip4addr_aton("255.255.255.0"); // 设置子网掩码
        ip_info.gw.addr = esp_ip4addr_aton("192.168.1.1");        // 设置默认网关
        // 应用配置的IP信息到网络接口
        ESP_ERROR_CHECK(esp_netif_set_ip_info(wifi_netif, &ip_info));

esp_bridge_wifi_set_config(WIFI_IF_STA, &wifi_sta_cfg);
esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_ap_cfg);

tswen, 2024-05-28:
不好意思，上面说错了，可以使用 iot_bridge 组件里面提供的接口 esp_bridge_netif_set_ip_info