espressif/esp-mesh-lite #88: build error

Code-Xiaofeng, 2024-05-17:
...NOTICE: Skipping optional dependency: espressif/esp_modem_usb_dte
......CMake Error at F:/Programs/Espressif/447/frameworks/esp-idf-v4.4.7/tools/cmake/build.cmake:463 (message):
  ERROR: Because no versions of mesh_lite match >=0.0.0,<0.10.3 || >0.10.3

   and mesh_lite (0.10.3) depends on espressif/iot_bridge (0.*), mesh_lite (>=0.0.0) requires espressif/iot_bridge (0.*).

  So, because no versions of espressif/iot_bridge match 0.*

   and project depends on mesh_lite (*), version solving failed.

Call Stack (most recent call first):
  F:/Programs/Espressif/447/frameworks/esp-idf-v4.4.7/tools/cmake/project.cmake:476 (idf_build_process)
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "E:/my_code/Origincell/esp32_project/esp-mesh-lite/examples/mesh_local_control/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

Has anyone encountered this problem？

Aliothxt, 2024-05-17:
参考了#73 后面的回复, 试试将`examples/mesh_local_control/main/idf_component.yml`下的第6行`override_path: "../../../components/mesh_lite"`注释掉试一下,我的IDF版本是5.1.2的注释掉就可以编译了

Code-Xiaofeng, 2024-05-20:
非常感谢，屏蔽掉就可以了

Code-Xiaofeng, 2024-05-20:
> 参考了#73 后面的回复, 试试将`examples/mesh_local_control/main/idf_component.yml`下的第6行`override_path: "../../../components/mesh_lite"`注释掉试一下,我的IDF版本是5.1.2的注释掉就可以编译了

非常感谢，屏蔽掉就可以了

closed 2024-05-27 by xcguang