espressif/esp-mesh-lite #89: How to using smart config in mesh lite ?

rikkapro0128, 2024-05-22:
i used smart config when config esp_mesh_lite and start smartconfig ? 
```cpp
ESP_ERROR_CHECK(esp_netif_init());
  ESP_ERROR_CHECK(esp_event_loop_create_default());

  esp_bridge_create_all_netif();

  wifi_config_t wifi_cfg;
  memset(&wifi_cfg, 0x0, sizeof(wifi_config_t));
  size_t softap_ssid_len = sizeof(wifi_cfg.ap.ssid);
  if (esp_mesh_lite_get_softap_ssid_from_nvs((char *)wifi_cfg.ap.ssid, &softap_ssid_len) != ESP_OK)
  {
    snprintf((char *)wifi_cfg.ap.ssid, sizeof(wifi_cfg.ap.ssid), "%s", CONFIG_BRIDGE_SOFTAP_SSID);
  }
  size_t softap_psw_len = sizeof(wifi_cfg.ap.password);
  if (esp_mesh_lite_get_softap_psw_from_nvs((char *)wifi_cfg.ap.password, &softap_psw_len) != ESP_OK)
  {
    strlcpy((char *)wifi_cfg.ap.password, CONFIG_BRIDGE_SOFTAP_PASSWORD, sizeof(wifi_cfg.ap.password));
  }
  esp_bridge_wifi_set_config(WIFI_IF_AP, &wifi_cfg);

  esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
  esp_mesh_lite_init(&mesh_lite_config);

  esp_mesh_lite_start();

  /* Register our event handler for Wi-Fi, IP and Provisioning related events */
  ESP_ERROR_CHECK(esp_event_handler_register(WIFI_PROV_EVENT, ESP_EVENT_ANY_ID, &Mesh::event_handler, self));
  ESP_ERROR_CHECK(esp_event_handler_register(PROTOCOMM_SECURITY_SESSION_EVENT, ESP_EVENT_ANY_ID, &Mesh::event_handler, self));
  ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &Mesh::event_handler, self));
  ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &Mesh::event_handler, self));
  ESP_ERROR_CHECK(esp_event_handler_register(SC_EVENT, ESP_EVENT_ANY_ID, &Mesh::event_handler, self));
  ESP_ERROR_CHECK(esp_event_handler_register(ESP_HTTPS_OTA_EVENT, ESP_EVENT_ANY_ID, &Mesh::event_handler, self));
```

and then i run func contain code 
```cpp
  esp_smartconfig_stop();
  wifi_config_t config{};
  esp_wifi_disconnect();
  esp_wifi_set_config(WIFI_IF_STA, &config);
  // esp_wifi_set_mode(WIFI_MODE_STA);
  ESP_ERROR_CHECK(esp_smartconfig_set_type(SC_TYPE_ESPTOUCH));
  smartconfig_start_config_t cfg = SMARTCONFIG_START_CONFIG_DEFAULT();
  ESP_ERROR_CHECK(esp_smartconfig_start(&cfg));
```

smartconfig just config ssid & pass wifi on first time (mesh_lite not yet connect with router wifi), but if i re-run module smartconfig i will cant recieve info ssid & pass when app send

can help me resolve this problem (code examle, ...) =((

rikkapro0128, 2024-05-22:
one more thing, when i run smart config:
- Sometime smart config will log 
``
smartconfig: smartconfig errno -1@sc_sniffer.c 155
I (1071425) smartconfig: smartconfig errno 258@sc_sniffer.c 155
W (1071575) wifi:STA is scanning or connecting, or AP has connected with external STAs, cannot set channel

I (1071575) smartconfig: smartconfig errno -1@sc_sniffer.c 178
W (1071735) wifi:STA is scanning or connecting, or AP has connected with external STAs, cannot set channel
``
so this will be the time when send info wifi in app, i will recieve in esp32, but something cant recieve info wifi =))

- other case when i start smartconfig, not thing log in terminal just log like
``
I (1437375) smartconfig: SC version: V3.0.1
I (1439555) smartconfig: scan status 1
I (1439605) smartconfig: scan status 4
I (1439655) smartconfig: scan status 1
I (1439705) smartconfig: scan status 4
``

i guess module smartconfig cant scan AP around so make this not working
Sumary, i want sure module smartconfig cant run and receive info wifi when app thanks all for help me =((

rikkapro0128, 2024-05-22:
When touch receive event TYPE_LONG i will be start smartconfig, but i cant receive config wifi
[image]

tswen, 2024-06-07:
If you disconnect the Wiif before reconfiguring the network, it will trigger the reconnection mechanism within the mesh. The reconnection within the mesh will involve scanning, which may affect the scanning results in SmartConfig.

rikkapro0128, 2024-06-07:
> If you disconnect the Wiif before reconfiguring the network, it will trigger the reconnection mechanism within the mesh. The reconnection within the mesh will involve scanning, which may affect the scanning results in SmartConfig.

ok, i see, thanks <3

closed 2024-06-07 by rikkapro0128