espressif/esp-mesh-lite #90: What is Maximum No of layer or Hops can be achieved?

mightChamp, 2024-05-23:
I have 1000 nodes which are in a line(Street light) 50-100 meter apart. So, I want to make sure that in mesh I can connect to last one node. Data Speed is not issue, if node get data in 30-60 seconds, it will be ok. Will this system Use ESP32-C6 as Gateway, as well as for Nodes.

tswen, 2024-05-27:
Currently, mesh lite supports up to 15 layers. In addition, the number of nodes deployed in a mesh network is recommended to be controlled within 100.

mightChamp, 2024-05-29:
Will ESP-MDF Supports my system(1000 nodes which are in a line(Street light) 50-100 meter apart)?

xcguang, 2024-05-29:
Will you directly connect to the last node for communication? Is the data both downlink and uplink, or only downlink?

mightChamp, 2024-05-29:
@xcguang  All node will send data to gateway, So last 1000th Node(50Km apart from gateway) will send data to gateway, latency is not issue, if took time, it will be ok, but data must be reached to gateway. and both uplink and downlink should work with 1000th node from gateway.

mightChamp, 2024-05-29:
@grodansparadis  ESPNOW can support mesh?

mightChamp, 2024-05-29:
My System is as follows:

- AC Power Supply 24 Hours running
- Each Node is 50 Meter Apart on street light (last 1000th node is at 25km from gateway).
- Node will never move it is static
- Data Frame will be Max 50 Bytes
- Each node send Data to next 5 nodes, for some trigger.
- Each node send an heartbeat data to node every 15 minute.
- Gateway can send downlink for configurations any time.

So, Based on above could you suggest best Technology to use :
Zigbee/ESP-MDF/ESP-Mesh-Lite/PainlessMesh/ESP-NOW/Openthread

grodansparadis, 2024-05-29:
ESP-NOW would solve that scenario easy and at a low cost. The others to of course but at a higher complexity.  I'm a fan of KISS (Keep  It Simple Stupid) .   :)    I have never tested PainlessMesh but OpenThread is nice.

xcguang, 2024-05-30:
Mesh-Lite supports downlink data communicate on esp-now, which api is `esp_mesh_lite_espnow_send`.

mightChamp, 2024-06-06:
Hello to all,

In my all research, I reached at conclusion that 1000 node is not feasible in single mesh. So, I am changing my implementation to minimal requirement of 100 nodes in a single mesh of street lights, which is as follow:  

- Node has AC Power Supply 24 Hours running
- Each Node is 50 Meter Apart on street lights (last 100th node is at 5km from gateway).
- Node will never move it is static
- Data Frame will be Max 50 Bytes
- Each node send Data to next 5 nodes, for some trigger.
- Each node send an heartbeat data to node every 15 minute.
- Gateway can send downlink for configurations any time.

So, Based on above could you suggest best Technology to use :
Zigbee/ESP-MDF/ESP-Mesh-Lite/PainlessMesh/ESP-NOW/Openthread

xcguang, 2024-06-07:
hello @mightChamp, could you give me a mail, I'd like to know more about your project.

mightChamp, 2024-06-12:
@xcguang Above is only my requirement, Could you provide which technology would be best Zigbee/ESP-MDF/ESP-Mesh-Lite/PainlessMesh/ESP-NOW/Openthread/BLE-Mesh in Street light line topology mesh communication.

xcguang, 2024-06-13:
Maybe esp_now still is better.

mightChamp, 2024-06-13:
esp-now is not an official mesh support, So I am afraid to use it. So I am ready to compromise on no of nodes and communication delay in single mesh. So what you suggest best.

xcguang, 2024-06-13:
If the layer is less than 15, I will say ESP-Mesh-Lite should meet your requirements as discussed in https://github.com/espressif/esp-mesh-lite/issues/96, however, we'd like to evaluate it.

To be honest, the others mesh technology(Zigbee/ESP-MDF/PainlessMesh/Openthread/BLE-Mesh) may work well in your scenario, but I did not test them, so I can not guarantee it.

mightChamp, 2024-06-13:
@xcguang thanks for your response.  Last thing I want to know that, Can We use two root nodes in mesh?

xcguang, 2024-06-13:
Yes, it supports multiple root nodes

closed 2024-06-15 by mightChamp