espressif/esp-mesh-lite #91: send msg to node target with mac (dont care leaf, parent or root node)

rikkapro0128, 2024-06-06:


tswen, 2024-06-06:
You can use esp-now https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/include/esp_mesh_lite_espnow.h

rikkapro0128, 2024-06-06:
> You can use esp-now https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/include/esp_mesh_lite_espnow.h

- esp now it is peer to peer, I just want to send data in the mesh network with a clear address similar to esp_wifi_mesh
- in the case where I want to turn on all devices in the mesh network when a certain device is activated, there will be a node that is too far from that node and esp_now will not respond, for example from floor 1 to floor 5 such as

rikkapro0128, 2024-06-06:
> You can use esp-now https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/include/esp_mesh_lite_espnow.h

I hope there is a way to solve this problem, thank you

tswen, 2024-06-07:
Then either use espnow broadcast, the payload contains the device mac that needs to respond, the device responds to the corresponding mac after receiving it, and then all device application layers need to do a layer of forwarding processing.
Then use the communication interface provided by esp-mesh-lite, esp_mesh_lite_send_broadcast_msg_to_child, the root node sends it to the next level, the next level receives it and then broadcasts it to the next level, the payload also contains the device mac that needs to respond, the device responds to the corresponding mac after receiving it.

rikkapro0128, 2024-06-07:
> Then either use espnow broadcast, the payload contains the device mac that needs to respond, the device responds to the corresponding mac after receiving it, and then all device application layers [...]

- oke i see, thanks so mush, i hope esp_mesh_lite will have be more func helpful like esp_mesh_lite_send_msg(uint8_t *mac), if this occur, it will very excellent =))

tswen, 2024-06-07:
If using UDP or TCP, it is difficult to achieve point-to-point communication in a tree topology.

To achieve point-to-point communication, you can only use ESP-NOW communication, which is not based on Wi-Fi connections. However, to avoid packet loss over long distances and require other devices to relay, you can only use ESP-NOW broadcast, and then filter through specific fields or MAC addresses.
The Rainmaker example does this, as shown in https://github.com/espressif/esp-mesh-lite/blob/master/examples/rainmaker/led_light/components/group_control/app_espnow.c

rikkapro0128, 2024-06-08:
> If using UDP or TCP, it is difficult to achieve point-to-point communication in a tree topology.
>
> To achieve point-to-point communication, you can only use ESP-NOW communication, which is not based [...]

oke thank u <3

closed 2024-06-08 by rikkapro0128