espressif/esp-mesh-lite #93: ESP32-mesh-lite:MQTT报错

LYL14, 2024-06-07:
1.
(2223341) transport_base: tcp_read error, errno=Socket is not connected
(2223341) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read() error: errno=128
(2223351) local_control: error
(2223351) mqtt_client: mqtt_process_receive: mqtt_message_receive() returned -1
(2223361) local_control: disconnected
2.
(2375361) transport_base: Poll timeout or error, errno=Connection already in progress, fd=58, timeout_ms=10000
(2375361) mqtt_client: Writing didn't complete in specified timeout: errno=119
(2375361) mqtt_client: Error to resend data 
(2375371) local_control: disconnected
3.
(3166061) mqtt_client: Writing didn't complete in specified timeout: errno=119
(3166061) mqtt_client: Error to resend data 
4.
(1274291) transport_base: Poll timeout or error, errno=Connection already in progress, fd=-1, timeout_ms=10000
(1274291) mqtt_client: Writing didn't complete in specified timeout: errno=11
(1274291) mqtt_client: Error sending ping
(1274301) mqtt_client: Can't send ping, disconnected
(1274311) mqtt_client: Publishing skipped: client is not connected
 (1274321) mqtt_client: Publishing skipped: client is not connected
5.
(1483441) esp-tls: [sock=58] select() timeout
(1483441) transport_base: Failed to open a new connection: 32774
(1483441) mqtt_client: Error transport connect
(1483451) mqtt_client: Publishing skipped: client is not connected
使用ESP32C3和S3系列芯片共60个，mesh-lite组网，然后通过MQTT收发数据，串口打印错误日志
设备mesh组网中而且获取到IP,但是通讯失败。

rikkapro0128, 2024-06-07:
> (2223341) transport_base: tcp_read error, errno=Socket is not connected (2223341) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read() error: errno=128 (2223351) local_control: error [...]

sometime this problem also happen with me, hope this problem fix =((

LYL14, 2024-06-07:
将设备减少至30-40，设备可以重新组网，并进行MQTT双向通信

LYL14, 2024-06-12:
我设置了wifi_cfg.ap.beacon_interval = 200;，然后设备放置间距1m~2m。60个设备依然出现网络问题。
 (3761745) transport_base: Poll timeout or error, errno=Connection already in progress, fd=58, timeout_ms=10000
 (3761745) mqtt_client: Writing didn't complete in specified timeout: errno=119
 (3761745) mqtt_client: Error to resend data

 (3771785) transport_base: Poll timeout or error, errno=Connection already in progress, fd=-1, timeout_ms=10000
 (3771785) mqtt_client: Writing didn't complete in specified timeout: errno=119
(3771785) mqtt_client: Error sending ping
 (3771795) mqtt_client: Can't send ping, disconnected
 (3771805) mqtt_client: Publishing skipped: client is not connected
(3771815) mqtt_client: Publishing skipped: client is not connected

(3776815) mqtt_client: Publishing skipped: client is not connected

client IP:50636992 MASK:16777215 GW:17082560
local_control: System information, channel: 2, layer: 2, self mac: ec:da:3b:cc:be:40, parent bssid: 3c:84:27:27:5a:f9, parent r
ssi: -48, free heap: 164716[0m

 (3818315) transport_base: Poll timeout or error, errno=Bad file number, fd=58, timeout_ms=10000
 (3818315) mqtt_client: Writing didn't complete in specified timeout: errno=9
 (3818315) local_control: disconnected
(3818325) mqtt_client: Publishing skipped: client is not connected

(3823335) mqtt_client: Publishing skipped: client is not connected
 (3828335) mqtt_client: Publishing skipped: client is not connected
[image]

LYL14, 2024-06-19:
昨天进行批量断线重连，设备网络掉线
1.
 IP:33925312 MASK:16777215 GW:17148096
(271555) local_control: System information, channel: 2, layer: 2, self mac: 3c:84:27:27:53:58, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 245404[0m
(271565) local_control: child node number: 0[0m
 (271565) local_control: Child mac: 3c:84:27:27:55:94[0m
2.
[19:15:39.129]收←◆client IP:0 MASK:0 GW:0
  (16111640) local_control: System information, channel: 2, layer: 3, self mac: 3c:84:27:27:57:18, parent bssid: 3c:84:27:27:5e:d1, parent rssi: -90, free heap: 215508[0m
  (16111640) local_control: child node number: 0[0m
  (16111650) local_control: Child mac: 3c:84:27:27:55:78[0m
  (16111650) local_control: Child mac: 3c:84:27:27:66:c4[0m

[19:15:48.911]收←◆[0;31mE (16121420) esp-tls: [sock=59] connect() error: Host is unreachable[0m
[0;31mE (16121420) transport_base: Failed to open a new connection: 32772[0m
[0;31mE (16121420) mqtt_client: Error transport connect[0m
  (16121430) local_control: error[0m
  (16121430) local_control: disconnected[0m

[19:16:03.919]收←◆[0;31mE (16136430) esp-tls: [sock=59] connect() error: Host is unreachable[0m
[0;31mE (16136430) transport_base: Failed to open a new connection: 32772[0m
[0;31mE (16136430) mqtt_client: Error transport connect[0m
  (16136440) local_control: error[0m
  (16136440) local_control: disconnected[0m

[19:16:09.129]收←◆client IP:0 MASK:0 GW:0
  (16141640) local_control: System information, channel: 2, layer: 3, self mac: 3c:84:27:27:57:18, parent bssid: 3c:84:27:27:5e:d1, parent rssi: -91, free heap: 216180[0m
  (16141640) local_control: child node number: 0[0m
  (16141650) local_control: Child mac: 3c:84:27:27:55:78[0m
  (16141650) local_control: Child mac: 3c:84:27:27:66:c4[0m

xcguang, 2024-06-19:
信号强度太弱了，可以用 esp_mesh_lite_set_rssi_threshold 设置下信号强度，比如 -75，这样连接时，会跳过 低于 -75 的 AP

LYL14, 2024-06-19:
这个函数的传参您理解吗，我有些没看懂 esp_mesh_lite_rssi_threshold_list_t 这个变量是设置在不同层的RSSI值吗。

xcguang, 2024-06-19:
```
static const esp_mesh_lite_rssi_threshold_list_t rssi_level[] = {
      {0,  -35, 20},
      {-35,  -55, 15},
      {-55,  -75, 10}
  };
 
esp_err_t result = esp_mesh_lite_set_rssi_threshold(rssi_level, sizeof(rssi_level) / sizeof(esp_mesh_lite_rssi_threshold_list_t), -75);
if (result == ESP_OK) {
     // RSSI threshold list set successfully
} else {
     // Failed to set the RSSI threshold list
}
```
这样试试呢

closed 2024-06-20 by LYL14