espressif/esp-mesh-lite #96: Can we increase Max No of layer of ESP-MESH-LITE from 15 to 20.

mightChamp, 2024-06-12:
As Mentioned in documentation, there is max 15 no of layer can be achieved. 
[image]

But my requirement is of 20.  May be there is some latency issues, But it will be ok of us.

So Can I change this value, and How to do it?

xcguang, 2024-06-13:
Sorry, you can not change it, btw, how do you evaluate 20 layers?

dipakbhutada, 2024-06-13:
@xcguang  Thanks for your response. Since each node is 50 meters apart, with the possibility of 20 layers, I can achieve 30-40 nodes in a single mesh. This allows me to cover a distance of 2 kilometers.

mightChamp, 2024-06-13:
@xcguang  Thanks for your response. Since each node is 50 meters apart, with the possibility of 20 layers, I can achieve 30-40 nodes in a single mesh. This allows me to cover a distance of 2 kilometers.

xcguang, 2024-06-13:
Now the limit is in software, for this change, and it needs some time to evaluate it.

closed 2024-06-15 by mightChamp