espressif/esp-mesh-lite #99: ESP-MESH-LAN-OTA失败

LYL14, 2024-06-19:
根节点和子节点共用下面代码，出现mesh-lan-ota和http-ota同时进行，mesh-lan-ota进行100%后失败。
.I (108395) local_control: OTA Progress: 26.79%
I (108395) local_control: OTA Progress: 26.89%
I (108435) local_control: OTA Progress: 26.99%
I (108505) Mesh-Lite: LAN OTA Percentage: 100%
I (108505) esp_image: segment 0: paddr=00210020 vaddr=3c0c0020 size=2c180h (180608) map
I (108535) esp_image: segment 1: paddr=0023c1a8 vaddr=3fc98a00 size=03d5ch ( 15708) 
I (108535) esp_image: segment 2: paddr=0023ff0c vaddr=40374000 size=0010ch (   268)
I (108535) esp_image: segment 3: paddr=00240020 vaddr=42000020 size=b3ac4h (735940) map
I (108645) esp_image: segment 4: paddr=002f3aec vaddr=4037410c size=1488ch ( 84108) 
E (108655) esp_image: Checksum failed. Calculated 0x3b read 0x3
I (108655) esp_image: segment 0: paddr=00210020 vaddr=3c0c0020 size=2c180h (180608) map
I (108685) esp_image: segment 1: paddr=0023c1a8 vaddr=3fc98a00 size=03d5ch ( 15708) 
I (108685) esp_image: segment 2: paddr=0023ff0c vaddr=40374000 size=0010ch (   268)
I (108685) esp_image: segment 3: paddr=00240020 vaddr=42000020 size=b3ac4h (735940) map
I (108795) esp_image: segment 4: paddr=002f3aec vaddr=4037410c size=1488ch ( 84108) 
E (108805) esp_image: Checksum failed. Calculated 0x3b read 0x3
E (108805) [ESP_Mesh_Lite_OTA]: esp_ota_set_boot_partition failed! err=0x1503 
I (108805) [ESP_Mesh_Lite_OTA]: OTA End
E (108815) Mesh-Lite: LAN OTA Fail! Reason: 7

"""
    int now_version = get_current_version();
    esp_err_t ota_finish_err = ESP_OK;
    esp_http_client_config_t config = {
        .url = "http://192.168.1.77:8888/s3-mesh-ota.bin",
    };

    esp_https_ota_config_t ota_config = {
        .http_config = &config,
        .http_client_init_cb = _http_event_handler, // Register a callback to be invoked after esp_http_client is initialized
    };

    esp_https_ota_handle_t https_ota_handle = NULL;
    esp_err_t err = esp_https_ota_begin(&ota_config, &https_ota_handle);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "ESP HTTPS OTA Begin failed");
        vTaskDelete(NULL);
    }
    esp_app_desc_t new_app_desc;
    err = esp_https_ota_get_img_desc(https_ota_handle, &new_app_desc);
    printf("11111111esp_https_ota_handle_t===%s\n", new_app_desc.version);

    if (err != ESP_OK)
    {
        printf("esp_https_ota_read_img_desc failed\n");
        ESP_LOGE(TAG, "esp_https_ota_read_img_desc failed");
        // esp_https_ota_abort(https_ota_handle);
        vTaskDelete(NULL);
    }

    // Root node, start the OTA process
    if (esp_mesh_lite_get_level() > 1)
    {
        /* Processing flow of child nodes. */
        int filesize = esp_https_ota_get_image_size(https_ota_handle);
        printf("esp_https_ota_get_image_size=%d\n", filesize);
        esp_mesh_lite_file_transmit_config_t transmit_config = {
            .type = ESP_MESH_LITE_OTA_TRANSMIT_FIRMWARE,
            .size = filesize,
            .extern_url_ota_cb = NULL,
        };
        esp_mesh_lite_transmit_file_start(&transmit_config);
    }
    else
    {
        esp_mesh_lite_ota_notify_child_node_pause();
    }

    if (esp_mesh_lite_get_level() >= 1)
    {
        while (1)
        {
            if (esp_mesh_lite_wait_ota_allow() != ESP_OK)
            {
                err = ESP_FAIL;
                break;
            }
            err = esp_https_ota_perform(https_ota_handle);
            if (err != ESP_ERR_HTTPS_OTA_IN_PROGRESS)
            {
                printf("--err-%0x\n", err);
                break;
            }
            int image_len_read = esp_https_ota_get_image_len_read(https_ota_handle);
            int total_size = esp_https_ota_get_image_size(https_ota_handle);
            float progress = (float)image_len_read / total_size * 100;
            ESP_LOGI(TAG, "OTA Progress: %.2f%%", progress);
            ESP_LOGD(TAG, "Image bytes read: %d", esp_https_ota_get_image_len_read(https_ota_handle));
        }
        if (err != ESP_OK)
        {
            printf(" esp_https_ota_perfo=%0x\n", err);
            esp_https_ota_abort(https_ota_handle);
            // vTaskDelete(NULL);
        }
    }
    printf("otaend/n");

    ota_finish_err = esp_https_ota_finish(https_ota_handle);
    if (ota_finish_err == ESP_OK)
    {
        ESP_LOGI(TAG, "ESP_HTTPS_OTA upgrade successful. Rebooting ...");
        vTaskDelay(5000 / portTICK_PERIOD_MS);
        esp_restart();
    }
    else
    {
        if (ota_finish_err == ESP_ERR_OTA_VALIDATE_FAILED)
        {
            ESP_LOGE(TAG, "Image validation failed, image is corrupted");
        }
        ESP_LOGE(TAG, "ESP_HTTPS_OTA upgrade failed 0x%x", ota_finish_err);
        // vTaskDelete(NULL);
    }
"""

tswen, 2024-06-20:
正确的流程应该是：
1. 设备收到 OTA 指令，开始准备 OTA
2. 判断设备当前处于的层级，如果是根节点，进行 http ota；如果是子节点，进行 esp_mesh_lite_transmit_file_start

tswen, 2024-06-20:
参考文档：https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota