espressif/esp-mesh-lite #101: mesh lite use http client call api error

yel-best, 2024-06-20:
use esp-idf v5.1
use http client

When using http client to call API, there is a certain probability that the following error will appear. Although it is not 100%, it will appear occasionally. And it is found that if the node is at the front of the hierarchy, it is not easy to appear, but when it is at the back of the hierarchy, such as 5th and 6th layers, the problem will occur with a high probability, which makes me unable to understand

```
E (297708) esp-aes: Failed to allocate memory
E (297708) esp-tls-mbedtls: write error :-0x0001:
E (297708) transport_base: esp_tls_conn_write error, errno=Connection already in progress
E (297716) HTTP_CLIENT: Error write request
E (297722) app_lvgl_http: Failed to open connection to https://xxxxxxxx/api/xxxx/xxxxxxx?paging=true&page=1&limit=20
E (297735) app_lvgl_http: Failed to terminal api reponse.
```

tswen, 2024-06-23:
Hello, when the problem occurs during the test, how many devices are there in a Mesh? Are there any test conditions where some devices are frequently powered on and off?

yel-best, 2024-06-26:
> Hello, when the problem occurs during the test, how many devices are there in a Mesh? Are there any test conditions where some devices are frequently powered on and off?

I have 30 **ESP32S3 N16R8** in my mesh, all of which need to connect to MQTT and HTTP. I tested that the problem occurred on a level 2 node. The level 1 nodes connected to the level 2 nodes are also next to me. The level 1 nodes did not restart and disconnect. However, after the level 2 node connected to the level 1 node, I called the HTTP API and this problem occurred. It kept prompting an exception. Every API request was abnormal. Only after I restarted, there was a probability that it could be restored. I really don't understand the reason.

Because I use the N16R8 version, I wonder if I can migrate the HTTP or MESH TASK to the external ARM, so that there will be more space, will this problem be solved? But I don't know how to migrate to the external ARM

tswen, 2024-06-27:
https://docs.espressif.com/projects/esp-idf/zh_CN/v5.1.2/esp32/api-reference/kconfig.html?highlight=psram#esp-psram

yel-best, 2024-08-02:
Hi, pls help me,

**1. When I use http to request the api, there will be such a prompt. In my project, I use components such as usb & modbus & gpio at the same time, and sometimes the http request is slow to send out. The url address is https. Is it because of the slow https resolution that causes the problem?**

**2. Is there any configuration that can increase the size of mesh lite node forwarding, which can help improve the speed of https sub-node requests? This problem has troubled me for a long time. I have no choice but to ask for help. I hope to get help.**

Thank you so much!!!!

```
I (4941) esp_bee_claim: bee_initialize_api to: things/DCDA0C1894FC/initialize
I (4949) app_bee: Bee Claim Started.
E (5078) bignum: Failed to allocate RSA interrupt 261

abort() was called at PC 0x4209df08 on core 0
0x4209df08: esp_mpi_isr_initialise at E:/MyProject/IoT/esp/esp-idf/components/mbedtls/port/esp_bignum.c:97
 (inlined by) esp_mpi_exp_mod at E:/MyProject/IoT/esp/esp-idf/components/mbedtls/port/esp_bignum.c:410
 (inlined by) mbedtls_mpi_exp_mod at E:/MyProject/IoT/esp/esp-idf/components/mbedtls/port/esp_bignum.c:469

Backtrace: 0x40379c9a:0x3c1817e0 0x403892c1:0x3c181800 0x40390f2e:0x3c181820 0x4209df08:0x3c181890 0x4209a6e8:0x3c1818f0 0x4209b61a:0x3c181920 0x4209b68b:0x3c181950 0x42098c21:0x3c181980 0x42098855:0x3c1819b0 0x4208ede0:0x3c1819d0 0x4208d8ec:0x3c181a70 0x4208c1bb:0x3c181a90 0x4208c1fc:0x3c181ab0 0x4208525a:0x3c181ad0 0x42084d19:0x3c181af0 0x42084e4a:0x3c181b30 0x420867e8:0x3c181b70 0x420fb095:0x3c181ba0 0x420882ce:0x3c181bc0 0x42089040:0x3c181bf0 0x4200e9f5:0x3c181c20 0x4200efa9:0x3c181e30 0x4200d6e0:0x3c181e50 0x4200dd7f:0x3c181e80
0x40379c9a: panic_abort at E:/MyProject/IoT/esp/esp-idf/components/esp_system/panic.c:466

0x403892c1: esp_system_abort at E:/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
```

tswen, 2024-08-09:
1. You can pay attention to whether the remaining memory is insufficient when the problem occurs.
2. In theory, increasing the buffer related to Wi-Fi and Lwip can achieve higher throughput, but the side effect is that it will consume more memory. If there is not much memory left, it is not recommended to increase the buffer of Wi-Fi and Lwip related configurations.