espressif/esp-mesh-lite #104: mseh dhcps ip assignment problem

yel-best, 2024-06-26:
Hi ,

Because my device is running in a group, and the group's router uses 192.168.xx.xx DHCP, and has some restrictions on this IP segment. When we use mesh, the mesh dhcp IPs obtained by the nodes that make up the network seem to start from 192.168.1.1, and different levels will have different, gradually, for example: 192.168.1.1, 192.168.1.2. Can I define this network segment? For example, if I use 10.22.xx.xx, I don't know if some of the slow intranet access problems I encounter are related to the DHCP allocation restrictions? Or is the internal network of the mesh actually one IP for the route, and this IP is allocated by the router?

I have read some instructions and tried to check the router management interface. In fact, only the mesh root node will be displayed in the list of connected terminals on the router.

tswen, 2024-06-27:
For Mesh-Lite devices, the IP of each child node device is assigned by its parent node. For a Mesh-Lite network, only the root node is visible to the router.

You can set a specific network segment for softap through esp_bridge_netif_set_ip_info API.

tswen, 2024-06-27:
For station netif, it is to set the static IP, but for softap netif, it is to set the softap network segment.

tswen, 2025-07-08:
1. Static IP usage is not recommended in mesh networks. If absolutely necessary, ensure proper IP conflict management is implemented. If dhcps_stop is called, the parent node's softAP will not assign IP addresses to child nodes.
2. The esp_wifi_set_inactive_time interface is unrelated to the described issue. If the problem persists, please provide logs for further investigation.

tswen, 2025-09-19:
Can you capture Wi-Fi air packets? If possible, could you set the softAP to OPEN mode and check if there are DHCP message exchanges in the air packets? Have you tried the mesh_local_control example? Does the same issue of failing to obtain an IP address occur there as well?