espressif/esp-mesh-lite #106: Running webserver on each node in mesh lite network.

Rbhagwat62, 2024-07-18:
Hello,

I am currently working on one of the POC where I need to connect existing esp32-c6 based device in mesh network. Existing device runs webserver for accessing device configuration and other feature through webpage. I need to keep same functionality of webserver and want to create mesh lite network to increase Wi-Fi coverage through mesh network.

I started exploring mesh lite example of mesh_local_control on IDF v 5.2.1 for esp32 c6. I am able to create mesh network using 3 devices. Root node gets IP assigned by router and other node gets IP assigned by root DHCP server. All nodes in the network are able to connect tcp server running on my laptop connected to router network. Every thing looks ok till here.

When I send ping request to root node from Laptop connected to router network, I get the response. But when I ping to other node IP which are on different subnet I didn't get any response. Like wise I added HTTP server code instead of tcp write task. When I open root node IP on browser I am able see http page with root MAC but when I tried to open other node's IP it didn't.

When I explore more about sub network, we might need to do IP forwarding in root node and router device. In menuconfig CONFIG_LWIP_IP_FORWARD flag under LWIP is set. Please let me know if this is feasible to access webpage from any node in the network with node IP or is there any other way to achieve the same.

Please find the modified webserver code in mesh local control example here [mesh_local_control.zip](https://github.com/user-attachments/files/16278723/mesh_local_control.zip)
Also find test setup and result here [ESP_c6_mesh_lite_local_control.pptx](https://github.com/user-attachments/files/16278830/ESP_c6_mesh_lite_local_control.pptx)

Thanks.

Rbhagwat62, 2024-07-18:
@tswen,
thanks for quick response.

"_You can try binding the web server to the SoftAP netif. Connect a device to the corresponding device's SoftAP, and then access the gateway IP to access the web server of the corresponding device._"

This you are telling in context of WiFi mesh lite or simple AP and station connection between two esp. Can you elaborate more.

Can you refer any link or sample code for binding the web server to the SoftAP netif.

Rbhagwat62, 2024-07-25:
I tested on no router example of mesh lite with three devices. Made one device as root node and other as non root child node. I also added mdns and simple http server code in each node. 

When created two level network level 1 root node and level 2 two child node and connected laptop to root node softAP, I was able to open webpage of each node with their respective mdns host name. 

When created three level network, level 1 root, level2 child1 and level3 child2 and connected laptop with root node softAP, I was able to open webpage from root and child1 only with their respective mdns hostname. But the not able to access child 2 webpage. When connected Laptop to child 1 softAP able to access child1 and child 2 webpage but not the root node.

Is their any way to access any node's webpage through mdns or IP by connecting laptop to root node soft AP.

tswen, 2024-07-25:
It is difficult to do. You can only access the current node and other nodes under the same parent node. Cannot access different branches or cross-level nodes.

Rbhagwat62, 2024-07-31:
@tswen : I can understand it would be difficult. Where I can start, if I still want to achieve accessing any device using IP?

My requirement is run BACnetIP device in mesh lite network. Each BACnetIP device runs UDP server. When UDP client from laptop connected to router broadcast discover request (who-Is BACnet packet) all devices in the mesh should able receive it and send back response.

It would be great if you can guide me to start with implementing IP routing at each level in mesh lite network such that we can access any device in the mesh lite. 

Thanks

xcguang, 2024-07-31:
Maybe you can try:

After root node received the packet, it responds the laptop, at the same time it packs the UDP data included the laptop ip and port, then forwards the packed packet to the children. After the children received the packet, responds the the laptop directly, and forward the packet to their children continuously

Rbhagwat62, 2024-08-02:
@xcguang: Thanks for reply.

I checked the code to understand how I can pack the UDP data and forward to children. Each node is functioning as UDP server so I need to add UDP client code in each device such that it will broadcast the packets to the children. Also I need to broadcast the packet in the AP subnet to which children are connected. Is my understanding is correct or is there any other way to forward packet.

I have find in code where the data is received on root node from BACnet explore. I will add UDP client to forward the packets.

Thanks

Rbhagwat62, 2024-08-07:
@xcguang: 

I created mesh network with two node, one as root and other as child. Root is connected to the router. I run BACnet explorer on Laptop connected to router.

As per your suggestion at root node packed UDP data including Laptop IP, port and forwarded to root's AP subnet. 

At the child I am able to receive all messages, able to unpack laptop IP and port. When use UDP broadcast address of child IP network I am able receive response from child to root node, but when I use UDP broadcast IP of Laptop IP network no packets received at child and Laptop.

For better understanding, My laptop IP is 192.168.42.161, root station IP 192.168.42.104, root AP IP 192.168.64.10, Child station IP 192.168.64.11 (in range of Root AP subnetwork)
So at child when I use 192.168.64.255 as UDP broadcast IP, I am able receive child BACnet response at root node. But when I use 192.168.42.255 as UDP broadcast IP I am not able to receive BACnet response neither at root nor at laptop.

Thanks.

xcguang, 2024-08-07:
> So at child when I use 192.168.64.255 as UDP broadcast IP, I am able receive child BACnet response at root node. But when I use 192.168.42.255 as UDP broadcast IP I am not able to receive BACnet response neither at root nor at laptop.

This step, 192.168.64.11 send udp packet to 192.168.42.161, the port is the laptop's.

Rbhagwat62, 2024-08-07:
Port is same everywhere 47808.

> 
> This step, 192.168.64.11 send udp packet to 192.168.42.161, the port is the laptop's.
>
192.168.64.11 sends upd packet to 192.168.42.161 port 47808

Rbhagwat62, 2024-08-07:
When I gave same instance number (Unique number for BACnet device) to both node and send read object request after discovery, I am able to see two response of BACnet-APDU in Wireshark. I kept two distinct value for same property and I can see both values in response for corresponding device in Wireshark. This means unicast message from 192.168.64.11 (child node) to 192.168.42.161(Laptop) is working not the broadcast.

xcguang, 2024-08-08:
> When I gave same instance number (Unique number for BACnet device) to both node and send read object request after discovery, I am able to see two response of BACnet-APDU in Wireshark. I kept two [...]

The children node cannot respond the 192.168.42.161(Laptop) in broadcast mode. 
Now does this solution meet your requirement?

Rbhagwat62, 2024-08-08:
BACnet uses broadcast packets for discovery, since child is not able respond in broadcast mode I am able to discover child at my laptop. I not sure whether BACnet use broadcast other than discovery.

If you are sure if child node cannot respond in broadcast mode, I will check how I can send Broadcast messages received at root (from child) back to laptop.

Rbhagwat62, 2024-08-23:
In BACnet there is a concept of BBMD (BACnet Broadcast Management Devices) which simply used to forward the Broadcast messages between two subnetwork. With the BBMD configured on my root node I can forward broadcast messages received from my laptop to root nodes subnetwork and able to discover all devices under root node in my BACnet explorer running on Laptop.

All devices will get register with their respective station IP on the BACnet explorer. But when we send any read request to device apart from root node, BACnet explorer sends unicast request to device IP which doesn't get any response and timeout error occurs. Since laptop can't send any request directly to device IP after root node, no response receive.

Is there any way I can forward all messages send for particular device to root node and through root node to destination device.

xcguang, 2024-08-23:
> BACnet explorer sends unicast request to device IP which doesn't get any response and timeout error occurs
Is this request UDP or TCP? Is it possible creating a server(UDP or TCP) on BACnet, and the node devices connects this server after receiving discovery packet?

Rbhagwat62, 2024-08-23:
It is on UDP. The BACnet explorer (Laptop) it self work as BACnet device. Each device in BACnet works as UDP server. After discovery any device can send unicast message directly to other device IP using UDP.

In our case of mesh all device in the network can send response directly to BACnet explorer (Laptop) but to get request from Laptop to device we are facing issue.

Rbhagwat62, 2024-08-26:
Is it possible I will add one ESP device in my router network which will sniff all UDP packets, if packet is send from my Laptop's IP device will forward all the packets to root node.

Rbhagwat62, 2024-09-24:
In the BACnet protocol, there is a option for BACnet router which works as routing device for two different IP network. I found demo code for the router in BACnet stack, ported for mesh lite. After some correction in routing packets as per BACnet standard I am able discover BACnet device through multilayer mesh network.

Thanks for all your support.

closed 2024-09-24 by Rbhagwat62