espressif/esp-mesh-lite #111: layer 2 using with mqtt error

rikkapro0128, 2024-08-08:
Even though layer 1 has a network connection, layer 2 is very susceptible to mqtt socket errors. I still haven't determined the cause. Have you often had this problem when trying with mqtt?
[image]

rikkapro0128, 2024-08-08:
[image]

tswen, 2024-08-09:
How many devices are there in the current network?
In addition, what are the two configuration items CONFIG_BRIDGE_IP_NAPT_SIZE and CONFIG_BRIDGE_IP_NAPT_TIMEOUT_MS_TCP in your project?

rikkapro0128, 2024-08-09:
my current network is 10 device 
CONFIG_BRIDGE_IP_NAPT_SIZE
[image]
CONFIG_BRIDGE_IP_NAPT_TIMEOUT_MS_TCP
[image]

rikkapro0128, 2024-08-09:
> How many devices are there in the current network? In addition, what are the two configuration items CONFIG_BRIDGE_IP_NAPT_SIZE and CONFIG_BRIDGE_IP_NAPT_TIMEOUT_MS_TCP in your project?

Currently, I still see packets sent from the server, but every time the server sends data it reports a socket error. I don't know exactly what is going on and whether this error is due to the application layer or not.

rikkapro0128, 2024-08-10:
Currently I am on version 1.0.0 and am currently having an error about the 2nd layer mqtt and the device cannot create a mesh network if it is 1m or more away, however if I return to version 0.10.6 everything works. still fine in terms of mesh creation ability, however the mqtt problem of the 2nd layer about the socket is still there but it only occasionally encounters that situation

robocircuits, 2024-08-10:
I am facing a somewhat similar issue. I am using ESP_MESH_LITE with AWS IoT MQTT connection. 
The connection is very instable and I am getting delays in telemetry and commands.

rikkapro0128, 2024-08-10:
[image]
[image]

robocircuits, 2024-08-10:
[image]

tswen, 2024-08-12:
这个和 

> Currently I am on version 1.0.0 and am currently having an error about the 2nd layer mqtt and the device cannot create a mesh network if it is 1m or more away, however if I return to version 0.10.6 [...]

What does `the device cannot create a mesh network` mean? Is the Wi-Fi connection failed? Can you provide a complete log file from power on?

rikkapro0128, 2024-08-12:
> What does `the device cannot create a mesh network` mean? Is the Wi-Fi connection failed? Can you provide a complete log file from power on?

- this means that these devices are mostly layer 1, they have difficulty connecting to each other compared to version 0.10.6, as I see it
- With 0.10.6, I'm still having an mqtt socket error at layer 2. Can you please diagnose the error for me? Layer 1 still has normal mqtt operation. I suspect TCP packet forwarding is having problems. problem

tswen, 2024-08-13:
你有通过 esp_mesh_lite_set_networking_mode 设置 ESP_MESH_LITE_ROUTER 模式吗？
另外你说的 1.0.0 是通过 components register 拉下来的组件，还是 esp-mesh-lite 仓库 master 分支最新的组件，建议用 master 最新的。

rikkapro0128, 2024-08-13:
> Have you set the ESP_MESH_LITE_ROUTER mode via esp_mesh_lite_set_networking_mode? In addition, the 1.0.0 you mentioned is the component pulled down through the components register, or the latest component of the master branch of the esp-mesh-lite repository. It is recommended to use the latest master.

- No, I never called that function in my code
- now I'm only using version 0.10.6
- I noticed it only really moves from level 1 to level 2, 3, 4... when your internal code calls the function esp_mesh_lite_wifi_scan_start()
- Once it has connected to the router, it does not change the level (from level 1 to 2) or a long time later it calls the function esp_mesh_lite_wifi_scan_start in your internal code and it begins to change the level (time can take quite a bit longer than 10 minutes for example, I've seen that)
- Is there any way for me to be proactive in searching for mesh networks (I set the soft-ap network interfaces of the esp to hidden wifi_cfg.ap.ssid_hidden = 1;)
- Can you advise me to fix the mqtt problem or please test with mqtt again to see if the tcp socket handle has any problems?

closed 2024-08-20 by rikkapro0128