espressif/esp-mesh-lite #112: esp-mesh-lite espnow deleting peers after send

vinicius-smartme, 2024-08-09:
### Description
After calling the `esp_mesh_lite_espnow_send` to send an espnow message, the esp-mesh-lite removes the peer. There is no documentation or rationale for such action.

The peer deletion is here:
https://github.com/espressif/esp-mesh-lite/blob/e65961b4443cb97406e08073dd17e1417e862501/components/mesh_lite/src/esp_mesh_lite_espnow.c#L84

### Problem
Unjustified peer deletion causes overhead to the user, that needs to re-add the peer after every send.

### Proposed solution
Either remove the peer deletion or document the rationale/need.

tswen, 2024-08-12:
Thank you for pointing out this issue, we will improve it as soon as possible.

tswen, 2024-08-22:
Hello, the new mesh lite component has been updated, and the peer will no longer be deleted in `esp_mesh_lite_espnow_send`. At the same time, an API to delete the peer after sending is added: `esp_mesh_lite_espnow_send_and_del_peer`.