espressif/esp-mesh-lite #113: E (624658) ESP_Mesh_Lite_Comm: Disconnect reason : 8 E (624660) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 58

robocircuits, 2024-08-10:
I am using 5 ESP32-wroom modules to build a mesh. The devices are connecting to AWS IoT core using ttl/ssl cert. 

I am facing following issue
1. Every device is working on Layer 1
2. ESP mesh is not stabilizing as it should
[image]

I am getting these errors

> E (624658) ESP_Mesh_Lite_Comm: Disconnect reason : 8
> E (624660) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 58

It is highly unstable with MQTT

Please tell me if I am doing something wrong here.

tswen, 2024-08-12:
Which version of idf are you using? Which version of mesh-lite are you using? Can you provide a complete log from the time of power-on?

robocircuits, 2024-08-13:
MESH-LITE latest pull. ESP-IDF - 5.3

robocircuits, 2024-08-13:
This matter is solved ... may be issue with same client_id trying to connect to the mqtt broker. 
Now this is solved