espressif/esp-mesh-lite #114: MQTT with MESH-LITE Error :  mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): EOF

robocircuits, 2024-08-12:
I am connecting to AWS IoT Core using my esp32 in mesh network. I am getting the following errors.

[image]

The connection is not stable. Can anyone suggest me how can I make it more stable and better.

yel-best, 2024-08-12:
Hi, 

What version of` mesh lite `are you using?

How many esp32s are there? Can you try to connect MQTT in the local LAN to see if there will be any problems?

robocircuits, 2024-08-13:
This was issue due to multiple same client_id connecting to broker. i changed names and it worked fine after that.

yel-best, 2024-08-13:
> This was issue due to multiple same client_id connecting to broker. i changed names and it worked fine after that.

Glad to see your problem has been solved 😄

closed 2024-08-15 by xcguang