espressif/esp-mesh-lite #115: mesh_lite upgrade v1.0.0 wifi_prov_mgr compilation exception

yel-best, 2024-08-12:
Hi

`use mesh_lite v1.0.0 version`

After compiling, the following error will appear. What is the reason? After upgrading, an error is reported. I originally used v0.10.3 to upgrade to v1.0.0 version, through the configuration file:

`espressif/mesh_lite: ">=1.0.0"`

The latest mesh_lite version of `components.espressif.com` is only `v1.0.0`. When will `v1.0.1 `be updated?

```
E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/mesh-lite-terminal/managed_components/espressif__mesh_lite/src/wifi_prov/wifi_prov_mgr.c:203:17: error: 'strncpy' output may be truncated copying 32 bytes from a string of length 39 [-Werror=stringop-truncation]
  203 |                 strncpy((char*)wifi_prov_mgr_mesh_lite_cfg.ssid, softap_ssid, sizeof(wifi_prov_mgr_mesh_lite_cfg.ssid));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E:/MyProject/IoT/PlatformIO-ESP/esp-data-collect/mesh-lite-terminal/managed_components/espressif__mesh_lite/src/wifi_prov/wifi_prov_mgr.c:196:17: error: 'strncpy' output may be truncated copying 32 bytes from a string of length 39 [-Werror=stringop-truncation]
  196 |                 strncpy((char*)wifi_prov_mgr_mesh_lite_cfg.ssid, softap_ssid, sizeof(wifi_prov_mgr_mesh_lite_cfg.ssid));
```

tswen, 2024-08-12:
What version is your idf? After updating the mesh lite component, you can delete the build folder and recompile it.
Theoretically, this [patch](https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/src/wifi_prov/wifi_provisioning.patch) will be automatically applied to esp-idf during compilation.
You can check whether the idf contains the changes in this patch.
In addition, if you do not use the zero provisioning function of mesh_lite, you can only manually change this part
[image]

yel-best, 2024-08-12:
> Contributor

Hi 

Use `IDF Version 5.1.4`

I modified the configuration file version number from` 0.10.3 to 1.0.0` and recompiled it and the error occurred

If you modify the mesh_lite component without performing other operations and compile directly, an error message will appear

tswen, 2024-08-12:
What's the error? Can you provide a screenshot or log file?

tswen, 2024-08-14:
Did you add any compilation options to your project CMakeLists? I can compile it normally without any errors.