espressif/esp-mesh-lite #118: mesh lite OTA error

yel-best, 2024-08-19:
Hi

use ESP-IDF 5.1.3
use MESH-LITE 0.10.3

I don't understand why. My firmware suddenly had such an error today when performing OTA. The app_1 partition could not be overwritten normally, causing the OTA to fail. There were not many error prompts. Please help me find out what caused it.

```C
I (258771) app_bee_ota: Image bytes read: 1331200
I (259570) app_bee_ota: Image bytes read: 1382400
I (260411) app_bee_ota: Image bytes read: 1433600
I (261200) app_bee_ota: Image bytes read: 1484800
I (261834) app_bee_ota: Image bytes read: 1536000
I (262553) app_bee_ota: Image bytes read: 1587200
I (262698) app_bee_ota: Reporting IN_PROGRESS: Firmware Image download complete
W (262698) app_bee_ota_using_topics: app_bee_ota_using_topics::Firmware Image download complete
I (262705) app_bee_ota_using_topics: {"id":"9bd8dbcc-e1c0-4af8-be30-f259fea8370b","status":"IN_PROGRESS","message":"Firmware Image download complete"}
W (262719) app_bee_mqtt: Publishing to /esquel/xz7v8bn3my/otareply
I (262732) app_bee: bee mqtt msg published.
I (263227) wifi:Set ps type: 1, coexist: 0

I (263227) esp_https_ota: esp_https_ota_finish start...
I (263228) esp_image: segment 0: paddr=00220020 vaddr=3c100020 size=6cce0h (445664) map
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403783c3
```

yel-best, 2024-08-20:
pls help me. thanks

tswen, 2024-08-20:
It seems that the long OTA process triggered the watchdog and caused the system to reset. You can increase the task watchdog timeout or feed the watchdog in time.

yel-best, 2024-08-22:
> It seems that the long OTA process triggered the watchdog and caused the system to reset. You can increase the task watchdog timeout or feed the watchdog in time.

Yes，you are right. I want to know, I have upgraded to ESP-IDF 5.2.2. Which version of mesh-lite should I use to be the most stable? Currently, I see that the version released on `components.espressif.com` is only v1.0.0

I used `xTaskCreateStatic `to create the OTA Task I developed, which eventually led to the main reason for the failure of OTA. I put the OTA Task in `EXT_RAM_BSS_ATTR`, and I switched to using `xTaskCreate `to create the task, and it returned to normal.

tswen, 2024-08-26:
You can use version 1.0.0 or the latest master branch on GitHub. However, the updates on GitHub will be faster than the tagged version updates.