espressif/esp-mesh-lite #121: WiFi在WIFI_PROTOCOL_11B模式下，esp_mesh_lite_try_sending_msg 发送300字节不能发出，一直超时

guo652917087, 2024-08-29:
1. 请问 WiFi在WIFI_PROTOCOL_11B模式下，esp_mesh_lite_try_sending_msg 最大可以发送多少个字节？发包间隔多少？
2. WiFi在WIFI_PROTOCOL_LR模式下，esp_mesh_lite_try_sending_msg 最大可以发送多少个字节？发包间隔多少？

固定在11B模式下，只能ping路由器成功，发包一直发不出。
固定在11G模式下，就能发包成功。

guo652917087, 2024-08-29:
经过反复测试，在11B和LR模式下，mesh内部发送接口esp_mesh_lite_try_sending_msg 和 esp_mesh_lite_try_sending_raw_msg都不能发出数据。可以通过socket发送数据到外网。应该是个大的BUG！！

tswen, 2024-09-11:
11B 模式和 LR 模式下通信是没有问题的，你是怎么测试使用的，方便发一下你的测试代码吗？

guo652917087, 2024-09-13:
> 11B 模式和 LR 模式下通信是没有问题的，你是怎么测试使用的，方便发一下你的测试代码吗？

两个设备，一个设备固定根节点，一个子设备。

条件1

根节点设置：
esp_mesh_lite_set_allowed_level ( 1 );
esp_wifi_set_protocol ( WIFI_IF_AP,  WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR );

子节点设置
esp_mesh_lite_set_allowed_level ( 0 );
esp_mesh_lite_set_disallowed_level ( 1 );
esp_wifi_set_protocol ( WIFI_IF_AP,  WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR );
 esp_wifi_set_protocol ( WIFI_IF_STA,  WIFI_PROTOCOL_11B  | WIFI_PROTOCOL_LR );
或者
 esp_wifi_set_protocol ( WIFI_IF_STA,  WIFI_PROTOCOL_LR );
此种条件，
1）esp_mesh_lite_try_sending_msg 和 esp_mesh_lite_try_sending_raw_msg 不能发送数据，会发不出，
2）用ping命令，指定ping包长度-s 1200一直能ping到路由器上的IP。
3）移植iperf，测试到路由器上的IP速度在0.5Mbs，说明tcp socket ok.

-------------------------------------------------------------------------------------------------------------------------------
条件2

根节点设置：
esp_mesh_lite_set_allowed_level ( 1 );
esp_wifi_set_protocol ( WIFI_IF_AP,  WIFI_PROTOCOL_11B| WIFI_PROTOCOL_LR );

子节点设置
esp_mesh_lite_set_allowed_level ( 0 );
esp_mesh_lite_set_disallowed_level ( 1 );
esp_wifi_set_protocol ( WIFI_IF_AP,  WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR );
 esp_wifi_set_protocol ( WIFI_IF_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR);
此种条件，
esp_mesh_lite_try_sending_msg 和 esp_mesh_lite_try_sending_raw_msg 能发送300度个字节数据，大概率会发不出，用ping命令，指定ping包长度-s 1200一直能ping到路由器上的IP。
超过512字节就歇菜发不出了。
-------------------------------------------------------------------------------------------------------------------------------------------
条件3，限定esp_wifi_set_protocol ( WIFI_IF_AP,   WIFI_PROTOCOL_LR ); esp_mesh_lite_try_sending_msg 和 esp_mesh_lite_try_sending_raw_msg就什么都不能发。几个字节都不行。

guo652917087, 2024-09-13:
@tswen 
还发现：

用了ESP32S3做了SPI ETH网口，作为根节点的WAN口功能，发现在同一个路由器下，modbus tcp一直超时，经测试对比，发现两个根设备在同时接入在同一局域网，同网段后，ping包超时，mqtt也经常掉线，modbus tcp slave也经常超时，只要断开其中一个根设备的WAN口，另一台esp32s3根设备WAN口的网络通信就正常。

我猜测，应该是espressif__iot_bridge 软件上 的TCP/IP没处理好，互相反复转发，或者esp32s3内部路由表回环。

tswen, 2024-09-23:
https://github.com/espressif/esp-mesh-lite/blob/master/examples/rainmaker/led_light/components/app_bridge/app_mesh_lite_comm.c
可以参考下这个文件里面的使用方法。
是否有通过 esp_mesh_lite_msg_action_list_register 注册，定义的 type 是什么呢？可以将通信这部分代码发出来帮您看一下吗？

wifi mode 是不会影响这个通信的，我怀疑还是大概率是接口使用问题。

tswen, 2024-09-23:
> @tswen 还发现：
> 
> 用了ESP32S3做了SPI ETH网口，作为根节点的WAN口功能，发现在同一个路由器下，modbus tcp一直超时，经测试对比，发现两个根设备在同时接入在同一局域网，同网段后，ping包超时，mqtt也经常掉线，modbus tcp slave也经常超时，只要断开其中一个根设备的WAN口，另一台esp32s3根设备WAN口的网络通信就正常。
> 
> 我猜测，应该是espressif__iot_bridge 软件上 的TCP/IP没处理好，互相反复转发，或者esp32s3内部路由表回环。

你的意思是 根节点A（Station 连接路由器）和 根节点B（ETH 连接路由器）同时连接路由器时，两个设备的应用层通信会超时？断开 根节点A，根节点 B 就好了？

根节点 A 和根节点 B 从路由器那边拿到的 ip 分别是多少，路由器可以抓包吗？

guo652917087, 2024-09-30:
@tswen  提供一份简单的测试代码给你验证，针对11B和LR模式，不能发出数据问题。

/*
 * SPDX-FileCopyrightText: 2022-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <inttypes.h>
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"

#include "esp_wifi.h"
#include "nvs_flash.h"
#include <sys/socket.h>

#include "esp_mac.h"
#include "esp_bridge.h"
#include "esp_mesh_lite.h"

#define IS_ROOT 1

#define TEST_MESSGAE_LENGTH								1024 // 200 500

#define WIFI_MESH_ACTION_JSON_DOWN 						"down"
#define WIFI_MESH_ACTION_JSON_DOWN_ACK 					"down_ack"
#define WIFI_MESH_ACTION_JSON_UP 						"up"
#define WIFI_MESH_ACTION_JSON_UP_ACK 					"up_ack"

#define WIFI_PASSWORD 									"88888888"

static int g_msgid = 0;
static char g_node_eui[13];
static char g_root_eui[13];

static SemaphoreHandle_t list_mutex;

static int g_lost_connect = 0;

#define MSG_LIST_MAX 100

struct msg_list_t
{
    char eui[13];
    uint32_t msgid;
    uint32_t time;
};

struct msg_list_s
{
    uint16_t num;
    struct msg_list_t messages[MSG_LIST_MAX];
};

static struct msg_list_s msg_list;

static int msg_list_compare ( const void *a, const void *b, void *arg )
{
    struct msg_list_t *p = ( struct msg_list_t * ) a;
    struct msg_list_t *q = ( struct msg_list_t * ) b;
    int *counter = ( int * ) arg;
    int p_count, q_count;

    p_count = p->time;
    q_count = q->time;

    if ( p_count > q_count )
        *counter = *counter + 1;
[... 667 lines cut ...]
#######ERR: msg id:40 timeout##########
#######ERR: msg id:41 timeout##########
#######ERR: msg id:42 timeout##########
#######ERR: msg id:43 timeout##########
#######ERR: msg id:44 timeout##########
#######ERR: msg id:45 timeout##########
#######ERR: msg id:46 timeout##########
#######ERR: msg id:47 timeout##########
#######ERR: msg id:48 timeout##########
#######ERR: msg id:49 timeout##########
#######ERR: msg id:50 timeout##########
#######ERR: msg id:51 timeout##########
#######ERR: msg id:52 timeout##########
#######ERR: msg id:53 timeout##########
#######ERR: msg id:54 timeout##########
#######ERR: msg id:55 timeout##########
#######ERR: msg id:56 timeout##########
#######ERR: msg id:57 timeout##########
#######ERR: msg id:58 timeout##########
#######ERR: msg id:59 timeout##########
#######ERR: msg id:60 timeout##########
#######ERR: msg id:61 timeout##########
I (70752) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK 
child try to reconnect
I (71354) [vendor_ie]: Mesh-Lite Scan done

guo652917087, 2024-09-30:
> 你的意思是 根节点A（Station 连接路由器）和 根节点B（ETH 连接路由器）同时连接路由器时，两个设备的应用层通信会超时？断开 根节点A，根节点 B 就好了？
>
> 根节点 A 和根节点 B 从路由器那边拿到的 ip 分别是多少，路由器可以抓包吗？

两台IP192.168.31.100和192.168.31.101 ，提供wireshark抓包modbus tcp文件，解压后，用wireshark打开。

[抓包文件.zip](https://github.com/user-attachments/files/17183407/default.zip)

tswen, 2024-10-16:
```c
        if ( time_us  > list->messages[i].time + ( 5 * 1000000 ) )
        {
            printf ( "#######ERR: msg id:%ld timeout##########\n" , list->messages[i].msgid );
```
看代码跟 mesh-lite 逻辑以及发送接口没有关系，只是你这里的判断导致走到了 ERR，你可以再检查下你的应用层代码。

guo652917087, 2024-10-16:
> ```c
>         if ( time_us  > list->messages[i].time + ( 5 * 1000000 ) )
>         {
>             printf ( "#######ERR: msg id:%ld timeout##########\n" , list->messages[i].msgid );
> ```
> 
> 看代码跟 mesh-lite 逻辑以及发送接口没有关系，只是你这里的判断导致走到了 ERR，你可以再检查下你的应用层代码。

@tswen 

请问阁下，有没有认真去执行一下测试代码呢？拿两个模块跑一跑呢？你这样的回答，我是真没想到。

我这里打印是 没有收到ack才执行。收到ack就删除msgid。

说实话，测试代码也提供，估计你也没能看清楚代码的逻辑。

开发项目用到esp平台，测试好几个问题。同时反馈给了esp-modbus、ble mesh、esp-idf，就这个项目反馈巨慢，你还要去测试代码，我也临时写了一份，真是不珍惜别人的劳动。

tswen, 2024-10-17:
```c
    esp_bridge_wifi_set_config ( WIFI_IF_AP, &wifi_config );

#if IS_ROOT
    esp_wifi_set_protocol ( WIFI_IF_AP,  WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR );
#else
    // esp_wifi_set_protocol ( WIFI_IF_AP,  WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR );
    // esp_wifi_set_protocol ( WIFI_IF_STA,  WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_LR );
    esp_wifi_set_protocol ( WIFI_IF_STA, WIFI_PROTOCOL_11B| WIFI_PROTOCOL_LR );
#endif

    esp_wifi_set_ps ( WIFI_PS_NONE );

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    mesh_lite_config.join_mesh_ignore_router_status = true;

#if IS_ROOT
    esp_wifi_get_mac ( WIFI_IF_AP, mac );
    sprintf ( g_root_eui, "%02x%02x%02x%02x%02x%02x", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5] );
    mesh_lite_config.join_mesh_without_configured_wifi = false;
#else
    esp_wifi_get_mac ( WIFI_IF_STA, mac );
    sprintf ( g_node_eui, "%02x%02x%02x%02x%02x%02x", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5] );
    mesh_lite_config.join_mesh_without_configured_wifi = true;
#endif
    esp_mesh_lite_init ( &mesh_lite_config );

    list_mutex = xSemaphoreCreateMutex();

    app_wifi_set_softap_info();

#if IS_ROOT
    printf ( "Root node\r\n" );
    esp_mesh_lite_set_allowed_level ( 1 );
#else
    printf ( "Child node\r\n" );
    esp_mesh_lite_set_disallowed_level ( 1 );
#endif

    msg_list_init ( &msg_list );
```
- 这样子试一下，将 set_protocol 放到 mesh_lite 初始化之前。
- 因为从没有出现过因为 wifi 的 protocol 影响到 udp 通信的问题，所以下意识怀疑了用法的问题，对此表示抱歉。