espressif/esp-mesh-lite #122: OTA LAN in inverted direction?

Magnatto, 2024-08-29:
Let's say I have 50 devices on a single mesh-lite and I want to do a LAN OTA update and only have the local network (so no external update).
If i replace the root device with an updated version it will update all others eventually through `esp_mesh_lite_transmit_file_start`, that's great.
But what if it's impossible or very incovenient to replace or know who is the root. Is there a way for me to replace ANY device with the updated version, so it may update the entire network, so the OTA LAN is trasmitted both ways, not just father to children?

tswen, 2024-09-05:
https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota

[image]

Magnatto, 2024-09-05:
> https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/User_Guide_CN.md#esp-mesh-lite-lan-ota
> 
> [![image](https://private-user-images.githubusercontent.com/55544524/364607938-be7466c3-c765-4d82-b1f7-af40e5688a19.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1NjExNjksIm5iZiI6MTcyNTU2MDg2OSwicGF0aCI6Ii81NTU0NDUyNC8zNjQ2MDc5MzgtYmU3NDY2YzMtYzc2NS00ZDgyLWIxZjctYWY0MGU1Njg4YTE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA1VDE4Mjc0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiNDY2YzE2ZmIzOThmMjk5OWRlNWM3MDU1ZmRkNTkyMWFmYzAzN2M3NTgzMjdiODlhNThkNmY4NDUzYWVhZjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.eAafBuNbA8S3HUZovKqJOwK3FAoJqPJKX7ZBb0pWBIA)](https://github.com/tswen/esp-mesh-lite/blob/docs/mesh_lite_LAN_OTA/components/mesh_lite/docs/_static/rootnode_ota_timing_chart_en.svg)

Thanks for the reply, I understand this logic, which relies on an external URL and ALWAYS from root to children.

What I'm asking is if it's possible to NOT to rely on an external URL (replacing one of the devices in the network with an upgraded version), and for the OTA to spread from ANY device (not just from root to children).

This way I don't need an outside network connection (for client safety reasons), but can still update all devices from any given device in the network (since the root is not easily found among many devices and strict information requirements)

tswen, 2024-09-11:
外部 URL 只是在向其父系节点获取固件失败之后的一个保底的 OTA 方式，如果不想要通过外部 URL 进行升级，可以将 extern_url_ota_cb 设置为 NULL。

对于 LAN OTA 来说，每个节点只能向其父系节点获取固件，如果您的根节点在某个分区已经存储了子节点所需要的二进制文件，可以通过 esp_mesh_lite_ota_register_file_transfer_cb 方式将该二进制文件提供给子节点。简单的使用示例请参考：https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/include/esp_mesh_lite_core.h#L828-L885

Magnatto, 2024-09-18:
Thank you, I understand that what I want (LAN OTA from child to parent, instead of parent to child) is not currently possible.
@tswen would it be possible then to add it in the plan for a future feature?
In a network of 50 devices it would be amazing to start the LAN upgrade from any device, not just the root.

tswen, 2024-09-20:
If a child node obtains a firmware and wants to upgrade all devices directly through LAN OTA, the best way is to transfer the firmware to the root node first, and then upgrade it layer by layer.

LAN OTA cannot meet the requirement of upgrading all devices from any node.

Magnatto, 2024-09-24:
> If a child node obtains a firmware and wants to upgrade all devices directly through LAN OTA, the best way is to transfer the firmware to the root node first, and then upgrade it layer by layer.
> 
> LAN OTA cannot meet the requirement of upgrading all devices from any node.

Understood, will try that, thanks