espressif/esp-mesh-lite #123: esp_mesh_lite_get_level  returns 0

highhandli, 2024-09-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

rarely

### Expected behavior

esp_mesh_lite_get_level should be return 1 or grater then 1

### Actual behavior (suspected bug)

When the wifi mesh is networked, when the function esp_mesh_lite_get_level is called to obtain the level, 0 is returned. According to the code on GitHub, the function returns 1 to indicate that it is a router node. If the return value is greater than 1, it is a leaf node or an intermediate node. However, the return value is not defined as 0, and now 0 is returned, and the data is not processed, resulting in the module not being able to send the data to the route through the mesh.

### Error logs or terminal output

```shell
void mesh_node_task(void *arg)
{
    unsigned int retry_timeout = 0;
    EventBits_t bits = 0;
    EventBits_t waitBits = IPV4_CONNECTED_BIT;
    // xEventGroupWaitBits(network_event_group, waitBits, false, true, portMAX_DELAY);

    bits = xEventGroupWaitBits(network_event_group, waitBits, false, true, pdMS_TO_TICKS(30000));
    if((bits & waitBits) != waitBits){
        ESP_LOGE(TAG, "IPV4_CONNECTED_BIT fail....");
        esp_wifi_disconnect();
        esp_netif_destroy_default_wifi(g_station_netif);
        esp_netif_destroy_default_wifi(g_softap_netif);
        esp_wifi_deinit();
        esp_mesh_lite_erase_rtc_store();
        esp_restart();
    }

    sbus_set_dbg_indicate_cb(dbg_indicate);
    sbus_setEventCallback(&dev_listener);
    sbus_rpc_setEventCallback(&dev_rpc_listener);
    
    TickType_t next_wake_time = xTaskGetTickCount();
    while (true) {
        uint8_t layer = esp_mesh_lite_get_level();//the problem is here returns 0
        if(g_layer!=layer){
            g_layer=layer;
            is_force_sbus_restart = 1;
            dev_prop_updata(29, g_layer);
            // status_light_set(g_layer);
        }
        if(is_force_sbus_restart){
            if(is_rpc_client_conn){
                sbus_rpc_reportStatus(g_dev_id, false);
                sbus_rpc_disconnect();
                is_rpc_client_conn = false;
            }
            if(is_sbus_srv_start){
                sbus_reportStatus(g_dev_id, false);
                sbus_stop();
                is_sbus_srv_start = false;
            }
            is_force_sbus_restart = 0;
        }

        //ESP_LOGI(TAG, "g_layer[%d]", g_layer);
        //g_layer = 0;
        if(g_layer==0){
        }else if(g_layer==1){
            // if(g_need_sync_time){
            //     obtain_time();
            // }
            // ROOT node, sbus srv mode
            if(is_rpc_client_conn){
                sbus_rpc_reportStatus(g_dev_id, false);
                sbus_rpc_disconnect();
                is_rpc_client_conn = false;
            }
            if(!is_sbus_srv_start){
                if(retry_timeout==0){
                    is_sbus_srv_start=sbus_start(SBUS_SERVER_PORT, RPC_SERVER_PORT, g_dev_id);
                    if(!is_sbus_srv_start){
                        retry_timeout = 2000/DEV_WORK_LOOP_INTERVAL;
                    }
                    else{
                        send_reportDev();
                    }
                }else{
                    retry_timeout--;
                }
            }else{
                if(g_need_sync_time){
                    obtain_time();
                }
                if(!ota_flag)
                    sbus_loop(DEV_WORK_LOOP_INTERVAL);
            }
        }else{
            // device node, sbus rpc client mode
            if(is_sbus_srv_start){
                sbus_reportStatus(g_dev_id, false);
                sbus_stop();
                is_sbus_srv_start = false;
            }
            if(!is_rpc_client_conn){
                if(retry_timeout==0){
                    char * root_ip = NULL;
                    if(!get_mesh_root_ip(&root_ip) || !(is_rpc_client_conn=sbus_rpc_connect(root_ip, RPC_SERVER_PORT, g_dev_id))){
                        retry_timeout = 2000/DEV_WORK_LOOP_INTERVAL;
                    }
                    else{
                        send_reportDev();
                    }
                }else{
                    retry_timeout--;
                }
            }else{
                if(g_need_sync_time){ //lgw . add 20240912
                    obtain_time();
                }
                if(!ota_flag)
                    sbus_rpc_loop(DEV_WORK_LOOP_INTERVAL);

                if(retry_timeout==0){
                    if(is_mesh_root_ip_changed())
                    {
                        ESP_LOGI(TAG, "detect mesh root ip changed, force sbus restart");
                        is_force_sbus_restart = 1;
                    }else
                        retry_timeout = 2000/DEV_WORK_LOOP_INTERVAL;
                }else{
                    retry_timeout--;
                }
            }
        }

        vTaskDelayUntil(&next_wake_time, pdMS_TO_TICKS(DEV_WORK_LOOP_INTERVAL));    
    }
    sbus_stop();
    vTaskDelete(NULL);
}
```

### Steps to reproduce the behavior

I used 15 modules to network, and after 7 days of operation, 2 modules had this situation

### Project release version

mesh lite version:v0.9.0 - 2023-10-31

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Windows

### Operating system version

freeRTOS

### Shell

other (details in Additional context)

tswen, 2024-09-18:
Can you provide the log? esp_mesh_lite_get_level returns 0, which means WiFi is not connected properly.

highhandli, 2024-09-18:
> Can you provide the log? esp_mesh_lite_get_level returns 0, which means WiFi is not connected properly.

there is no more log.this happen in the code "uint8_t layer = esp_mesh_lite_get_level();". If I execute the function "esp_restart" and then restart, it's still the same. If I reset via hardware, the function "esp_mesh_lite_get_level" returns 1 or greater than 1 and then it works.

highhandli, 2024-09-18:
> Can you provide the log? esp_mesh_lite_get_level returns 0, which means WiFi is not connected properly.

[Uploading 勘察箱异常数据日志.txt…]()

highhandli, 2024-09-18:
> Can you provide the log? esp_mesh_lite_get_level returns 0, which means WiFi is not connected properly.

设备就在路由旁边，感觉应该不是连不上Wifi

highhandli, 2024-09-18:
> Can you provide the log? esp_mesh_lite_get_level returns 0, which means WiFi is not connected properly.

The device is next to the router, and it doesn't feel like it's not connected to Wifi

tswen, 2024-09-20:
Please update mesh lite to the latest version and try again.