espressif/esp-mesh-lite #124: The compilation shows that the enumerated variable WIFI_PROV_CUST_DATA_RECV does not exist

highhandli, 2024-09-20:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Compiled through

### Actual behavior (suspected bug)

The compilation did not pass

### Error logs or terminal output

```shell
/home/ligaowen/smart_gw/esp_projs/esp-idf-all-5.1/components/wifi_provisioning/src/manager.c: In function 'wifi_prov_mgr_configure_meshlite':
/home/ligaowen/smart_gw/esp_projs/esp-idf-all-5.1/components/wifi_provisioning/src/manager.c:1340:22: error: 'WIFI_PROV_CUST_DATA_RECV' undeclared (first use in this function); did you mean 'WIFI_PROV_CRED_RECV'?
 1340 |     execute_event_cb(WIFI_PROV_CUST_DATA_RECV, data, size);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~
      |                      WIFI_PROV_CRED_RECV
```

### Steps to reproduce the behavior

I couldn't find the definition of the enumerated variable WIFI_PROV_CUST_DATA_RECV from the latest documentation on the official website, and the compilation did not pass

### Project release version

the latest

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

freeRTOS

### Shell

other (details in Additional context)

tswen, 2024-09-20:
https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/src/wifi_prov/wifi_provisioning.patch

Please check if this patch is not applied to esp-idf successfully. If you do not use the zero-provisioning feature, you can manually add the line of code https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/src/wifi_prov/wifi_provisioning.patch#L9 to esp-idf first.

tswen, 2025-02-26:
Because some customers have reported that they do not want to modify esp-idf without knowing why, we maintain the https://github.com/espressif/esp-mesh-lite/tree/master/components/wifi_provisioning component separately. If you want to use the network provisioning in mesh, you can add a dependency in main/idf_component.yml under the project: https://github.com/espressif/esp-mesh-lite/blob/master/examples/mesh_wifi_provisioning/main/idf_component.yml#L16-L18