espressif/esp-mesh-lite #126: How to assign unique IP to each AP of same layer nodes

Rbhagwat62, 2024-10-07:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

I am using mesh_lite version 1.0.0 and ESP IDF 5.2.1

Currently I am working on the mesh local control example of the mesh lite. When the network form each node check for AP IP is already in used or not by calling esp_bridge_netif_network_segment_is_used API from bridge_common.c. This works correctly when only single device is present on each layer. Each layer gives unique IP to it AP interface.

But when there are multiple devices connect as child to any node at particular layer each child device will assigns same IP to AP interface. 

I want to assign unique IP assigned to each node AP though the child nodes are at same layer. How I can achieve it.

I have attached one ppt file to demonstrate the issue.
[Mesh_unique_AP_IP_issue.pptx](https://github.com/user-attachments/files/17274042/Mesh_unique_AP_IP_issue.pptx)

Thank you

tswen, 2024-10-09:
Hello, as demonstrated in the second PPT slide, both C1 and C2 have the same softAP network segment, and they are only responsible for assigning IPs to the devices directly connected to them. 

Although C3 and C4 have the same IP address, it will not affect their communication with the router and PC. 

In the current framework, there is no IP conflict detection between devices connected to different parent nodes at the same level because this does not affect communication, so it does not seem to be necessary. However, if it's essential for each device to have a unique IP address, you can set different softAP network segments for each device using the `esp_bridge_netif_set_ip_info` API.

Rbhagwat62, 2024-10-09:
@tswen,

Thanks for the reply. 

With esp_bridge_netif_set_ip_info API do I need to manually set IP to each device or it will set automatically as implemented in esp_bridge_netif_request_ip.

I want to set all device IPs automatically, if it is possible with esp_bridge_netif_set_ip_info can you please guide me for its implementation.

Rbhagwat62, 2024-10-10:
In function **esp_bridge_netif_network_segment_conflict_update** in **bridge_common.c** , after getting new IP using **esp_bridge_netif_request_ip** API, I used **esp_bridge_netif_set_ip_info** instead of **esp_netif_set_ip_info** like

**esp_bridge_netif_set_ip_info(p->netif, &allocate_ip_info, true, true);**

I tried enabling and disabling the **save_to_nvs** option.

After doing above changes, I was expecting , when function **esp_bridge_netif_request_ip** is called from **esp_bridge_netif_network_segment_conflict_update**,  **list->custom_check_cb** should give **ip_sengment_is_used** true when second device on layer 2 check for IP which is already allotted for the first device softAP. But it return false and same IP gets assigned to second device softAP.

Is my understanding and implementation for  esp_bridge_netif_set_ip_info is correct? Please suggest to correct solution.

tswen, 2024-10-21:
It is not recommended to modify the iot_bridge component. You need to manage the IP subnets of different devices at the application layer, and then set the IP information for different devices using esp_bridge_netif_set_ip_info. This is because the iot bridge and mesh lite itself do not check the IP subnets of child nodes in other branches

Rbhagwat62, 2024-10-24:
@tswen,

Thanks for your response. Got your point.

To set the IP information using **esp_bridge_netif_set_ip_info** we need to pass the esp_netif_t* object. Since I want to set softAP IP I will required softap netif object created during **esp_bridge_create_all_netif** correct?

How can I get the softAP esp_netif_t object in application my application code?

tswen, 2024-10-25:
Through this API, the netif of softap can be returned: `esp_netif_get_handle_from_ifkey("WIFI_AP_DEF")`

Rbhagwat62, 2024-10-28:
Tested esp_netif_get_handle_from_ifkey works correctly.

Thanks for support

closed 2024-10-28 by Rbhagwat62