espressif/esp-mesh-lite #127: Need understanding for esp_mesh_lite_set_rssi_threshold API

Rbhagwat62, 2024-10-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

With reference to [comment from one of issue](https://github.com/espressif/esp-mesh-lite/issues/93#issuecomment-2178209897) I want to have understanding of **esp_mesh_lite_rssi_threshold_list_t** object. 

To set the RSSI threshold we need to use **esp_mesh_lite_set_rssi_threshold** API and need to pass **esp_mesh_lite_rssi_threshold_list_t** array. In the reference comment it is given as 

static const esp_mesh_lite_rssi_threshold_list_t rssi_level[] = {
        {0,  -35, 20},
        {-35,  -55, 15},
        {-55,  -75, 10}
    };

for each entry first is min RSSI, second is max RSSI and last is threshold which is unsigned value. What is threshold value and how it is used?
There are three entries in the array, what is the significant of each entry?

Can you please provide the explanation.

Thanks.

tswen, 2024-10-25:
When a device is trying to join a network, it determines the best parent node based on the signal strength of surrounding nodes that have already successfully joined the network (relative to the current device) and their hierarchy level. Typically, the device will prioritize selecting nodes that are at a lower hierarchy level (i.e., closer to the root node) as the best parent node. However, it will make exceptions if a higher-level node has a signal strength that exceeds the signal strength of a lower-level node by a certain threshold (default threshold is 15 dB), or if a lower-level node has a signal strength that is not weaker than the signal strength of a higher-level node by a certain threshold (also defaulting to 15 dB).

This interface allows for a stepped setting of different threshold sizes within specific signal strength ranges:

- `{0, -35, 20}` means that when the signal strength of surrounding nodes (relative to the current device) is between 0 and -35 dB, the threshold is 20 dB.
  
- `{-55, -75, 10}` means that when the signal strength of surrounding nodes is between -55 and -75 dB, the threshold is 10 dB.
  
- `{-75, -125, 5}` means that when the signal strength of surrounding nodes is between -75 and -125 dB, the threshold is 5 dB.

To encourage the device to connect to nearby nodes, you should reduce the threshold. To push the device to prefer connecting to lower hierarchy nodes, you should increase the threshold.

Since vendor information is passively received during scanning, it's uncertain which information will be received first. If the root node's information is received first, followed by the second-level node's information, the device will assess the signal strength of the second-level node according to its threshold to determine whether it can replace the original optimal parent node.

Rbhagwat62, 2024-10-28:
@tswen 

thanks for detailed explanation I will test it by changing the values.

While testing I faced one case, let me explain. 

I used 4 nodes running with mesh lite local control example on each device, also configured the router details. I placed devices as shown in image 
[image]

From the image node1 and node 2 are in the range of router but not in the range of each other. Also node 3 and node 4 in the range of node 1 and node 2 respective. So when the network is formed I observed node 1 and node 3 formed their own network with node 1 as root node. In other side Node 2 and Node 4 formed another mesh network with node 2 as root node.

So is there any way I will have only one mesh network with one root with this type of device placement? or I need to always ensure each device will be in range of other device.

tswen, 2024-10-28:
Root node 1 and root node 2 cannot connect to each other because they are outside each other’s wireless range. Therefore, they cannot fusion properly in this situation.

Rbhagwat62, 2024-10-28:
Ok, Thanks for your support

closed 2024-10-28 by Rbhagwat62