espressif/esp-mesh-lite #129: My ESP-IDF v5.2.2 is not the same as the github v5.2.2 tagged version (IEP-1267) (IDFGH-13924)

sfeutrier, 2024-11-12:
**Describe the bug**
I'm trying for hours to make the wifi provisioning sample to work, without any luck!
After investigation of the source code, I found out that everything related to "esp_wifi_start" (starting Wi-Fi in Station Mode) is commented out.
This is not the case here: https://github.com/espressif/esp-idf/blob/v5.2.2/components/wifi_provisioning/src/manager.c (line 1556) while in my folder "C:\Espressif\frameworks\esp-idf-v5.2.2\components\wifi_provisioning\src\manager.c", this is the line 1658... How ?
What is the correct way to solve my issue?

**To Reproduce**
Steps to reproduce the behavior:
I guess install Espressif IDE with esp-idf v5.2.2

**Expected behavior**
Folder files and tagged version files should be identical

**Screenshots**
What I have in my local version of esp-idf
[image]
What I should have when looking at the tag v5.2.2 in Github
[image]

tswen, 2024-11-14:
@sfeutrier Hello, you can update the mesh-lite component to the latest master branch. The new mesh lite component will maintain [wifi_provisioning](https://github.com/espressif/esp-mesh-lite/tree/master/components/wifi_provisioning) separately without affecting idf.

sfeutrier, 2024-11-26:
Hey @tswen I tried last week to ypdate my mesh-lite component but there are managed by the IDF Component manager and I can succeed to update.
Do you know where I can find the proper way to update it and test?

tswen, 2024-12-18:
@sfeutrier mesh lite component has been updated to 1.0.1, which includes the changes mentioned above.

Indastri, 2025-01-21:
Hi @sfeutrier, If your issue was resolved can this issue be closed? Thanks!

Indastri, 2025-02-19:
As there was no further response from the user - closing this issue. If the problem persists feel free to reopen this or create a new issue.

Cheers!

closed 2025-02-19 by Indastri