espressif/esp-mesh-lite #131: Hidden mode AP not working!

ruslanslobozhenuk, 2024-11-25:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm planning to develop a firmware for controlling devices in a mesh network (>100 units). I updated to the latest commit (necessary for OTA), and found this bug! I took the local_control example and slightly modified the code to reproduce the situation. Source code and logs are attached. Thanks in advance!

### Actual behavior (suspected bug)

Hidden mode not working!

Source code:

/*
 * SPDX-FileCopyrightText: 2022-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <inttypes.h>
#include "esp_log.h"
#include "esp_mac.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"

#include "esp_wifi.h"
#include "nvs_flash.h"
#include <sys/socket.h>

#include "esp_bridge.h"
#include "esp_mesh_lite.h"

#define PAYLOAD_LEN       (1456) /**< Max payload size(in bytes) */

static int g_sockfd    = -1;
static const char *TAG = "local_control";

/**
 * @brief Create a tcp client
 */
static int socket_tcp_client_create(const char *ip, uint16_t port)
{
    ESP_LOGD(TAG, "Create a tcp client, ip: %s, port: %d", ip, port);

    esp_err_t ret = ESP_OK;
    int sockfd    = -1;
    struct ifreq iface;
    memset(&iface, 0x0, sizeof(iface));
    struct sockaddr_in server_addr = {
        .sin_family = AF_INET,
        .sin_port = htons(port),
        .sin_addr.s_addr = inet_addr(ip),
    };

    sockfd = socket(AF_INET, SOCK_STREAM, 0);
    if (sockfd < 0) {
        ESP_LOGE(TAG, "socket create, sockfd: %d", sockfd);
        goto ERR_EXIT;
    }

    esp_netif_get_netif_impl_name(esp_netif_get_handle_from_ifkey("WIFI_STA_DEF"), iface.ifr_name);
    if (setsockopt(sockfd, SOL_SOCKET, SO_BINDTODEVICE,  &iface, sizeof(struct ifreq)) != 0) {
[... 530 lines cut ...]
Nothing

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

other (details in Additional context)

### Additional context

VisualGDB

ruslanslobozhenuk, 2024-11-26:
I tried to use esp-bootloader-plus and got some problems. Is there any plan to support esp-bootloader-plus with mesh-lite?

tswen, 2024-11-28:
> I tried to use esp-bootloader-plus and got some problems. Is there any plan to support esp-bootloader-plus with mesh-lite?

These two are different solutions and do not conflict. What problem do you encounter?

ruslanslobozhenuk, 2024-11-28:
Problem solved! If anyone is interested, then while experimenting, at some point I executed the following code:
...
wifi_country_t country = {
.cc = "JP",
.schan = 1,
.nchan = 14,
.policy = WIFI_COUNTRY_POLICY_AUTO,
};
esp_wifi_set_country(&country);
...
and after that I didn't return it back! Clearing the flash didn't lead to anything! Remembering this, I returned everything to default, and everything worked!

closed 2024-12-06 by ruslanslobozhenuk