espressif/esp-mesh-lite #133: Mesh level not correct.

ruslanslobozhenuk, 2024-12-06:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

I have 11 nodes. I make realtime diagram nodes hierachy (every 10 seconds nodes send heartbeat which contains mesh level. 
[image]
[Number 1] - realtime Mesh Level.  In [...] second string - parent node. Help me please. 
Thank you!

### Actual behavior (suspected bug)

Nodes 404CCA57F03C and  404CCA5E6D48 Mesh Level 1, but have parent Node  5432040C2100. Its problem (((.

### Steps to reproduce the behavior

Mesh build

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

tswen, 2024-12-09:
What is the mesh lite version (commit) used? Are there logs for Nodes 404CCA57F03C, 404CCA5E6D48 and Node 5432040C2100?

ruslanslobozhenuk, 2024-12-09:
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1954
load:0x4086c110,len:0x10a4
load:0x4086e610,len:0x3810
entry 0x4086c180
I (23) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (24) boot: compile time Dec  6 2024 09:31:17
I (24) boot: chip revision: v0.0
I (27) boot: efuse block revision: v0.2
I (32) boot.esp32c6: SPI Speed      : 80MHz
I (36) boot.esp32c6: SPI Mode       : DIO
I (41) boot.esp32c6: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 sec_cert         unknown          3f 06 00009000 00007000
I (69) boot:  1 nvs              WiFi data        01 02 00010000 00008000
I (77) boot:  2 otadata          OTA data         01 00 00018000 00002000
I (84) boot:  3 phy_init         RF data          01 01 0001a000 00001000
I (92) boot:  4 solo             WiFi data        01 02 0001b000 00003000
I (99) boot:  5 ota_0            OTA app          00 10 00020000 001db000
I (107) boot:  6 ota_1            OTA app          00 11 00200000 001db000
I (114) boot:  7 fctry            WiFi data        01 02 003f8000 00008000
I (122) boot: End of partition table
I (126) esp_image: segment 0: paddr=00200020 vaddr=42150020 size=4d3c0h (316352) map
I (189) esp_image: segment 1: paddr=0024d3e8 vaddr=40800000 size=02c30h ( 11312) load
I (197) esp_image: segment 2: paddr=00250020 vaddr=42000020 size=146168h (1335656) map
I (428) esp_image: segment 3: paddr=00396190 vaddr=40802c30 size=1fbcch (129996) load
I (461) esp_image: segment 4: paddr=003b5d64 vaddr=40822800 size=043b0h ( 17328) load
I (475) boot: Loaded app from partition at offset 0x200000
I (475) boot: Disabling RNG early entropy source...
I (487) cpu_start: Unicore app
I (496) cpu_start: Pro cpu start user code
I (496) cpu_start: cpu freq: 160000000 Hz
I (496) app_init: Application information:
I (499) app_init: Project name:     solo_sensor_v1
I (504) app_init: App version:      1.0.2_f4a097d
I (510) app_init: Compile time:     Dec  6 2024 16:18:15
I (516) app_init: ELF file SHA256:  38f9e6216...
I (521) app_init: ESP-IDF:          v5.3.2-dirty
I (526) efuse_init: Min chip rev:     v0.0
I (531) efuse_init: Max chip rev:     v0.99
I (536) efuse_init: Chip rev:         v0.0
I (541) heap_init: Initializing. RAM available for dynamic allocation:
I (548) heap_init: At 4082F050 len 0004D5C0 (309 KiB): RAM
I (554) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (560) heap_init: At 500018D0 len 00002718 (9 KiB): RTCRAM
I (567) spi_flash: detected chip: generic
I (571) spi_flash: flash io: dio
W (575) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (589) sleep: Configure to isolate all GPIO pins in sleep state
I (595) sleep: Enable automatic switching of GPIO sleep configuration
I (602) coexist: coex firmware version: cbb41d7
I (617) coexist: coexist rom version 5b8dcfa
I (617) main_task: Started on CPU0
I (617) main_task: Calling app_main()
I (617) uart: queue free spaces: 20
W (617) dali: Solo ESP32C6 DALI driver initialized
I (1627) led_indicator: LED Indicator Version: 0.9.3
I (1627) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1627) led_indicator: Indicator create successfully. type:LED Strips mode, hardware_data:0x40837320, blink_lists:custom
W (1637) main: app_main 138 app_main const heap 300736
W (1647) main: free heap 300736, minimum 300736
I (1657) main: Load SOLO configuration success.
I (1657) bridge_common: esp-iot-bridge version: 0.11.9
I (1657) pp: pp rom version: 5b8dcfa
[... 184 lines cut ...]
I (8587) [vendor_ie]: RTC store: temp_mesh_id:191; ssid:NODE_0c2101; bssid:54:32:04:0c:21:01; crc:44681661
W (8597) solo_mesh: Got IP, channels: [13,0], layer: 2, self mac: 40:4c:ca:57:f0:3c, parent bssid: 54:32:04:0c:21:01, parent rssi: -39, free heap: 170164
I (8617) mqtt: Publishing to node/404CCA57F03C/d
W (8617) mqtt_client: Publish: Losing qos0 data when client not connected
E (8627) mqtt: MQTT Publish failed
I (8627) esp_diag_var_data_pt_write: ts: 8140458, type: 1, index: 12, value type: 12, value len: 4, value[c0a80502]
I (8637) esp_diag_var_data_pt_write: ts: 8149774, type: 1, index: 13, value type: 12, value len: 4, value[ffffff00]
I (8647) esp_diag_var_data_pt_write: ts: 8160768, type: 1, index: 14, value type: 12, value len: 4, value[c0a80501]
W (8637) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x1000121, TALO:0xc043254, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
I (8677) esp_diag_var_data_pt_write: ts: 8182658, type: 1, index: 6, value type: 2, value len: 1, value[07]
I (8687) esp_diag_var_data_pt_write: ts: 8192704, type: 1, index: 7, value type: 2, value len: 1, value[01]
I (8697) esp_diag_var_data_pt_write: ts: 8203112, type: 1, index: 8, value type: 2, value len: 1, value[01]
I (8707) esp_diag_var_data_pt_write: ts: 8213651, type: 1, index: 10, value type: 2, value len: 1, value[07]
I (8717) esp_diag_var_data_pt_write: ts: 8224033, type: 1, index: 11, value type: 2, value len: 1, value[02]
I (8727) solo_cmd_resp: Enabling Command-Response Module.
I (8727) esp_solo_node: Node attribute cmd-resp created
I (8737) mqtt: Connecting to MQTT broker [192.168.240.1]
I (8747) esp_solo_core: Waiting for MQTT connection
I (8747) chip: CPU load: 21%
I (8797) mqtt: MQTT Connected
I (8797) app_solo: SOLO_MQTT_EVENT_CONNECTED.
I (8797) solo_cmd_resp: Checking for pending commands.
I (8797) mqtt: Publishing to node/404CCA57F03C/out
I (8807) esp_solo_node_config: Generated Node config of length 2718
I (8807) mqtt: Publishing to node/404CCA57F03C/config

ruslanslobozhenuk, 2024-12-09:
Mesh-Lite commit id: 7e45e27

tswen, 2024-12-23:
Based on the provided logs, there is a log entry: `solo_mesh: Got IP, channels: [13,0], layer: 2, self mac: 40:4c:ca:57:f0:3c, parent bssid: 54:32:04:0c:21:01, parent rssi: -39, free heap: 170164`. It seems that the layer display is correct.