espressif/esp-mesh-lite #135: no_route例程怎么联网呢？

heyangxinmei, 2024-12-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

no_route例程实现联网功能

### Actual behavior (suspected bug)

no

### Steps to reproduce the behavior

no

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows10

### Shell

ZSH

### Additional context

no

tswen, 2024-12-11:
根节点连接路由器，mesh 设备才能连接外网。no_router 示例中根节点不连接路由器，所以是无法连接外网的。

tswen, 2024-12-12:
可以的，你可以将 mesh lite 方案与 https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module 这个示例结合

heyangxinmei, 2024-12-12:
[image]
这个USB CDC endpoint address 怎么填写啊？我使用的是AIR780E，AT指令。

tswen, 2024-12-12:
modem 相关的问题可以在 solution 方案下面提 issue

heyangxinmei, 2024-12-12:
好