espressif/esp-mesh-lite #137: [ESP_Mesh_Lite_Comm]: Disconnect reason : 212，why cannot the device be connected after the device is restarted

yel-best, 2025-01-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Why cannot the device be connected after the device is restarted?

```
I (2300) main_task: Returned from app_main()
I (4762) [vendor_ie]: Mesh-Lite Scan done 
I (4763) [vendor_ie]: RTC store: temp_mesh_id:255; ssid:apps-iot; bssid:00:00:00:00:00:00; crc:1851272269
I (4767) [vendor_ie]: wifi_cfg apps-iot router_config apps-iot
I (4775) [vendor_ie]: esp_mesh_lite_wifi_connect return ESP_OK 
I (4781) app_main: mesh lite app_scan_end
I (4786) app_bee: WI-FI Station Finished scanning AP
E (7604) [ESP_Mesh_Lite_Comm]: Disconnect reason : 212 
I (7604) app_bee: WI-FI Station disconnected from AP
W (7605) app_bee: WI-FI Disconnected 1. Will try reconnecting in a while...
I (9702) [vendor_ie]: esp_mesh_lite_wifi_connect return ESP_OK 
I (9702) [ESP_Mesh_Lite_Comm]: Retry to connect to the AP
E (12531) [ESP_Mesh_Lite_Comm]: Disconnect reason : 212 
I (12531) app_bee: WI-FI Station disconnected from AP
W (12532) app_bee: WI-FI Disconnected 2. Will try reconnecting in a while...
I (14702) [vendor_ie]: esp_mesh_lite_wifi_connect return ESP_OK 
I (14702) [ESP_Mesh_Lite_Comm]: Retry to connect to the AP
E (17531) [ESP_Mesh_Lite_Comm]: Disconnect reason : 212 
I (17531) app_bee: WI-FI Station disconnected from AP
```

tswen, 2025-01-10:
WIFI_REASON_NO_AP_FOUND_IN_RSSI_THRESHOLD      = 212,    /**< No AP found in RSSI threshold */

If a rssi threshold is set in wifi_config.sta.threshold.rssi, and the rssi is lower than this threshold when connecting to the router, the connection will fail.

yel-best, 2025-01-20:
> WIFI_REASON_NO_AP_FOUND_IN_RSSI_THRESHOLD = 212, /**< No AP found in RSSI threshold */
> 
> If a rssi threshold is set in wifi_config.sta.threshold.rssi, and the rssi is lower than this threshold when connecting to the router, the connection will fail.

Is there any documentation available for this? 😃

tswen, 2025-01-20:
Currently not available. This is a new feature in the Wi-Fi lib in the new idf. The documentation has not been updated yet. 212 means No AP found in RSSI threshold.