espressif/esp-mesh-lite #138: What does join_mesh_without_configured_wifi ( CONFIG_JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO ) do?

mrjohannchang, 2025-01-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Could you please elaborate what `join_mesh_without_configured_wifi` of `esp_mesh_lite_config_t` ( `JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO` / `CONFIG_JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO` ) does? I want to have mesh formed regardless if the root node is connected to a router. I get why `join_mesh_ignore_router_status` should be set to true, it's straightforward, but why example has `join_mesh_without_configured_wifi` set to true for non-root nodes in this case?

tswen, 2025-01-14:
- JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO 设置为 true：
    - 设备在上电执行 esp_mesh_lite_start 之后就会尝试加入 Mesh 网络，即使设备没有配网（没有路由器信息）。
- JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO 设置为 false：
    - 设备如果已经配过网，存在路由器信息，那么设备在上电执行 esp_mesh_lite_start 之后就会尝试加入 Mesh 网络；
    - 如果设备没有配过网，不存在路由器信息，那么设备在上电执行 esp_mesh_lite_start 之后就不会尝试加入 Mesh 网络，除非收到配网信息之后，用户主动调用 esp_mesh_lite_set_router_config()； esp_mesh_lite_connect();

- join_mesh_ignore_router_status 设置为 true：
    - 指子节点尝试加入 Mesh 网络时，应当忽略根节点是否连接到路由器，即使根节点没有连接路由器，子节点与根节点 Mesh ID 保持一致其也可以加入到 Mesh 网络中。
    - 这个配置仅对于 no_router 示例里面的子节点有用，其他情况应当避免这个配置设置为 true。

mrjohannchang, 2025-02-03:
謝謝您 @tswen 

> 即使设备没有配网（没有路由器信息）

請問配網、路由器訊息指的是 Wi-Fi station 的 SSID、password 嗎？意即

```
esp_wifi_set_config(WIFI_IF_STA, <路由器信息>)
```

或

```
esp_bridge_wifi_set_config(WIFI_IF_STA, <路由器信息>);
```

關於第二點的「设备如果已经配过网」、「如果设备没有配过网」，請問什麼情況下是有配過網？是指曾經有 call 過 `esp_bridge_wifi_set_config(WIFI_IF_STA, ...)` 嗎？這個有配過、沒配過的狀態，ESP32 reset 後會維持嗎？

就我對 Wi-Fi 理解來推測 mesh-lite 的行為，一個節點如果沒有設定 Wi-Fi station 模式的 SSID、password，他就不會連接任何 access point。如果是這樣的話，我不明白 `JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO` 設定為 true 會有什麼作用。是 mesh-lite 裡有有設定預設的 access point 資料讓 station 去連接嗎？

> 这个配置仅对于 no_router 示例里面的子节点有用，其他情况应当避免这个配置设置为 true。

請問這個的原因是什麼？如果子節點加入的 mesh 的根節點沒有連接到路由器，會有什麼副作用嗎？

tswen, 2025-02-07:
- 配网指 esp_mesh_lite_set_router_config，即让设备知道路由器信息
- 如果 JOIN_MESH_WITHOUT_CONFIGURED_WIFI_INFO  设置为 true，设备即使没有配网，其也可以找到一个 Mesh ID 匹配的父节点并连接上去。
-  在不固定根节点的场景下，如果 join_mesh_ignore_router_status 设置为 true，有可能会造成设备之前的环形连接。

closed 2025-07-17 by mrjohannchang