espressif/esp-mesh-lite #140: How does it work when BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC ( CONFIG_BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC ) enabled?

mrjohannchang, 2025-01-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

How it works when `BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC` ( `CONFIG_BRIDGE_SOFTAP_SSID_END_WITH_THE_MAC` ) is enabled? Does the library strip out the appended partial MAC address from the AP scan result internally?

tswen, 2025-01-14:
仅仅是在 BRIDGE_SOFTAP_SSID 后加个 Mac 后缀而已。

mrjohannchang, 2025-07-17:
I guess esp-mesh-lite uses custom IEs (Information Elements) instead of fixed SSID to determine the parent and probably other mesh info. Thank you @tswen!

closed 2025-07-17 by mrjohannchang