espressif/esp-mesh-lite #144: mesh-lite + hosted 是否可以同时支持？

bingogogogogogogogo, 2025-01-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

mesh-lite + hosted 是否可以同时支持？ 一个SOC通过SDIO连接esp32当网卡使用，同时可以mesh组网是否可以实现，麻烦指导一下谢谢

tswen, 2025-01-14:
可以，将 BRIDGE_DATA_FORWARDING_NETIF_SDIO 配置项打开就好了

xcguang, 2025-01-14:
esp32 类此路由器，和 soc 的协议栈不冲突

bingogogogogogogogo, 2025-01-16:
把esp32通过USB插入到Ubuntu当网卡使用，esp32 WIFI sta连接路由器。路由器连接的设备有esp32和一台PC电脑，PC想访问Ubuntu启动的http服务器，该如何做到？
[image]
@xcguang @tswen

tswen, 2025-01-16:
https://github.com/espressif/esp-iot-bridge/issues/90#issuecomment-2451489800
可以参考这个回答。对于 iot-bridge 那种一层 napt 这样做是没问题的，但是如果要涉及到 mesh，就不建议这样做了，这样会很复杂。

bingogogogogogogogo, 2025-01-16:
请问下上述图片的连接模式，我在办公室环境测试吞吐，下载2.86Mbps 上传4.14Mbps，这样的数据是否太低了，有什么方法可以优化

tswen, 2025-01-20:
这个取决于周围的无线环境干扰，如果周围无线环境干扰较小，相应吞吐也会高一些。对于 usb 方案来说，在较好的环境中可以达到五六兆左右的吞吐。