espressif/esp-mesh-lite #145: Linker : undefined reference to `esp_mesh_lite_set_fusion_config'

BlastMan64, 2025-01-24:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Call esp_mesh_lite_set_fusion_config function to allow fusion between networks

### Actual behavior (suspected bug)

The code compiles but linker fails.
However, the function is declared in esp_mesh_lite_core.h l.544
`esp_err_t esp_mesh_lite_set_fusion_config(esp_mesh_lite_fusion_config_t *config);`

If the call of esp_mesh_lite_set_fusion_config is removed, the code compiles & runs well.

### Error logs or terminal output

```shell
[7/9] Linking CXX executable no_router.elf
FAILED: no_router.elf
C:\windows\system32\cmd.exe /C "cd . && C:\esp\tools\tools\xtensa-esp-elf\esp-14.2.0_20241119\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/esp/projects/exodus-next-gen/pocs/esp-mesh-lite/examples/no_router/build/no_router.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.bt_funcs.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\no_router.elf.rsp -o no_router.elf && cd ."
C:/esp/tools/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(no_router.cpp.obj):(.literal.app_main+0x24): undefined reference to `esp_mesh_lite_set_fusion_config'
C:/esp/tools/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(no_router.cpp.obj): in function `app_main':
C:/esp/projects/exodus-next-gen/pocs/esp-mesh-lite/examples/no_router/main/no_router.cpp:138:(.text.app_main+0x74): undefined reference to `esp_mesh_lite_set_fusion_config'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### Steps to reproduce the behavior

Reproduced with no_router example in app_main :

```c++
esp_mesh_lite_fusion_config_t fusion_config = {
      .fusion_start_time_sec = 60,   // Set the fusion to start 60 seconds after the mesh network is up.
      .fusion_frequency_sec = 120,   // Set the network to perform fusion every 120 seconds thereafter.
};
 
// Apply the fusion configuration to the mesh network.
esp_mesh_lite_set_fusion_config(&fusion_config);

esp_mesh_lite_init(&mesh_lite_config);
``` 

### Project release version

latest - commit 2ad8162

### System architecture

other (details in Additional context)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

other (details in Additional context)

### Additional context

VSCode 1.96.4
ESP-IDF 5.4.0
ESP32S3 / ESP32C6

tswen, 2025-02-07:
- Devices do not need to call `esp_mesh_lite_set_fusion_config`. As long as the device is a root node, it will periodically trigger fusion. `esp_mesh_lite_set_fusion_config` is only used to set some parameters for fusion.
- Additionally, it is important to note that `esp_mesh_lite_set_fusion_config` should be called after `esp_mesh_lite_init`.
- The issue `undefined reference to 'esp_mesh_lite_set_fusion_config'` primarily arises because the internal API of the library was previously updated, but the external code has not been modified accordingly. Changing `esp_mesh_lite_set_fusion_config` to `esp_mesh_lite_comm_set_fusion_config` will resolve this issue.

BlastMan64, 2025-02-07:
Thanks for this explanations.
I was not sure that the fusion was enabled by default - it is

I tried `esp_mesh_lite_comm_set_fusion_config`, however it does not work either :
```
C:/esp/projects/exodus-next-gen/pocs/esp-mesh-lite/examples/no_router/main/no_router.cpp:274:5: error: 'esp_mesh_lite_comm_set_fusion_config' was not declared in this scope; did you mean 'esp_mesh_lite_set_fusion_config'?
  274 |     esp_mesh_lite_comm_set_fusion_config(&fusion_config);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     esp_mesh_lite_set_fusion_config
```

tswen, 2025-02-07:
The esp_mesh_lite_core.h header file also needs to be modified.