espressif/esp-mesh-lite #149: Raw Data Transfer between nodes and API methods?

mjgci, 2025-02-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Reference: Mesh Local Control Example

I am looking to transfer large amount of data between the root node and the child nodes.

Using ESP-NOW api such as "esp_mesh_lite_espnow_send()", the data size gets limited to ~250B which is insufficient for large data transfers in one transaction. It can address my requirements to communicate with a specific node(using MAC address as a reference) however, I will need to send the chunks of data if the length is >250B in my application code. This will in turn will affect the speed of the data transfers.

The OTA mechanism to transmit a file as an example in "esp_mesh_lite_core.h" using "esp_mesh_lite_transmit_file_start()" depends on the URL and is not useful in my case. Any suggestions on how to overcome this will be helpful?

tswen, 2025-02-17:
The function `esp_mesh_lite_transmit_file_start` can fulfill the requirement of transmitting long data between nodes. By registering the relevant callback through `esp_mesh_lite_ota_register_file_transfer_cb`, child nodes can invoke `esp_mesh_lite_transmit_file_start` to request the corresponding file with the specified `file_name` from their parent node.

For further details, please refer to: https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/include/esp_mesh_lite_core.h#L845-L902

mjgci, 2025-02-18:
@tswen Thank you for your response and suggestions. The File Transfer mechanism using the APIs could have been useful however, there will be extra steps to modify OTA elements in the library as per my requirement. Understanding the library is taking time and I will keep testing if the suggestions help. 

The ESP-NOW API is quite helpful. The only caveat to this is the size of bytes transferred in one transaction. Following this helped : [https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide.md#espnow-usage-guide](url)

tswen, 2025-04-15:
@nopnop2002 mesh lite 最新的 master 分支已经支持 29185cc323d199e609d1ff75fe21e5dab95c4e34

closed 2025-06-06 by mjgci