espressif/esp-mesh-lite #151: After Root OTA update, child not update

ruslanslobozhenuk, 2025-03-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Mesh-Lite OTA update

### Actual behavior (suspected bug)

Last version, after ROOT node HTTPS OTA update, child nodes not update!
This error message on all child nodes

E (51737) [ESP_Mesh_Lite_Comm]: Failed to duplicate JSON payload! 

### Error logs or terminal output

```shell
E (51737) [ESP_Mesh_Lite_Comm]: Failed to duplicate JSON payload!
```

### Steps to reproduce the behavior

 OTA update

### Project release version

last version mesh-lite, IDF 5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Windows 11

### Shell

ZSH

tswen, 2025-03-07:
Hello, thank you for pointing out the issue. We will fix this bug as soon as possible. In the meantime, you can revert to a version prior to the commit 0b8918b28f9bb6a8ccf2f676c2005e5b0d30cdf6

tswen, 2025-03-18:
The release of version 1.0.2 has not been completed yet.

tswen, 2025-04-11:
@ruslanslobozhenuk Hi, this issue has been fixed in the master branch. https://github.com/espressif/esp-mesh-lite/commit/62655031e5dc527d35ebb2c51389f100208991cd