espressif/esp-mesh-lite #152: MQTT/tcp client instability - seen after using with Mesh_lite (IDFGH-14830)

BR-Coding-cmd, 2025-03-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/5.3

### Espressif SoC revision.

esp32s3 rev0.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32_S3_WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

The firmware I'm running is a merge of the mesh Lite "local control" and MQTT v3.1.1 via TCP examples.

What I'm expecting is the client to connect when called, maintain the ping keep alive to the broker and not disconnect unless there is an event issue/problem.

I should see a structure echoing the nodes within the mesh network seen in MQTT topics done through a freeRTOS task

### What is the actual behavior?

What I'm seeing is the following;
- repeated attempt to connect to the broker whilst a connection attempt is already occurring. 
- when a failed connection is logged via the ESP_LOGI the error comes back as "success"
- repeated warning print through ESP_LOG of " W (221282) wifi:Haven't to connect to a suitable AP now!"

### Steps to reproduce.

1. open ESP_MESH_LITE example "local_control"
2. open MQTT/TCP example from ESP_IDF release/v5.3
3. merge the two main files, accounting for the set up calls.
3.1. the merge should look something like the c file attached.
[mqtt_mesh-lite_merge_issue.txt](https://github.com/user-attachments/files/19182452/mqtt_mesh-lite_merge_issue.txt)
4. run after setting up mosquito broker.
5. point project to broker
6. observe logs...

### Debug Logs.

```plain
Occasionally the Wifi logw line comes through and repeats for hours on end until power reset. This Error will obviously stop the ESP from reaching out to external controllers - something  that I'm wanting to avoid.

Upon boot the MQTT will fail it's first connection call then more often than not be successful on the second. I've been unable to find the cause to this by delaying function calls until a flag is met to allow for ample time and to avoid blocking.
```

### More Information.

I'm seeing this in my own file also which is a more complex project than the examples. My thinking was it was due to what I added into the project around these libraries. I'm curious to know what there is in ways of rectifying the issue seen.

nopnop2002, 2025-03-18:
You don't need the mqtt_tree task.

Just replace the tcp_client_write_task.

```
void tcp_client_write_task(void *arg)
{
    esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.uri = CONFIG_BROKER_URL,
    };

    esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);
    /* The last argument may be used to pass data to the event handler, in this example mqtt_event_handler */
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
    esp_mqtt_client_start(client);

    while(1) {
        vTaskDelay(1);
    }

    vTaskDelete(NULL);
}
```

BR-Coding-cmd, 2025-03-19:
@nopnop2002 

Thanks for the reply. You are correct, the TCP client functions in the code were not needed (nor really used too).

nopnop2002, 2025-03-19:
@BR-Coding-cmd 

What mqtt broker are you using?

BR-Coding-cmd, 2025-03-19:
@nopnop2002 

I'm using a mosquito broker through docker

nopnop2002, 2025-03-19:
I also use a local broker.

The distance between WiFi and the root node is 3M.

It's very stable.

BR-Coding-cmd, 2025-03-19:
I think it be worth noting, that I'm using a similar style of structure to essentially send out data from a sensor over Mesh_lite back to the MQTT broker. 

After testing over night, I come back to the mesh still being active (seen in logs) but then no longer connected to the broker, which has not gone down over the night.

nopnop2002, 2025-03-19:
>I (4992373) mqtt_client: MQTT_EVENT_DISCONNECTED
>I (23186) mqtt_client: MQTT_EVENT_DISCONNECTED

Both indicate that you have been disconnected from the MQTT broker.
There is no subsequent logging of a connect.

nopnop2002, 2025-03-21:
>Could this now not be a problem with the integration of the mesh_lite library but the MQTT/TCP?

Yes.
This is due to the MQTT transport.

BR-Coding-cmd, 2025-03-21:
I'm using is the following structure "mqtt://[USERNAME]:[PASSWORD]@[HOST]:[PORT]" for the URL param within the client struct, could this be mistaken as MQTTS? 

I've combed through the MQTT client struct, and compared to what was set up in the mqtt_app_start.

`esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.uri = MQTTHost,
        .credentials.client_id = client_id,
    };
`
Within "mqtt_cfg -> Broker -> address -> transport" there is the obvious definitions of TCP, SSL, WS & WSS. By defining the TCP transport layer within the client setup, the following error and warning was provided.

> W (1569) mqtt_client: Transport config set, but overridden by scheme from URI: transport = 1, uri scheme = mqtt
> E (1577) mqtt_client: Client was not initialized

which I thought was down to not using "TCP://" in the URL. Upon making the change the following was logged.

> W (1571) mqtt_client: Transport config set, but overridden by scheme from URI: transport = 1, uri scheme = tcp
> E (1579) mqtt_client: Client was not initialized

removing the transport config but retaining the tcp:// in the URL the following is provided

> E (1572) mqtt_client: Not support this mqtt scheme tcp
> E (1578) mqtt_client: Failed to create transport list

A more complete log of what can be seen when using `esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.uri = "mqtt://[USERNAME]:[PASSWORD]@[HOST]:[PORT]",
        .credentials.client_id = client_id,
    };
` is below; (appologies for the long comment)

> E (11572) esp-tls: [sock=58] select() timeout
> E (11579) transport_base: Failed to open a new connection: 32774
> E (11585) mqtt_client: Error transport connect
> I (11590) mesh_mqtt: MQTT_EVENT_ERROR
> E (11594) mesh_mqtt: Last error reported from esp-tls: 0x8006
> I (11601) mesh_mqtt: Last errno string (Success)
> I (11606) mesh_mqtt: MQTT_EVENT_DISCONNECTED
> I (21611) mesh_mqtt: Other event id:7
> I (21753) mesh_mqtt: MQTT_EVENT_CONNECTED
> I (21771) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=59041
> I (21781) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=4767
> I (21784) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=41436
> I (21788) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=21564
> I (21791) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=846
> I (21796) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=33695
> I (21802) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=61132
> I (21809) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=14572
> I (21815) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=14883
> I (21821) mesh_mqtt: MQTT_EVENT_DATA
> I (21825) mesh_mqtt: MQTT_EVENT_DATA
> I (21829) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=43035
> I (21835) mesh_mqtt: MQTT_EVENT_DATA
> I (21842) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=6800
> W (22569) [vendor_ie]: [Unexpected restart]: the rtc data: temp_mesh_id:156, level:1, ssid:[SSID], bssid 00:00:00:00:00:00
> W (22570) [vendor_ie]: Try to connect the original parent node
> W (22577) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
> I (22588) wifi:state: run -> init (0x0)
> I (22590) wifi:pm stop, total sleep time: 0 us / 14985300 us
> 
> I (22595) wifi:new:<6,0>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> E (22605) [ESP_Mesh_Lite_Comm]: Disconnect reason : 8
> I (22609) [vendor_ie]: esp_wifi_connect return ESP_OK
> E (22610) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 58
> E (22624) mqtt_client: Poll read error: 119, aborting connection
> I (22631) mesh_mqtt: MQTT_EVENT_DISCONNECTED
> I (22683) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> I (22685) wifi:state: init -> auth (0xb0)
> I (22726) wifi:state: auth -> assoc (0x0)
> I (22745) wifi:state: assoc -> run (0x10)
> I (22833) wifi:connected with [SSID], aid = 4, channel 6, BW20, bssid = [BSSID]
> I (22834) wifi:security: WPA2-PSK, phy: bgn, rssi: -58
> I (22836) wifi:pm start, type: 1
> 
> I (22839) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
> I (22888) wifi:AP's beacon interval = 102400 us, DTIM period = 1
> I (23849) esp_netif_handlers: sta ip: [MYIP], mask: 255.255.248.0, gw: []
> I (23850) bridge_wifi: Connected with IP Address:[MYIP]
> I (23854) [vendor_ie]: RTC store: temp_mesh_id:156; ssid:[SSID]; bssid:00:00:00:00:00:00; crc:771362977
> I (28436) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> I (28437) wifi:station: 10:52:1c:75:49:38 join, AID=1, bgn, 40D
> I (28457) bridge_wifi: STA Connecting to the AP again...
> I (28489) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.5.2
> I (32635) mesh_mqtt: Other event id:7
> I (32655) mesh_mqtt: MQTT_EVENT_CONNECTED
> I (32669) mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=7989
> I (32674) mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=12746

nopnop2002, 2025-03-21:
TCP transport
mqtt://[HOST]:[PORT]

 SSL/TLS transport
mqtts://[HOST]:[PORT]

WS transport
ws://[HOST]:[PORT]

WSS transport
wss://[HOST]:[PORT]

BR-Coding-cmd, 2025-03-21:
Thank you for spending the time on this. I've corrected the syntax to your cfg to match the tcp statement. I'm not planning on using ws/wss so have omitted it.

Okay, so using the corrected syntax 

> E (31638) esp-tls: [sock=58] select() timeout
> E (31639) transport_base: Failed to open a new connection: 32774
> E (31639) mqtt_client: Error transport connect
> I (31644) iMesh_mqtt: MQTT_EVENT_ERROR
> E (31648) iMesh_mqtt: Last error reported from esp-tls: 0x8006
> I (31655) iMesh_mqtt: Last errno string (Success)
> I (31660) iMesh_mqtt: MQTT_EVENT_DISCONNECTED

The above is still seen upon initial start. The below happens immediately after the connection. Note the "E (42692) connection abort" is the same socket number as the esp-tls error above

> W (42651) [vendor_ie]: [Unexpected restart]: the rtc data: temp_mesh_id:156, level:1, ssid:[SSID], bssid 00:00:00:00:00:00
> W (42652) [vendor_ie]: Try to connect the original parent node
> W (42659) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
> I (42669) wifi:state: run -> init (0x0)
> I (42672) wifi:pm stop, total sleep time: 0 us / 17974714 us
> 
> I (42677) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> E (42687) [ESP_Mesh_Lite_Comm]: Disconnect reason : 8
> E (42692) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 58
> I (43047) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> I (43048) wifi:state: init -> auth (0xb0)
> E (43051) mqtt_client: Poll read error: 119, aborting connection
> I (43053) wifi:state: auth -> assoc (0x0)
> I (42691) [vendor_ie]: esp_wifi_connect return ESP_OK
> I (43064) wifi:state: assoc -> run (0x10)
> I (43060) Mesh_mqtt: MQTT_EVENT_DISCONNECTED
> I (43095) wifi:connected with [SSID], aid = 2, channel 6, BW20, bssid = [BSSID]
> I (43095) wifi:security: WPA2-PSK, phy: bgn, rssi: -65
> I (43097) wifi:pm start, type: 1

following these errors the client connects up to the broker and will continue publishing. I expect the disconnect error to pop through in a few hours time at which point I will capture and update this thread.

Note: I've attempted to capture the packets/traffic with wireshark but have been unsuccessful

nopnop2002, 2025-03-21:
Please tell us the uri you are using.
```
.broker.address.uri = uri,
```

BR-Coding-cmd, 2025-03-21:
```
#define Setting_MQTTHost "192.168.1.42"
#define Setting_PORT_MQTT "1883"
#define Setting_USR_MQTT "username"
#define Setting_PSWD_MQTT "password"

sprintf(URL,"mqtt://%s:%s", Setting_MQTTHost, Setting_PORT_MQTT);

esp_mqtt_client_config_t mqtt_cfg = {
                // format of "mqtt://IP:port"
        .broker.address.uri = URL, // 
        #if CONFIG_MQTT_TRANSPORT_OVER_SSL
		// .broker.verification.certificate = (const char *)root_cert_pem_start,
        #endif
        .credentials = {
            .client_id = client_id,
            .username = Setting_USR_MQTT,
            .authentication.password = Setting_PSWD_MQTT,
        },
        .session.protocol_ver = MQTT_PROTOCOL_V_3_1_1,
    };
```

As far as I'm aware I've not placed anything in the SDK or defined it in my file to use the SSL port..

As to why it's using the SSL port instead of the TCP i'm not sure

nopnop2002, 2025-03-21:
Try using this as the uri.
No username or password is required.
```
uri=[mqtt://broker.emqx.io:1883]
```

BR-Coding-cmd, 2025-03-21:
I'm getting the logs through but cannot see the topics on the broker via mqtt explorer. 

The logs on start up are below;

> I (43053) Mesh_mqtt: MQTT_EVENT_CONNECTED
> I (43182) Mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=42388
> I (43184) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=39379
> I (43226) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=58460
> W (43628) [vendor_ie]: [Unexpected restart]: the rtc data: temp_mesh_id:156, level:1, ssid:[SSID], bssid 00:00:00:00:00:00
> W (43629) [vendor_ie]: Try to connect the original parent node
> W (43636) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
> I (43646) wifi:state: run -> init (0x0)
> I (43649) wifi:pm stop, total sleep time: 0 us / 18971660 us
> 
> I (43654) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> E (43664) [ESP_Mesh_Lite_Comm]: Disconnect reason : 8
> I (43668) [vendor_ie]: esp_wifi_connect return ESP_OK
> I (44021) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
> I (44022) wifi:state: init -> auth (0xb0)
> E (43669) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 58
> I (44027) wifi:state: auth -> assoc (0x0)
> E (44035) mqtt_client: Poll read error: 119, aborting connection
> I (44040) wifi:state: assoc -> run (0x10)
> I (44046) Mesh_mqtt: MQTT_EVENT_DISCONNECTED
> I (44060) wifi:connected with [SSID], aid = 2, channel 6, BW20, bssid = 10:27:f5:68:9a:be
> I (44060) wifi:security: WPA2-PSK, phy: bgn, rssi: -59
> I (44063) wifi:pm start, type: 1
> 
> I (44066) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
> I (44075) wifi:AP's beacon interval = 102400 us, DTIM period = 1
> I (45584) esp_netif_handlers: sta ip: [IP], mask: 255.255.248.0, gw: 10.0.0.2
> I (45584) bridge_wifi: Connected with IP Address:[IP]
> I (45588) [vendor_ie]: RTC store: temp_mesh_id:156; ssid:[SSID]; bssid:00:00:00:00:00:00; crc:771362977
> I (54051) Mesh_mqtt: Other event id:7
> I (54354) Mesh_mqtt: MQTT_EVENT_CONNECTED
> I (54470) Mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=50352
> I (54492) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=14137
> I (54544) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=2213
> I (58503) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=40741

nopnop2002, 2025-03-21:
Good!!

There is no esp-tls error message.

BR-Coding-cmd, 2025-03-21:
apologies, I missed this from just above the "I (43053) Mesh_mqtt: MQTT_EVENT_CONNECTED"

> E (25627) esp-tls: couldn't get hostname for :broker.emqx.io: getaddrinfo() returns 202, addrinfo=0x0
> E (25627) transport_base: Failed to open a new connection: 32769
> E (25633) mqtt_client: Error transport connect
> I (25638) Mesh_mqtt: MQTT_EVENT_ERROR
> E (25642) Mesh_mqtt: Last error reported from esp-tls: 0x8001
> I (25649) Mesh_mqtt: Last errno string (Success)
> I (25654) Mesh_mqtt: MQTT_EVENT_DISCONNECTED

nopnop2002, 2025-03-21:
Please erase-flash and build again.

```
esp-idf erase-flash
esp-idf build
``

BR-Coding-cmd, 2025-03-21:
upon every flash I run

idf.py erase-flash flash monitor -p COMX

nopnop2002, 2025-03-21:
I'll give up.

BR-Coding-cmd, 2025-03-21:
My esp-idf version is v5.3.2

nopnop2002, 2025-03-21:
Your logging
```
I (7614) wifi:state: init -> auth (0xb0)
I (7621) wifi:state: auth -> init (0x8a0)
I (7622) wifi:new:<6,0>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
E (7629) [ESP_Mesh_Lite_Comm]: Disconnect reason : 202
E (8618) esp-tls: couldn't get hostname for :broker.emqx.io: getaddrinfo() returns 202, addrinfo=0x0
E (8618) transport_base: Failed to open a new connection: 32769
E (8624) mqtt_client: Error transport connect
I (8629) mesh_mqtt: MQTT_EVENT_ERROR
E (8633) mesh_mqtt: Last error reported from esp-tls: 0x8001
I (8639) mesh_mqtt: Last errno string (Success)
I (8645) mesh_mqtt: MQTT_EVENT_DISCONNECTED
W (9618) wifi:Haven't to connect to a suitable AP now!
I (11422) [vendor_ie]: esp_wifi_connect return ESP_OK
I (11423) [ESP_Mesh_Lite_Comm]: Retry to connect to the AP
W (11485) wifi:Haven't to connect to a suitable AP now!
```

My logging
```
I (4200) wifi:state: init -> auth (0xb0)
I (4206) wifi:state: auth -> assoc (0x0)
I (4242) wifi:Association refused temporarily time 1024, comeback time 1124 (TUs)
I (5393) wifi:state: assoc -> assoc (0x0)
I (6395) wifi:state: assoc -> init (0x2700)
I (6396) wifi:new:<11,0>, old:<11,0>, ap:<11,2>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
E (6399) [ESP_Mesh_Lite_Comm]: Disconnect reason : 39
I (7352) [vendor_ie]: esp_wifi_connect return ESP_OK
I (7353) [ESP_Mesh_Lite_Comm]: Retry to connect to the AP
E (10194) [ESP_Mesh_Lite_Comm]: Disconnect reason : 205
W (11322) wifi:Haven't to connect to a suitable AP now!
```

BR-Coding-cmd, 2025-03-21:
The difference there would now point to a WiFi config difference then perhaps? We're using the same files so I'm doubting it's the case.

nopnop2002, 2025-03-21:
Got it.
It's trying to connect to MQTT without waiting for the WiFi connection to be completed.

nopnop2002, 2025-03-21:
```
void app_main()
{
    /**
     * @brief Set the log level for serial port printing.
     */
    esp_log_level_set("*", ESP_LOG_INFO);

    ESP_ERROR_CHECK(esp_storage_init());

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    ESP_LOGI(TAG, "wifi_init");
    wifi_init();

    ESP_LOGI(TAG, "esp_mesh_lite_init");
    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);

    ESP_LOGI(TAG, "app_wifi_set_softap_info");
    app_wifi_set_softap_info();

    ESP_LOGI(TAG, "esp_mesh_lite_start");
    esp_mesh_lite_start();

    /**
     * @breif Create handler
     */
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_sta_got_ip_handler, NULL, NULL));

    TimerHandle_t timer = xTimerCreate("print_system_info", 10000 / portTICK_PERIOD_MS,
                                       true, NULL, print_system_info_timercb);
    xTimerStart(timer, 0);
}
```

BR-Coding-cmd, 2025-03-21:
I suppose it'll be called from main the same way as before

`ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_sta_got_ip_handler, NULL, NULL));`

nopnop2002, 2025-03-21:
Please check the timing when tcp_client_write_task is started.
static void ip_event_sta_got_ip_handler(void *arg, esp_event_base_t event_base,
                                        int32_t event_id, void *event_data)
{
    static bool tcp_task = false;

    if (!tcp_task) {
        esp_logw(TAG, "Start tcp_client_write_task");
        xTaskCreate(tcp_client_write_task, "tcp_client_write_task", 4 * 1024, NULL, 5, NULL);
        tcp_task = true;
    }
}

nopnop2002, 2025-03-21:
```
I (33128) Mesh_mqtt: MQTT_EVENT_CONNECTED
I (33238) Mesh_mqtt: MQTT_EVENT_SUBSCRIBED, msg_id=52183
I (33266) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=55572
I (33287) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=7031
```

That's a different code than mine.

```
    case MQTT_EVENT_CONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        xEventGroupSetBits(mqtt_status_event_group, MQTT_CONNECTED_BIT);
        break;
```

BR-Coding-cmd, 2025-03-21:
Apologies, see the below for the logs

```
I (1376) [vendor_ie]: Mesh ID: 101
I (1378) Mesh-Lite: esp-mesh-lite component version: 1.0.1
I (1385) ESPNOW: espnow [version: 1.0] init
I (1388) [vendor_ie]: Mesh-Lite commit id: 2660e63

I (1395) [vendor_ie]: Mesh ID: 101
W (1399) [vendor_ie]: Error Get[4354]
W (1403) [vendor_ie]: Error Get[4354]
W (1407) wifi:Haven't to connect to a suitable AP now!
I (1413) [ESP_Mesh_Lite_Comm]: msg action add success
I (1419) [ESP_Mesh_Lite_Comm]: Bind Socket 54, port 6364
I (1424) [ESP_Mesh_Lite_Comm]: Bind Socket 55, port 6363
I (1430) [ESP_Mesh_Lite_Comm]: Bind Socket 56, port 6366
I (1436) [ESP_Mesh_Lite_Comm]: Bind Socket 57, port 6365
I (1443) [mesh-lite-espnow]: Start espnow task
I (1449) [vendor_ie]: Mesh ID: 101
I (1452) Mesh-Lite: Mesh-Lite connecting
I (1458) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (1466) main_task: Returned from app_main()
I (2453) [vendor_ie]: Mesh-Lite Scan done
I (2454) [vendor_ie]: RTC store: temp_mesh_id:255; ssid:[SSID]; bssid:00:00:00:00:00:00; crc:2952505822
I (2458) [vendor_ie]: wifi_cfg [SSID] router_config [SSID]
I (2466) [vendor_ie]: esp_wifi_connect return ESP_OK
I (3577) wifi:ap channel adjust o:1,1 n:6,2
I (3577) wifi:new:<6,0>, old:<1,1>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (3578) wifi:state: init -> auth (0xb0)
I (3585) wifi:state: auth -> init (0x8a0)
I (3585) wifi:new:<6,0>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
E (3593) [ESP_Mesh_Lite_Comm]: Disconnect reason : 202
I (7414) [vendor_ie]: esp_wifi_connect return ESP_OK
I (7414) [ESP_Mesh_Lite_Comm]: Retry to connect to the AP
I (7686) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (7687) wifi:state: init -> auth (0xb0)
I (8688) wifi:state: auth -> init (0x200)
I (8689) wifi:new:<6,0>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
E (8691) [ESP_Mesh_Lite_Comm]: Disconnect reason : 2
I (12414) [vendor_ie]: esp_mesh_lite_wifi_scan_start return ESP_OK
I (15249) [vendor_ie]: Mesh-Lite Scan done
I (15250) [vendor_ie]: RTC store: temp_mesh_id:255; ssid:[SSID]; bssid:00:00:00:00:00:00; crc:2952505822
I (15254) [vendor_ie]: wifi_cfg [SSID] router_config [SSID]
I (15262) [vendor_ie]: esp_wifi_connect return ESP_OK
I (15615) wifi:new:<6,2>, old:<6,0>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (15616) wifi:state: init -> auth (0xb0)
I (15621) wifi:state: auth -> assoc (0x0)
I (15647) wifi:state: assoc -> run (0x10)
I (15666) wifi:connected with [SSID], aid = 2, channel 6, BW20, bssid = 10:27:f5:68:9a:be
I (15666) wifi:security: WPA2-PSK, phy: bgn, rssi: -57
I (15668) wifi:pm start, type: 1

I (15671) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (15679) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (15705) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (16690) esp_netif_handlers: sta ip: [IP], mask: 255.255.248.0, gw: 10.0.0.2
I (16691) bridge_wifi: Connected with IP Address:[IP]
I (16695) bridge_common: [WIFI_AP_DEF ]Name Server1: [SERVER_IP]
I (16701) [vendor_ie]: RTC store: temp_mesh_id:156; ssid:[SSID]; bssid:00:00:00:00:00:00; crc:771362977
E (16712) [ESP_Mesh_Lite_Comm]: Error Get!
W (16716) [ESP_Mesh_Lite_Comm]: argot is not saved in flash
W (16726) local_control: ip_event_sta_got_ip_handler
W (16728) local_control: Start tcp_client_write_task
W (16734) Mesh_mqtt: tcp_client_write_task start
I (16739) Mesh_mqtt: Wait for connection to the MQTT broker
I (16740) Mesh_mqtt: Other event id:7
I (21141) Mesh_mqtt: MQTT_EVENT_CONNECTED
I (21142) Mesh_mqtt: Connected to MQTT Broker
I (21144) Mesh_mqtt: sent publish successful, msg_id=32230
I (21169) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=32230
I (22149) Mesh_mqtt: sent publish successful, msg_id=28469
I (22273) Mesh_mqtt: MQTT_EVENT_PUBLISHED, msg_id=28469
```

BR-Coding-cmd, 2025-03-24:
Update on connection performance using the below config for MQTT client with ESP_MESH_LITE library over MQTT v3.1.1 TCP

```
#define Setting_MQTTHost "broker.emqx.io"
#define Setting_PORT_MQTT "1883"

void func(void){
    uint8_t mac[6] = {0};
    char mac_str[MAC_MAX_LEN];
    int i = 0;
    int j = 0;
    esp_wifi_get_mac(WIFI_IF_STA, mac);
    snprintf(mac_str, sizeof(mac_str), MACSTR, MAC2STR(mac));
    while (mac_str[i] != '\0') {
        if (mac_str[i] != ':') {
            my_mac_str[j++] = mac_str[i];
        }
        i++;
    }
    my_mac_str[j] = '\0';
}    

sprintf(URL,"mqtt://%s:%s", Setting_MQTTHost, Setting_PORT_MQTT);
sprintf(client_id, "Node_%s",my_mac_str);

esp_mqtt_client_config_t mqtt_cfg = {
        // format of "mqtt://IP:port"
        .broker.address.uri = URL,
        #if CONFIG_MQTT_TRANSPORT_OVER_SSL
		// .broker.verification.certificate = (const char *)root_cert_pem_start,
        #endif
        .credentials = {
            .client_id = client_id,
            // .username = Setting_USR_MQTT,
            // .authentication.password = Setting_PSWD_MQTT,
        },
    };
```

- when using the public broker I saw no disconnection from it at all over >3days of activity with constant publishing every 5 seconds
- when using my own local broker with a username and password I saw the below error message from the log within idf.py monitor after maybe only a handful of hours. 
NOTE: The altered config params can be seen above the error message

```
#define Setting_MQTTHost "192.168.1.42"
#define Setting_PORT_MQTT "1883"
#define Setting_USR_MQTT "username"
#define Setting_PSWD_MQTT "password"

....

esp_mqtt_client_config_t mqtt_cfg = {
        // format of "mqtt://IP:port"
        .broker.address.uri = URL,
        #if CONFIG_MQTT_TRANSPORT_OVER_SSL
		// .broker.verification.certificate = (const char *)root_cert_pem_start,
        #endif
        .credentials = {
            .client_id = client_id,
            .username = Setting_USR_MQTT,
            .authentication.password = Setting_PSWD_MQTT,
        },
    };

```

```
E (243631655) esp-tls: [sock=58] select() timeout
E (243631656) transport_base: Failed to open a new connection: 32774
E (243631656) mqtt_client: Error transport connect
I (243631662) iMesh_mqtt: MQTT_EVENT_ERROR
E (243631667) iMesh_mqtt: Last error reported from esp-tls: 0x8006
I (243631673) iMesh_mqtt: Last errno string (Success)
I (243631679) iMesh_mqtt: MQTT_EVENT_DISCONNECTED
```

- it is my understanding now that there is an issue either within how I've set up my mosquito broker or that there is something in the back end of the ESP_MQTT lib

Any comments welcome!

nopnop2002, 2025-03-24:
>when using the public broker I saw no disconnection from it at all over >3days of activity with constant publishing every 5 seconds

This means that there are no issues with the backend and network of the ESP_MQTT library.

There seems to be a problem with the mosquito broker configuration.

BR-Coding-cmd, 2025-03-24:
Ah that's comforting. I'll have a look into the broker! Will close the issue as resolved once I've found the issue my end so others can be aware 

thanks for the help on this @nopnop2002 !

BR-Coding-cmd, 2025-03-26:
So, I've done some further digging following the recent findings - It's not a broker issue it's a WiFi config issue; what it is I'm still trying to understand.

Test broker1 has been a mosquito broker via docker

Test broker2 is a stable Vertex Broker ran on a ubuntu VM

- Using the TCP example from my IDF version, release v5.3, I can connect to my broker.
- Using a Merged version of TCP and esp_mesh_lite/local_control I can connect - I will eventually disconnect and also see that multiple downstream nodes disconnect from the broker but still be connected within the mesh.
    - I also see no connection to my vertex broker if it has a username and password set. My DHCP server on the AP I'm using shows that the node has IP and the master relays saying that the child nodes have vaild mesh DHCP IPs - However, the Broker shows that the Master/child node(s) don't attempt to connect to it. No requests are seen.

Difference between the two;
- mqtt/TCP example calls "example_connect()" and can connect to test broker 1&2 no problem and uses the full URI string to pass in Username and Password : "mqtt://Username:Password@Host:Port" port being 1883 because it's a TCP connection.
- esp_mesh_lite/local_control example sets up wifi via wifi_config_t with ssid and password then sets up "app_wifi_set_softap_info()"

When using "example_connect()" there are no esp-TLS issues regarding ports not being made, using the mesh_lite/local_control version of setting up the WiFi config (seen below) on the other hand does and the "mqtt handler" spits out the error seen prior (see quoted section below)

    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    wifi_init();

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);

    app_wifi_set_softap_info();

> E (243631655) esp-tls: [sock=58] select() timeout
> E (243631656) transport_base: Failed to open a new connection: 32774
> E (243631656) mqtt_client: Error transport connect
> I (243631662) Mesh_mqtt: MQTT_EVENT_ERROR
> E (243631667) Mesh_mqtt: Last error reported from esp-tls: 0x8006
> I (243631673) Mesh_mqtt: Last errno string (Success)
> I (243631679) Mesh_mqtt: MQTT_EVENT_DISCONNECTED

The connection to the broker, via the mqtt/tcp+mesh_lite, whilst has been seen to work prior, may have an issue passing through or managing the socket. I've gone down to the mqtt_client source files and see that the code will operate as expected (as seen when using example_connect)

BR-Coding-cmd, 2025-03-26:
See the attached txt for how i'm testing currently. Bare in mind the issue here is that I'm trying to use a username and password for this test. When parsing the username and password to the broker via the URI and wifi is config'ed via "example_connect()" it works - using the mesh_lite wifi_config method, does not

[tcp-mesh_lite-issue.txt](https://github.com/user-attachments/files/19463841/tcp-mesh_lite-issue.txt)

nopnop2002, 2025-03-26:
This code only sets the local my_mac_str, it does not set the global my_mac_str.
```
char my_mac_str[MAC_MAX_LEN];

void sign_MyMAC(void){
    uint8_t mac[6] = {0};
    char mac_str[MAC_MAX_LEN];
    int i = 0;
    int j = 0;
    esp_wifi_get_mac(WIFI_IF_STA, mac);
    snprintf(mac_str, sizeof(mac_str), MACSTR, MAC2STR(mac));
    while (mac_str[i] != '\0') {
        if (mac_str[i] != ':') {
            my_mac_str[j++] = mac_str[i];
        }
        i++;
    }
    my_mac_str[j] = '\0';
}
```

nopnop2002, 2025-03-26:
Here is a sample code for an MQTT client for Mesh-Lite.

https://github.com/nopnop2002/esp-mesh-lite-examples/tree/main/mesh_mqtt_client

You can specify the username and password in the menuconfig.

[image]

BR-Coding-cmd, 2025-03-26:
```
(0x8006): mbedtls parse certificates was partly successful
```

The above error code is what I'm still seeing when I place your wifi_init() functions and your suggested mqtt_cfg for TCP. 

I've the below mqtt_cfg's

```
sprintf(URL, "mqtt://%s:%s@%s:%s", Setting_USR_MQTT, Setting_PSWD_MQTT, Setting_MQTTHost, Setting_PORT_MQTT);
esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.uri = URL,
        .credentials.client_id = client_id,
        .session.protocol_ver = MQTT_PROTOCOL_V_3_1_1,
    };
```

```
esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.hostname = Setting_MQTTHost,
        .broker.address.port = atoi(Setting_PORT_MQTT),
        .credentials.client_id = client_id,
        .session.protocol_ver = MQTT_PROTOCOL_V_3_1_1,
        .broker.address.transport = MQTT_TRANSPORT_OVER_TCP,
    };
```

```
 sprintf(URL, "mqtt://%s:%s", Setting_MQTTHost, Setting_PORT_MQTT);
    
    esp_mqtt_client_config_t mqtt_cfg = {
        .broker.address.uri = URL,
        .credentials = {
            .username = Setting_USR_MQTT,
            .authentication.password = Setting_PSWD_MQTT,
         },
        .credentials.client_id = client_id,
        .session.protocol_ver = MQTT_PROTOCOL_V_3_1_1,
    };
```

The WiFi_cfg is the same as yours with the wifi_config looking like - along with the other functions.
```
wifi_config_t wifi_config = {
		.sta = {
			.ssid = CONFIG_ROUTER_SSID,
			.password = CONFIG_ROUTER_PASSWORD,
		},
	};
```

I'm going to pull your example and try building it to see if there is any difference in outcome

BR-Coding-cmd, 2025-03-26:
Hi,

I've pulled the repo and tested it on my end, the logs are attached

[Log_printout_no_auth 260325.txt](https://github.com/user-attachments/files/19469303/Log_printout_no_auth.260325.txt)

I'm sad to say that I wasn't able to get that to connect up to my Vertx broker. I was able to connect to the Mosquito broker with no Auth but not the Vertx one. I can however connect to the Vertx broker via MQTT Explorer and also other clients both with and without authentication...

I'm honestly stuck

nopnop2002, 2025-03-26:
Change the MQTT transport to SSL/TLS and try connecting to the Vertx broker.

BR-Coding-cmd, 2025-03-27:
My Broker isn't set up to operate with a TLS/SSL connection. So I won't be able to test for the time being. My Vertx Broker is set up to operate on a TCP connection with username and password. Why this is throwing an error when used with mesh_lite I'm not sure but without Mesh_lite enabled I don't get the below error which is what I'm seeing when connecting via TCP.

```
E (15474) esp-tls: [sock=58] select() timeout
E (15475) esp-tls: Failed to open new connection
E (15475) transport_base: Failed to open a new connection
E (15480) mqtt_client: Error transport connect
I (15485) PUB: MQTT_EVENT_ERROR
E (15488) PUB: Last error reported from esp-tls: 0x8006
I (15494) PUB: Last errno string (Success)
I (15499) PUB: MQTT_EVENT_DISCONNECTED
```

I see no reason for my code to even use TLS if the mqtt config is set to TCP and isn't passing through any cert

nopnop2002, 2025-03-27:
Is it correct that the port number used when connecting to WiFi via "example_connect()" is 1883?

BR-Coding-cmd, 2025-03-27:
yes and the Logs below

```
I (432) main_task: Calling app_main()
I (432) mqtt_example: [APP] Startup..
I (432) mqtt_example: [APP] Free memory: 330992 bytes
I (432) mqtt_example: [APP] IDF version: -128-NOTFOUND
I (452) example_connect: Start example_connect.
I (452) pp: pp rom version: e7ae62f
I (452) net80211: net80211 rom version: e7ae62f
I (472) wifi:wifi driver task: 3fcac738, prio:23, stack:6656, core=0
I (472) wifi:wifi firmware version: 9b6e5e2dd
I (472) wifi:wifi certification version: v7.0
I (472) wifi:config NVS flash: enabled
I (472) wifi:config nano formating: disabled
I (482) wifi:Init data frame dynamic rx buffer num: 32
I (482) wifi:Init static rx mgmt buffer num: 5
I (492) wifi:Init management short buffer num: 32
I (492) wifi:Init dynamic tx buffer num: 32
I (492) wifi:Init static tx FG buffer num: 2
I (502) wifi:Init static rx buffer size: 1600
I (502) wifi:Init static rx buffer num: 10
I (512) wifi:Init dynamic rx buffer num: 32
I (512) wifi_init: rx ba win: 6
I (512) wifi_init: accept mbox: 6
I (522) wifi_init: tcpip mbox: 32
I (522) wifi_init: udp mbox: 6
I (522) wifi_init: tcp mbox: 6
I (532) wifi_init: tcp tx win: 5760
I (532) wifi_init: tcp rx win: 5760
I (542) wifi_init: tcp mss: 1440
I (542) wifi_init: WiFi IRAM OP enabled
I (552) wifi_init: WiFi RX IRAM OP enabled
I (552) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
W (562) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
W (602) phy_init: saving new calibration data because of checksum failure, mode(2)
I (612) wifi:mode : sta (cc:ba:97:08:ee:ec)
I (612) wifi:enable tsf
I (622) example_connect: Connecting to [SSID]...
I (622) example_connect: Waiting for IP(s)
I (3502) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (3502) wifi:state: init -> auth (0xb0)
I (3522) wifi:state: auth -> assoc (0x0)
I (3522) wifi:state: assoc -> run (0x10)
I (3562) wifi:connected with [SSID], aid = 2, channel 6, BW20, bssid = 0c:9d:92:52:ab:c8
I (3562) wifi:security: WPA2-PSK, phy: bg, rssi: -50
I (3562) wifi:pm start, type: 1

I (3572) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3582) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (3632) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (5582) esp_netif_handlers: example_netif_sta ip: 192.168.2.104, mask: 255.255.128.0, gw: 0.0.0.0
I (5582) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.2.104
I (5582) example_common: Connected to example_netif_sta
I (5592) example_common: - IPv4 address: 192.168.2.104,
I (5602) mqtt_example: Other event id:7
I (5602) main_task: Returned from app_main()
I (5742) mqtt_example: MQTT_EVENT_CONNECTED
I (5742) mqtt_example: sent publish successful, msg_id=45753
I (5742) mqtt_example: sent subscribe successful, msg_id=57807
I (5752) mqtt_example: sent subscribe successful, msg_id=53142
I (5752) mqtt_example: sent unsubscribe successful, msg_id=57015
I (5802) mqtt_example: MQTT_EVENT_PUBLISHED, msg_id=45753
I (5822) mqtt_example: MQTT_EVENT_SUBSCRIBED, msg_id=57807
I (5822) mqtt_example: sent publish successful, msg_id=0
I (5822) mqtt_example: MQTT_EVENT_SUBSCRIBED, msg_id=53142
I (5832) mqtt_example: sent publish successful, msg_id=0
I (5832) mqtt_example: MQTT_EVENT_UNSUBSCRIBED, msg_id=57015
I (5842) mqtt_example: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
I (5862) mqtt_example: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
```

nopnop2002, 2025-03-27:
Thank you.

ESP-MESH-LITE seems to have problems connecting to TCP-MQTT servers that require a password.

BR-Coding-cmd, 2025-03-27:
@nopnop2002 
How would you suggest renaming the issue so that it is more appropriately handled??

nopnop2002, 2025-03-28:
You can edit the title of any issue you create.
Is there an edit button?

[image]

BR-Coding-cmd, 2025-03-28:
yes I can see it

nopnop2002, 2025-03-28:
I followed this tutorial to build an MQTT server that requires a username and password.
https://www.digitalocean.com/community/tutorials/how-to-install-and-secure-the-mosquitto-mqtt-messaging-broker-on-debian-10

Connections without a username and password will be refused.
```
$ mosquitto_sub -h 192.168.10.48 -t test
Connection error: Connection Refused: not authorised.
```
A username and password are required to connect.
```
$ mosquitto_sub -h 192.168.10.48 -t test -u "nop" -P "xxxxx"
Hello World
Hello World
```

I built this project.
https://github.com/nopnop2002/esp-mesh-lite-examples/tree/main/mesh_mqtt_client
I was able to connect this broker using my username and password.
```
I (4018) SUB: CONFIG_MQTT_BROKER=[192.168.10.48]
I (4023) convert_mdns_host: from=[192.168.10.48]
I (4000) PUB: Wait for connection to the MQTT broker
I (4025) PUB: MQTT_EVENT_CONNECTED
I (4029) SUB: ip=[192.168.10.48]
I (4040) PUB: Connected to MQTT Broker
I (4042) SUB: MQTT_TRANSPORT_OVER_TCP
I (4051) SUB: uri=[mqtt://192.168.10.48:1883]
```

I was also able to receive the topic.
```
I (93580) SUB: MQTT_EVENT_DATA
I (93580) SUB: TOPIC=[/topic/mesh/broadcast] DATA=[Hello World]
I (93583) SUB: ulTaskNotifyTake event_id=6
I (93588) SUB: -------------------------------------------------
I (93595) SUB: TOPIC=[/topic/mesh/broadcast]
I (93600) SUB: DATA=[Hello World]
I (93604) SUB: -------------------------------------------------
```

BR-Coding-cmd, 2025-04-01:
> If you add the following code to this example, will it be possible to connect to the Vertx broker from multiple ESP32s at the same time?

Yes I can connect multiple clients to the same broker, with the TCP example alteration

However when connecting via a mesh network I am unable to get any connection. I have tried using the example connect function from TCP as stated previously, however, the mesh_lite protocol cannot operate through this as a DHCPS is not initialised and the device is only set up as a Station and not both 'Station & SoftAP'

BR-Coding-cmd, 2025-04-01:
I'm currently working from this project, built from an unchanged sdkconfig from the local_control example from mesh_lite examples

```
#include <inttypes.h>
#include "esp_log.h"
#include "esp_mac.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"

#include "esp_wifi.h"
#include "nvs_flash.h"
#include <sys/socket.h>

#include "esp_bridge.h"
#include "esp_mesh_lite.h"

static int g_sockfd    = -1;
static const char *TAG = "local_control";

/* --- changes -- */
// event handling
#include "freertos/queue.h"
#include "freertos/event_groups.h"

//WIFI stuff
#define WIFI_SSID "SSID"
#define WIFI_PSWD "PSWD"

//mqtt stuff
#include "mqtt_client.h"
EventGroupHandle_t mqtt_status_event_group;
#define MQTT_CONNECTED_BIT BIT2
#define MQTT_USR "USR"
#define MQTT_PSWD "PSWD"
#define URL "mqtt://192.168.5.201:1883"
char *TAG_MQTT = "Mesh_mqtt";

//mesh stuff
#define MESHID 50

// globals
char my_mac_str[MAC_MAX_LEN];
static bool sys_info = false;
int msg_id;
/* -- end of changes -- */

/* 
    ====================================================
    ================== FUNCTIONS =======================
    ==================================================== 
*/

void sign_MyMAC(void){
    uint8_t mac[6] = {0};
    char mac_str[MAC_MAX_LEN];
    int i = 0;
    int j = 0;
    esp_wifi_get_mac(WIFI_IF_STA, mac);
    snprintf(mac_str, sizeof(mac_str), MACSTR, MAC2STR(mac));
    while (mac_str[i] != '\0') {
        if (mac_str[i] != ':') {
            my_mac_str[j++] = mac_str[i];
        }
        i++;
    }
    my_mac_str[j] = '\0';
}
     
/* -- ESP-CRAP -- */
[... 218 lines cut ...]
    configASSERT( mqtt_status_event_group );
    xEventGroupClearBits(mqtt_status_event_group, MQTT_CONNECTED_BIT);

    esp_storage_init();

    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    esp_bridge_create_all_netif();

    wifi_init();

    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_set_mesh_id(MESHID, true);
    esp_mesh_lite_init(&mesh_lite_config);

    app_wifi_set_softap_info();

    esp_mesh_lite_start();

    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_ip_event_handler, NULL, NULL));

}

```

BR-Coding-cmd, 2025-04-02:
it is worth noting that within the mesh network after connecting up to the MQTT broker via TCP with username&password the connection from the Mesh network to the broker can be seen in oddly in two different forms
- first root then child
- child then root
Both of which happens whilst there is still a vaild mesh network

I'm not sure as to how the packets are sent over the DHPCS from the Root to the Child nodes, however it would seem that there is an intermittent bug within the process which means the data from root to child for "internet access" misses a handshake or is unable to complete a transaction with the access point which means it disconnects.

What I'd expect to happen is for the nodes to retry the connection in the background rather than stay in the failed state and not connect up.

nopnop2002, 2025-04-02:
>I'm not sure as to how the packets are sent over the DHPCS from the Root to the Child nodes

1. The root node connects to the WiFi router in Station mode.
The IP address of the root node is obtained from the WiFi router.

2. The root node waits for child nodes in SoftAP mode.

3. The leaf nodes connect to the SoftAP of the root node.
The leaf node's IP address is obtained from the root node.

BR-Coding-cmd, 2025-04-02:
Thank you for the clarification. I did have a base knowledge of it, and that has cemented it.

Ahh yes, well I've checked it and now have turned off the config for it.

nopnop2002, 2025-04-03:
>Ahh yes, well I've checked it and now have turned off the config for it.

will it connect to the Vertx broker without  IPv6  turned off ?

BR-Coding-cmd, 2025-04-03:
I have been successful in connecting with example connect yes.

closed 2025-04-24 by BR-Coding-cmd