espressif/esp-mesh-lite #155: About example sdkconfig.defaults

nopnop2002, 2025-03-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

This repository provides several example code, but I have a question about the sdkconfig.defaults in the samples.

The following is registered in some sdkconfig.defaults, but I don't think this is necessary.

This will affect builds for the ESP32C series with 2M flash

```
#
# Serial flasher config
#
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y

#
# Partition Table
#
CONFIG_PARTITION_TABLE_CUSTOM=y
```

And I wonder why this default value is needed.

```
#
# FreeRTOS
#
CONFIG_FREERTOS_UNICORE=y
```

nopnop2002, 2025-03-18:
@tswen 

thank you.

tswen, 2025-05-19:
https://github.com/espressif/esp-mesh-lite/commit/c40b18e56562bec371149199bfb815522046a658

The reported issue has been fixed in the master branch. Please verify and close the issue if no additional problems are found.

closed 2025-05-19 by nopnop2002