espressif/esp-mesh-lite #157: Feature requests for the no_router sample

nopnop2002, 2025-03-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

I would like to add the following features to this sample:

- Sending messages from the root to all leaves

- Sending messages from all leaves to the root

thank you.

### Use cases

This is what I'm thinking.
```
Bluerooth spp <-----> root <-----> leaf
```

tswen, 2025-03-25:
You can refer to https://gist.github.com/tswen/25d2054e868ef2b6c798a3ca05f77c7f

nopnop2002, 2025-03-25:
@tswen 

Thank you for your reply.
I was able to send and receive messages between root and leaf.

```
I (351261) no_router: System information, channel: 11, layer: 1, self mac: a4:cf:12:05:c6:34, parent bssid: 00:00:00:00:00:00, parent rssi: -120, free heap: 203508
I (351266) no_router: Child mac: 3c:71:bf:9d:bd:00
1: 2, 3c:71:bf:9d:bd:00, 192.168.5.2
2: 1, a4:cf:12:05:c6:34, 0.0.0.0
[send to broadcast] number: 123456
[recv from child] level: 2, mac: 3c:71:bf:9d:bd:00
```

closed 2025-03-25 by nopnop2002