espressif/esp-mesh-lite #158: The files in the ESP component registry do not match those in the master branch

nopnop2002, 2025-03-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The files in the ESP component registry and the master branch are identical.

### Actual behavior (suspected bug)

The files in the ESP component registry do not match those in the master branch.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

- files in master branch

```
$ ls -l components/mesh_lite/include/esp_mesh_lite_core.h
-rw-rw-r-- 1 nop nop 47028  3月 18 07:49 components/mesh_lite/include/esp_mesh_lite_core.h
```

- file in esp componentregstory
```
$ ls -l managed_components/espressif__mesh_lite/include/esp_mesh_lite_core.h
-rw-rw-r-- 1 nop nop 44727  3月 22 11:25 managed_components/espressif__mesh_lite/include/esp_mesh_lite_core.h
```

### Project release version

ESP-Mesh-Lite Component v1.0.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Shell

Bash

tswen, 2025-03-26:
New commits are continuously merged into the master branch, so the master branch contains all the latest features and bug fixes. Once the content is refined and thoroughly tested, a new version is released to the ​ESP Component Registry.

nopnop2002, 2025-03-26:
When a new version is released to the ESP Component Registry, the version number should be incremented.

closed 2025-03-26 by nopnop2002