espressif/esp-mesh-lite #159: A node become root even though it's disallowed

Aric5301, 2025-03-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Node stays as an idle node

### Actual behavior (suspected bug)

Node becomes root node

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Setup ESP-IDF v5.4
Download Mesh Local Control example
Set ssid to hidden (I unfortunately haven't tested without it so I can't say whether that's relevant or not)
Between the init and the start of mesh lite, call `esp_mesh_lite_set_disallowed_level(1)` or `esp_mesh_lite_set_router_min_rssi_threshold(0)` (I've tried both)
Program a ESP32-S3 dev board with the example, and run it.

### Project release version

v1.0.1

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Irrelevant

### Shell

other (details in Additional context)

tswen, 2025-04-01:
May I ask which version you are using? On my end, I cannot reproduce the issue with either the ​master branch or ​version 1.0.1. When the device calls `esp_mesh_lite_set_disallowed_level(1)`, It will not attempt to connect to the router.

Aric5301, 2025-04-01:
I'm using version 1.0.1.
Did you have the ssid set to hidden like in my setup? That's the only difference I made compared to the supplied Mesh Local Control example

tswen, 2025-04-03:
I couldn't replicate the problem on my end. Would you mind sharing your code and the full logs for further investigation?

nopnop2002, 2025-04-11:
>Between the init and the start of mesh lite, call esp_mesh_lite_set_disallowed_level(1) or >esp_mesh_lite_set_router_min_rssi_threshold(0) (I've tried both)

These are not required for the Local Control example.
Select root/leaf in menuconfig.