espressif/esp-mesh-lite #160: Node Seems to Be Stuck in a “Quasi-Root” Mode

Aric5301, 2025-03-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Node either becomes a proper root or becomes a leaf node

### Actual behavior (suspected bug)

Node doesn't establish a stable connection to any parent node. Keeps printing mainly error 201.
When restared through the reset pin (i.e. when restarted **and** having the RTC memory reset) the node connects to a parent node (node 1 in my setup) normally.
If restarted through the program (with either `esp_restart()` or `abort()`), the situation persists. To my understanding, due to esp-mesh-lite saving some data in RTC memory which makes the next boot not really "clean".

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Setup ESP-IDF v5.4
Download Mesh Local Control example
Set ssid to hidden (I unfortunately haven't tested without it so I can't say whether that's relevant or not)
Program two ESP32-S3 dev

board with the example, and run them in a setup like in the attached picture:
[image]
Solid lines represent a good link (~-55dBm), while the dashed line between node 2 and the router represents a bad link. It's somewhat difficult for me to precisely define what a bad link is, since in my setup this link was created due to signal leakage and not intended.

Now I turn both node at the same time. The resulting topology is that node 1 becomes root, and node 2 becomes a leaf in layer 2.
Next, I disconnet node 1 from 2. This results in node 2 wanting to become the new root. To my understanding (I don't have access to the actual code since it comes pre-compiled), it manages to receive at least one beacon frame from the router, but does not manage to establish a connection to the router. This is what I call a "quasi-root" mode.
Next, I reconnet node 1 and 2, Node 2 doesn't connect to node 1 and remains forever stuck in this quasi state until I reset it.
When I completely eliminated the so called bad link between node 2 and the router the issue was no longer appearant, which is what made me think that the existance of this bad link is indeed what's going on.

### Project release version

v1.0.1

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Irrelevant

### Shell

other (details in Additional context)

tswen, 2025-04-03:
Are Node 1 and Node 2 on the same firmware? Any `set_allowed_level/set_disallowed_level` restrictions? When Node 2 keeps failing (event 201), is Node 1 online or offline? Please share logs/code for deeper analysis.

nopnop2002, 2025-04-11:
esp-mesh-lite may be getting some information from nvs.
Clearing nvs and then building may change the situation.