espressif/esp-mesh-lite #161: Ghost child nodes remain as connected

Aric5301, 2025-03-31:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Ungracefully disconnected child nodes are not printed as connected (the "Child mac:' print)

### Actual behavior (suspected bug)

Ungracefully disconnected child nodes are printed as connected (the "Child mac:' print) for at least several minutes (I didn't explicitly tried to test it for a long duration

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Setup ESP-IDF v5.4
Download Mesh Local Control example
Set ssid to hidden (I unfortunately haven't tested without it so I can't say whether that's relevant or not)
board with the example, and run them in a setup like in the attached picture:

[image]

Now I turn both nodes at the same time. The resulting topology is that node 1 becomes root, and node 2 becomes a leaf in layer 2.
Next, I violently disconnet node 1 from 2. Violently means that I literally disconnect the RF cable connecting them.
Node 1 keeps logging that node 2 is connected to it.
When I reconnect the 2 nodes, only that I see "station disconnected" logs followd by the successfull reconnection of the two.

### Project release version

v1.0.1

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Irrelevant

### Shell

other (details in Additional context)

tswen, 2025-04-01:
When a node abnormally disconnects (e.g., sudden power loss or forced reset), since it doesn't complete the Wi-Fi DEAUTH (deauthentication) handshake with its parent node, the parent node's SoftAP cannot immediately detect that the Station (child node) has gone offline.

The parent node's SoftAP will maintain the Station's connection state until no data frames are received within the ​inactive timeout period (default: 300 seconds), at which point it will consider the connection terminated and remove the Station from its sta_list.

You can adjust this timeout period (in seconds) using the esp_wifi_set_inactive_time() API. For example, setting a shorter duration (e.g., 60 seconds) can accelerate the cleanup of invalid connections.

Aric5301, 2025-04-01:
Understood.
Thank you very much!

closed 2025-04-01 by Aric5301