espressif/esp-mesh-lite #166: Mesh-Lite hosting via DPP

BR-Coding-cmd, 2025-05-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I've got a client requesting the limits of our products security which utilises mesh-lite, and I'm drawing a blank when it comes to the support of the mesh-lite protocol when it comes to wifi and its use of the wifi stack. 

Our communication layer is encrypted, but the level of security thus far is the supported 64 byte password for the mesh network.

The Questions:

1.  is it possible to use a certificate in the wifi connection up to the Access Point?
2. Is it possible to encrypt the data link between nodes on the mesh network to a higher level than a password? 

if it is possible could someone point me in the right direction or explain as to how it may be done?

nopnop2002, 2025-05-10:
>is it possible to use a certificate in the wifi connection up to the Access Point?

You can use HTTPS protocol for communication between root node and AP.

>Is it possible to encrypt the data link between nodes on the mesh network to a higher level than a password?

ESP-MESH-LITE uses this for communication between nodes.
```
esp_err_t esp_mesh_lite_try_sending_msg(char* send_msg,
                                        char* expect_msg,
                                        uint32_t max_retry,
                                        cJSON* req_payload,
                                        esp_err_t (*resend)(const char* payload));
```

The payload is JSON.
JSON cannot handle binary data.
The data encrypted using mbedtls must be converted to a null-terminated string using base64.

xcguang, 2025-05-12:
Raw data is supported by esp_mesh_lite_try_sending_raw_msg.

BR-Coding-cmd, 2025-05-12:
> You can use HTTPS protocol for communication between root node and AP.

Yes, this is true and we use this for a different service within the app. However the root connection between master and AP is what I'm curious about as currently we're using "WPA3-PSK". From what I can tell there isn't any other way for the device to get on the network that also is supported by mesh-Lite

Thank you for the suggestion of MD5, I shall look into it.

What I'm curious about is the pre-existing data security between nodes, master and the outside world. Beyond altering the library, what I gather is the background information and of the mesh is only secured via the 64-bit password that can be set for the IoT-bridge SAP. Which I'm guessing that in order to encrypt the data between nodes there would have to be a custom callback and forwarding function(s) to pass the data up to the master.

nopnop2002, 2025-05-12:
@xcguang 

Thank you. 

@BR-Coding-cmd 

>Which I'm guessing that in order to encrypt the data between nodes there would have to be a custom callback and forwarding function(s) to pass the data up to the master.

Callbacks for transferring data between nodes are built into MESH-Lite.

BR-Coding-cmd, 2025-05-12:
@nopnop2002 
Yes, I am talking about the root node here and it's communication via WiFi to the Access Point.

I understand that mesh communication of node-node or node to master can be achieved with the raw_msg function, however it's the actual WiFi layer that I'm referring to. Is it possible to make the WiFi connection from root to Access Point secure?

BR-Coding-cmd, 2025-05-13:
I appreciate the assistance and shall look into the ssh and sftp solutions,

Thanks,
BR

nopnop2002, 2025-05-15:
@BR-Coding-cmd 

I found the following in the documentation:
https://github.com/espressif/esp-mesh-lite/blob/master/components/mesh_lite/User_Guide.md
```
Communication between nodes within a Mesh network can be encrypted with AES128 using esp_mesh_lite_aes_set_key
```

@tswen 

Will simply calling this encrypt messages between nodes?

If so, when should I call this function?

tswen, 2025-05-15:
```c
    esp_mesh_lite_config_t mesh_lite_config = ESP_MESH_LITE_DEFAULT_INIT();
    esp_mesh_lite_init(&mesh_lite_config);

    app_wifi_set_softap_info();

    //attention this function must be called before Mesh-Lite initialization.
    //esp_mesh_lite_aes_set_key(key, 128);

    esp_mesh_lite_start();

```

closed 2025-06-03 by BR-Coding-cmd