espressif/esp-mesh-lite #167: esp_mesh_lite_try_sending_raw_msg does not update seq

nopnop2002, 2025-05-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

esp_mesh_lite_try_sending_raw_msg does update seq

### Actual behavior (suspected bug)

esp_mesh_lite_try_sending_raw_msg does not update seq

### Error logs or terminal output

```shell
I (1677015) mesh_lite_to_root_handler: seq=0
I (1687017) mesh_lite_to_root_handler: seq=0
I (1697018) mesh_lite_to_root_handler: seq=0
I (1707015) mesh_lite_to_root_handler: seq=0
```

### Steps to reproduce the behavior

I used esp_mesh_lite_try_sending_raw_msg to transfer data from Leaf to Root.
```
esp_err_t esp_mesh_lite_try_sending_raw_msg(uint32_t msg_id,
                                            uint32_t expect_resp_msg_id,
                                            uint32_t max_retry,
                                            const uint8_t* data,
                                            size_t size,
                                            esp_err_t (*raw_resend)(const uint8_t* data, size_t size));
```

The message reception handler on the Root side is this.
```
static esp_err_t mesh_lite_to_root_handler(uint8_t *data, uint32_t len, uint8_t **out_data, uint32_t* out_len, uint32_t seq)
{
    ESP_LOGI(__FUNCTION__, "seq=%"PRIi32, seq);
    //ESP_LOG_BUFFER_HEXDUMP(__FUNCTION__, data, len, ESP_LOG_INFO);
    *out_len = 0;
    return ESP_OK;
}
```
I found seq is always 0.

### Project release version

ESP-IDF v5.4.1-dirty

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 20.04.6 LTS

### Shell

Bash

### Additional context

```
git clone https://github.com/nopnop2002/esp-mesh-lite-examples
cd esp-mesh-lite-examples/mesh_ftp_client
idf.py menuconfig
idf.py flash
```

tswen, 2025-05-14:
[image]
The issue has been fixed in the master branch, but since the 1.0.2 tag hasn't been released yet, the fix is not included in version 1.0.1. You can use the latest component from the master branch instead.

nopnop2002, 2025-05-14:
Thank you for your reply.
I hope the 1.0.2 tag will be released as soon as possible.

closed 2025-05-14 by nopnop2002