espressif/esp-mesh-lite #171: Issue with MESH Lite Communication After Merging with MQTT — Data Not Exchanging Between Master & Slave

gajananmaske1234, 2025-05-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi all,

I'm working on a project where:
Master device connects to a Wi-Fi router and publishes data using MQTT (this part is working fine ✅).
Slave device is connected to the master via ESP-MESH Lite (using the no_router example as the reference).

 **Working :** 
MQTT connection and data publish/subscribe from master to cloud.

MESH network formation:
Logs from slave show parent MAC as the master.
Logs from master show child MAC as the slave.

Devices are connected in the MESH network (confirmed via logs).
Both MQTT and mesh components initialize without error.

  **Not Working:** 
After merging the no_router (MESH Lite) example with my MQTT code, the master and slave are no longer able to exchange data over MESH. I’ve confirmed:

 **What I Expect**
1. Master should send/receive data from slave over MESH Lite (no internet/router needed for MESH path).
2. At the same time, Master should continue its MQTT communication with the cloud over Wi-Fi.

 **What I'm Looking For**
Any working example that combines MESH Lite and MQTT?
Any suggestions on how to debug mesh data transfer specifically?
Best practices to combine Wi-Fi STA (for MQTT) and MESH (for slave comm) cleanly?

nopnop2002, 2025-05-16:
>Slave device is connected to the master via ESP-MESH Lite (using the no_router example as the reference).

Both root and leaf nodes can communicate directly with the MQTT broker, but why not?

BR-Coding-cmd, 2025-05-16:
I would suggest using the mesh Lite example of "local control" as you are using a router (Access Point) and you have a master connected to said router and in some way or another controlling it.

the no route configuration is best used when the mesh is self contained

gajananmaske1234, 2025-05-16:
I am using already developed code for MQTT with provising using scanner.
Mesh lite example : https://github.com/nopnop2002/esp-mesh-lite-examples/tree/main/router_less_message_exchange

Router   <-       Parent Node       <->        child node 
           **MQTT**                     **MESH LITE**

nopnop2002, 2025-05-16:
>I want to connect root device only to router rest of the communication should be on elite mesh with no router example.

I have confirmed that only the root node connects to WiFi, and that the root and leaf nodes can communicate internally.

I think there is something wrong with your code.

tswen, 2025-05-19:
https://github.com/espressif/esp-mesh-lite/edit/master/examples/mesh_local_control/sdkconfig.defaults#L21-L24

Some necessary configuration items also need to be checked for correct settings.