espressif/esp-mesh-lite #176: Query: Master has IP then logs "Mesh network cannot send join_me_request when not connected to a router"

BR-Coding-cmd, 2025-06-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I have a mesh of 1 node with 2 other meshs in the same room.

I can see that the `Test Log #### [vid mesh id dont match]` twice and that confirms the above

What I'm confused about is the following

[image]

tswen, 2025-06-10:
Has this device obtained an IP address normally?

BR-Coding-cmd, 2025-06-10:
yes this device had an IP from the router

BR-Coding-cmd, 2025-06-11:
[idf_py_stdout_output_6060-joinMeReqIssue.txt](https://github.com/user-attachments/files/20688505/idf_py_stdout_output_6060-joinMeReqIssue.txt)

tswen, 2025-06-16:
[image]
Comparing the two logs, the right log is missing the "mesh lite internal got IP" event (highlighted in green box). Did you implement any blocking operations in your application-layer "got IP" event handler that might be affecting the event task?

BR-Coding-cmd, 2025-06-20:
Thank you for your patience, your reply slipped past me
> application-layer "got IP" event handler that might be affecting the event task?

my got IP event handler is written as shown below;
```
static void ip_event_sta_got_ip_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    ESP_LOGW(TAG, "Driver has IP from AP");
    if(event_id == IP_EVENT_STA_GOT_IP) {
        WiFi_g = true;
        ESP_LOGW(TAG, " Wifi connection, wifi_g : %d", WiFi_g);

        if(!MQTT_started_FLAG)
        {
            ESP_LOGW(TAG, "Start MQTT task");
            mqtt_app_start();
            xTaskCreate(&mqtt_tree, "Mesh_Tree", 3 * ONE_KILOBYTE, NULL, 5, &Online_Handler);
            MQTT_started_FLAG = true;
        }

    } else if (event_id == IP_EVENT_STA_LOST_IP) {
        WiFi_g = false;
        ESP_LOGW(TAG, "No Wifi connection, wifi_g : %d", WiFi_g);
    }
}
```

tswen, 2025-06-24:
If this issue is easily reproducible, you can verify it by:

1. Adding debug logs ​before ip_event_sta_got_ip_handler exits​
2. Inserting additional debug logs ​inside the mqtt_app_start API